arendst/Tasmota #4619: OTA unavailable on Arduino IDE 2.5.0 (Beta 1)

reaper7, 2018-12-13:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : IDE arduino 1.8.8, Current arduino master branch for esp8266 2.5.0b1 - esp8285 board
- [x] Hardware used : sonoff t1, sonoff s20, sonoff rf
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :

Compilation sonoff-tasmota on current arduino master branch for esp8266(8285 board) causes OTA unavailable. 
Reference: https://github.com/esp8266/Arduino/issues/5487

ascillato, 2018-12-13:
Hi,

Thanks for the report, but have you tested with latest Tasmota v6.3.0.17 and core 2.5.0-beta1 ? :thinking: 

Have you tested the precompiled bins with latest STAGE core (http://thehackbox.org/tasmota/020500/) ?

`OTA` and `OTAurl` are working fine in Tasmota. Tasmota code identifies the core and uses the correct `OTAUpdate` syntax for each core.

That Arduino issue you are refering is not happening in Tasmota code. The `Arduino OTA` is another OTA option usable with Arduino IDE software as a WIFI COM port. That option is disabled in Tasmota by default. If you want to use that, you have to compile Tasmota yourself with the key `USE_ARDUINO_OTA` enabled in _my_user_config.h_ file. Nevertheles, Tasmota uses the mDNS discovery with ArduinoOTA as explained as a requirement in the referenced Arduino Issue. So, will work too.

**devyte** has explained that the issue that the user **Swiftnesses** had, was just the use of an incomplete example. So, it is not a code bug in the beta arduino core. It was just a documentation issue:

> Now, MDNS.update() is mandatory, or it won't work. No in-between half-sometimes-maybe working state.
> About the examples, you're correct, and art of the problem was that nobody knew the need of calling update() and that the examples weren't quite correct.
> I'll keep this open to track fixing them.

If you have tested, please try again. Might be just a wifi issue. Also remember that you have to choose the firmware with the same memory map than you have in your previous firmware (1MB or 4MB).

Closing issue as can't be reproduced, sorry.

reaper7, 2018-12-13:
so... arduino and sonoff-tasmota updated from git's yesterday,
same code compiled on 2.4.2 is visible

visible for OTA:
```
Program Version     6.3.0.17(sonoff)
Build Date & Time   2018-12-12T23:02:18
Core/SDK Version    2_4_2/2.2.1(cfd48f3)
```

not visible for OTA:
```
Program Version     6.3.0.17(sonoff)
Build Date & Time   2018-12-12T23:02:18
Core/SDK Version    STAGE/3.0.0-dev(c0f7b44)
```

I tested this many times flashing first or second binary

andrethomas, 2018-12-13:
The binaries at http://thehackbox.org/tasmota/020500/ have been rebuilt many times and tested extensively over the last few weeks - every day we updated the core and performed tests.

The last build was earlier this evening and tested in all possible scenarios and no point of failure was found yet.

[image]

reaper7, 2018-12-13:
what is wrong?
all updated now and still the same problem:
[image]

maybe something with settings on new core???
 [image]

but I do not change nothing only SDK

andrethomas, 2018-12-13:
The only thing we do different is use Generic ESP8286 as opposed to you using 8285, but since the microcontroller core is the same so in theory, it should not matter.

The other thing is we use LwIP V2 High Bandwidth Low Flash (hb2n) - I am not sure what impact this has in your scenario.

andrethomas, 2018-12-13:
Full log dump of a ESP8285 Wemos D1 mini doing OTA

```
21:16:20 RSL: RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin"}
21:16:20 RSL: RESULT = {"Upgrade":"Version 6.3.0.17 from http://thehackbox.org/tasmota/020500/sonoff.bin"}
21:16:22 HTP: Web server stopped
21:16:23 HTP: Web server active on sonoff1-6382.local with IP address 192.168.42.222
21:16:24 HTP: Web server stopped
21:16:49 HTP: Web server active on sonoff1-6382.local with IP address 192.168.42.222
21:16:50 RSL: UPGRADE = Successful. Restarting
21:16:52 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
va6df83be
@cp:0
ld

00:00:00 Project sonoff Sonoff (Topic sonoff1, Fallback DVES_B118EEEE, GroupTopic sonoffs) Version 6.3.0.17(aec8430-minimal)-STAGE
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff1-6382...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff1-6382 with IP address 192.168.42.222
00:00:07 RSL: INFO1 = {"Module":"Generic","Version":"6.3.0.17(aec8430-minimal)","FallbackTopic":"DVES_B118EEEE","GroupTopic":"sonoffs"}
00:00:07 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-6382","IPAddress":"192.168.42.222"}
00:00:07 RSL: INFO3 = {"RestartReason":"Software/System restart"}
21:17:13 HTP: Web server stopped
21:17:49 HTP: Web server active on sonoff1-6382 with IP address 192.168.42.222
21:17:51 RSL: UPGRADE = Successful. Restarting
21:17:52 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
va6df83be
@cp:0
ld

00:00:00 Project sonoff Sonoff (Topic sonoff1, Fallback DVES_B118EEEE, GroupTopic sonoffs) Version 6.3.0.17(aec8430-sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff1-6382...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff1-6382.local with IP address 192.168.42.222
```

I think this issue is closed and shut case now.

reaper7, 2018-12-13:
in my first post there are references to the 'bug'...

I thought we were talking about the same thing :-) 

When you gave the logs I understood that we are talking about other matters...
but the discussion was blocked

andrethomas, 2018-12-13:
Yes, apologies for not understanding your issue from your first post. After some discussion with @ascillato it becomes clear.

ascillato2, 2018-12-13:
Thanks for reporting and testing.

Fixed with PR 

**Console Output:**

```
00:00:00 CFG: Cargado desde flash en 3F5, Conteo 22
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_59171919, GroupTopic sonoffs) Versión 6.3.0.17(sonoff)-STAGE
00:00:00 UPL: Arduino OTA Habilitado Puerto 8266        <===== 
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 DNS: Inicializado
00:00:04 HTP: Servidor web activo en living-5913.local con dirección IP 192.168.1.33
00:00:06 MQT: Intentando conectar...
00:00:06 MQT: Conectado
00:00:06 MQT: tele/living/LWT = Online (Grabado)
00:00:06 MQT: cmnd/living/POWER = 
00:00:06 MQT: Suscribir a cmnd/living/#
00:00:06 MQT: Suscribir a cmnd/sonoffs/#
00:00:06 MQT: Suscribir a cmnd/DVES_59171919/#
00:00:06 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.3.0.17(sonoff)","FallbackTopic":"DVES_59171919","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
00:00:06 MQT: tele/living/INFO3 = {"RestartReason":"External System"}
00:00:06 MQT: stat/living/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/living/POWER1 = OFF
00:00:10 APP: Conteo Reinicios 12
00:00:10 CFG: Grabado a la flash en 3F4, Conteo 23, Bytes 3584
00:00:14 MQT: tele/living/STATE = {"Time":"1970-01-01T00:00:14","Uptime":"0T00:00:14","Vcc":3.012,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":2,"RSSI":50}}
```

**Arduino IDE - OTA PORT:**

[image]

closed 2018-12-13 by ascillato2