arendst/Tasmota #4622: Pulsing LED (Breathable LED effect) for status lights etc.

jascdk, 2018-12-14:
Hi.

Would it be possible to integrate a "Breathable LED effect" in the LED light options.

Mainly for use in projects to indicate that MQTT / wifi etc. is not connected.

I used this code in another project linked below:

https://arduining.com/2015/08/20/nodemcu-breathing-led-with-arduino-ide/

It gives a nice effect - a bit more cool than just standard blinking :)

Jason2866, 2018-12-14:
It is not the goal of Tasmota to have cool LED blinking effects which needs rare resources.
LED indication for status is already implemented
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

ascillato2, 2018-12-14:
Nice idea. Added your request to the list of requested effects in the already open issue #3480

Thanks for sharing your ideas.

closed 2018-12-14 by ascillato2