arendst/Tasmota #4623: Support for multiple channels

jpadie, 2018-12-14:
Hi

Is there a tutorial or similar for how tasmota might be extended to cater for non-sonoff scenarios?  I'm in the process of building a 10ch central heating controller and am thinking of basic it on tasmota. 

Tia
Justin

Jason2866, 2018-12-14:
Yes take a look in wiki https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017

If you need further assistance with that, please find us in the Tasmota Support Chat.

ascillato2, 2018-12-14:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-14 by ascillato2

Jason2866, 2018-12-14:
Since you have a question not an issue closing is just the way github issue works.
For questions the Tasmota community on Discord will give you a warm welcome

ascillato, 2018-12-14:
@jpadie Sorry if you felt that way. It was never the intention to produce that feeling in you. Sorry.

Github issues are meant for bugs reporting and also we provide 3 templates that you have ignored completely. Those templates gives us a lot of information for troubleshoting and bug reporting. Github issues are not a forum.

As explained in the Tasmota [Readme](https://github.com/arendst/Sonoff-Tasmota#support-information):

* **Tasmota Wiki**: Github Wiki (https://github.com/arendst/Sonoff-Tasmota/wiki)
* **Tasmota Forum**: Google Forum (https://groups.google.com/d/forum/sonoffusers)
* **Tasmota Support**: Discord (https://discord.gg/Ks2Kzd4)
* **Tasmota Software Bug Reporting**: Github (https://github.com/arendst/Sonoff-Tasmota/issues/new/choose)

For general questions we maintain the Tasmota support chat where a bunch of great people are there, giving support to this free software.

Sorry again if you felt the closing as agressive. It was never the intention for that. We track and follow the issues to solve them and to improve Tasmota, but we do that for free and in our free time and as a hobby, so, we must be as organized as possible because time is very limited here. If we left the answered issues opened, most of the people don't close them and don't give feedback. So, in very few time we will reach an unmanageable amount of issues that can not be tracked/followed anymore. In January of this year (2018) we had 1248 issues opened and a huge amount of people were complaining about their issues that were left unanswered and were left behind. So, we (the Tasmota Community) were working hard looking for different ways of improving and making more efficient this issue tracking.

We address and solve about 1000 issues every 3 month. Now that number is decreasing because most of questions and support are being addressed in the Tasmota Chat that has proved to be way more efficient for Q&A and it is a growing community. Great ideas are coming from there and a lot of people are sharing their experiences.

Anyway, you have received a valid answer to your question. If that don't satisfy your needs, you are free to ask again or clarify your idea (that is why the chat is more efficient and faster). There is no need for complainig.

In Discord Chat you don't need to provide any personal data. It is just another communication channel.

Hope you undertand now, that the closing was nothing personal neither agressivity upon you. It was just the way it is organized. Questions are better to be addresed in the Chat. Questions in Github are answered and closed (not tracked), but are not locked (as in other repositories) so you can continue the thread if you need to and will be reopened if it is needed.

I hope this misunderstanding was cleared and that you can continue with Tasmota. Thanks.