arendst/Tasmota #4624: Shelly1 input not published on MQTT (does not switch relay either)

s0170071, 2018-12-14:
**Describe the bug**
_A clear and concise description of what the bug is._
Shelly1 - 
1. No MQTT Message fired if relay is toggled using the switch input.
2. Relay does not toggle if the switch input is toggled.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Shelly1 V3
- [x ] _Development/Compiler/Upload tools used_ :  ESP Easy Flasher 0.02.005
- [ ] _If a pre-compiled release or development binary was used, which one?_ : sonoff-classic.bin
- [ ] _You have tried latest release or development binaries?_ : 
- [x ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
15:10:52 CMD: status 0
15:10:52 MQT: stat/shelly1/STATUS = {"Status":{"Module":46,"FriendlyName":["Sonoff"],"Topic":"shelly1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"off","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:10:52 MQT: stat/shelly1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:04:34","StartupUTC":"2018-12-14T14:06:18","Sleep":100,"BootCount":19,"SaveCount":49,"SaveAddress":"F6000"}}
15:10:52 MQT: stat/shelly1/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:35:18","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:10:52 MQT: stat/shelly1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MNET",""],"TelePeriod":300,"SetOption":["00008009","55898000","00000000"]}}
15:10:52 MQT: stat/shelly1/STATUS4 = {"StatusMEM":{"ProgramSize":483,"Free":520,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","6050A790","008003A8","210217C6","000001C0"]}}
15:10:52 MQT: stat/shelly1/STATUS5 = {"StatusNET":{"Hostname":"shelly1-4773","IPAddress":"192.168.1.150","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:1D:92:A5","Webserver":2,"WifiConfig":5}}
15:10:52 MQT: stat/shelly1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.52","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D92A5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:10:52 MQT: stat/shelly1/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 14 14:10:52 2018","Local":"Fri Dec 14 15:10:52 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
15:10:52 MQT: stat/shelly1/STATUS10 = {"StatusSNS":{"Time":"2018-12-14T15:10:52","Switch1":"OFF"}}
15:10:52 MQT: stat/shelly1/STATUS11 = {"StatusSTS":{"Time":"2018-12-14T15:10:52","Uptime":"0T00:04:34","Vcc":3.218,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MNET","BSSId":"5C:49:79:2C:0C:61","Channel":11,"RSSI":92}}}

**To Reproduce**
 
- program, configure as shelly1 and try to toggle the relay by applying AC power to the switch input. 
(Beware - high voltage )
Problem: the internal relay is not switched then the input toggles and there is no MQTT message about the intput toggling. There is, however a log line that reads 
'15:06:33 MQT: cmnd/off/POWER1 = TOGGLE' 
which is odd as the device is not named "off" but "shelly1" (full log below)

**Expected behavior**
I would expect immediate relay switch when the switch input is toggled or at least an immediate MQTT message about the input change. (so I can do the switching in OpenHab)

Console shows: 

00:00:00 Project sonoff Sonoff (Topic shelly1, Fallback DVES_1D92A5, GroupTopic sonoffs) Version 6.3.0-2_3_0
00:00:00 WIF: Connecting to AP1 MNET in mode 11N as shelly1-4773...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on shelly1-4773.local with IP address 192.168.1.150
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/shelly1/LWT = Online (retained)
00:00:08 MQT: cmnd/shelly1/POWER = 
00:00:08 MQT: tele/shelly1/INFO1 = {"Module":"Shelly 1","Version":"6.3.0","FallbackTopic":"DVES_1D92A5","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/shelly1/INFO2 = {"WebServerMode":"Admin","Hostname":"shelly1-4773","IPAddress":"192.168.1.150"}
00:00:08 MQT: tele/shelly1/INFO3 = {"RestartReason":"Power on"}
00:00:08 MQT: stat/shelly1/RESULT = {"POWER":"OFF"}
00:00:08 MQT: stat/shelly1/POWER = OFF
15:06:29 MQT: cmnd/off/POWER1 = TOGGLE
15:06:31 MQT: cmnd/off/POWER1 = TOGGLE
15:06:33 MQT: cmnd/off/POWER1 = TOGGLE
15:06:34 MQT: tele/shelly1/STATE = {"Time":"2018-12-14T15:06:34","Uptime":"0T00:00:16","Vcc":3.239,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MNET","BSSId":"5C:49:79:2C:0C:61","Channel":11,"RSSI":84}}
15:06:34 MQT: tele/shelly1/SENSOR = {"Time":"2018-12-14T15:06:34","Switch1":"ON"}
15:06:35 MQT: cmnd/off/POWER1 = TOGGLE

Note: the device is named "shelly1" When I apply power to the switch input I see 
`15:06:31 MQT: cmnd/off/POWER1 = TOGGLE`
messages. They do not show up on the MQTT broker. The relay is also not switched.

ascillato, 2018-12-14:
Hi,

the message in your log: `15:06:35 MQT: cmnd/off/POWER1 = TOGGLE` means that the mqtt message is being sent to your broker. So, your issue is on your broker side. Sorry.

You need to check that. You can use a MQTT sniffer like Hivemq's web client.

If you need further support on this, please find us in the Tasmota Support Chat. Thanks

s0170071, 2018-12-14:
Why does that particular message go by the name "/off/ " and not "/shelly1/" like the others ?
On the MQTT broker I do not see that "off" message.

ascillato2, 2018-12-14:
Ah, ok.

You have changed the fulltopic key. So I thought you was trying to achieve that type of message.

If you didn't want to do that and was a change you did by mistake, please:

Erase all flash and flash the precompiled bins (6.3.0.17) from: http://thehackbox.org/tasmota/

OR

Upgrade by OTA to the latest Tasmotra (6.3.0.17) from: http://thehackbox.org/tasmota/ and then do a `reset 5` command in the console to wipe out all the configuration of your flash and start over from skratch

ascillato2, 2018-12-14:
Closing this issue as it has been answered.

If you need further support on this, please find us in the Tasmota Support Chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-14 by ascillato2