arendst/Tasmota #4627: Autodiscovery sending temperature values instead of humidity

blakadder, 2018-12-14:
**Describe the bug**
Home Assistant autodiscovery config sends the data from temperature as well as unit of measurement for the humidity sensor. Meanwhile the temperature sensor has no unit of measurement.
```
00:00:07 MQT: homeassistant/sensor/3D698E_AM2301_Temperature/config = {"name":"Sonoff AM2301 Temperature","state_topic":"testsonoff/tele/SENSOR","availability_topic":"testsonoff/tele/LWT","payload_available":"Online","payload_not_available":"Offline","value_template":"{{value_json['AM2301'].Temperature}}"} (retained)
00:00:07 MQT: homeassistant/sensor/3D698E_AM2301_Humidity/config = {"name":"Sonoff AM2301 Humidity","state_topic":"testsonoff/tele/SENSOR","availability_topic":"testsonoff/tele/LWT","payload_available":"Online","payload_not_available":"Offline","unit_of_measurement":"°C","value_template":"{{value_json['AM2301'].Temperature}}"} (retained)
2
```

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : Wemos D1 mini and DHT22 sensor on pin D3
- [ ] _Development/Compiler/Upload tools used_ : 
- [ ] _If a pre-compiled release or development binary was used, which one?_ : 6.3.0.17(5f560ba-sonoff) 2.5.0 core sonoff.bin
- [ ] _You have tried latest release or development binaries?_ : absolutely
- [ ] _Provide the output of command_``status 0`` :
```
20:54:27 CMD: status 0
20:54:27 MQT: testsonoff/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"testsonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
20:54:27 MQT: testsonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:26","StartupUTC":"2018-12-14T19:52:01","Sleep":50,"BootCount":12,"SaveCount":25,"SaveAddress":"F9000"}}
20:54:27 MQT: testsonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.3.0.17(5f560ba-sonoff)","BuildDateTime":"2018-12-14T10:16:58","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
20:54:27 MQT: testsonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HTPC","HTPCext"],"TelePeriod":300,"SetOption":["000A8029","558180C0","00000248"]}}
20:54:27 MQT: testsonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":456,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
20:54:27 MQT: testsonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"testsonoff-2446","IPAddress":"192.168.31.100","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.31.1","Mac":"80:7D:3A:3D:69:8E","Webserver":2,"WifiConfig":4}}
20:54:27 MQT: testsonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.31.112","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3D698E8E","MqttUser":"mosquitto","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:54:27 MQT: testsonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 14 19:54:27 2018","Local":"Fri Dec 14 20:54:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"07:16","Sunset":"16:01"}}
20:54:27 MQT: testsonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-14T20:54:27","AM2301":{"Temperature":23.5,"Humidity":51.9},"TempUnit":"C"}}
20:54:27 MQT: testsonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-14T20:54:27","Uptime":"0T00:02:26","Vcc":2.917,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":2,"SSId":"HTPCext","BSSId":"40:31:3C:93:F0:15","Channel":7,"RSSI":56}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_

SetOption19 1 with the sensor attached

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

http://prntscr.com/lv2rhh

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-14:
Thanks for reporting

Also the unit of measurement of humidity is not C is %

blakadder, 2018-12-14:
The bug puts nothing as unit of measurement for temp and °C for humidity as is visible in the screenshot

ascillato, 2018-12-15:
Confirmated Bug:

```
22:26:51 MQT: homeassistant/sensor/591719_AM2301_Temperature/config = {"name":"LivingTest AM2301 Temperature","state_topic":"livingtest/tele/SENSOR","availability_topic":"livingtest/tele/LWT","payload_available":"Online","payload_not_available":"Offline","value_template":"{{value_json['AM2301'].Temperature}}"} (Grabado)
22:26:51 MQT: homeassistant/sensor/591719_AM2301_Humidity/config = {"name":"LivingTest AM2301 Humidity","state_topic":"livingtest/tele/SENSOR","availability_topic":"livingtest/tele/LWT","payload_available":"Online","payload_not_available":"Offline","unit_of_measurement":"°C","value_template":"{{value_json['AM2301'].Temperature}}"} (Grabado)
```

For Temperature is not sending the unit
For Humidity is sending the unit C that is wrong (should be %) and the topic sent is the same as temperature.

ascillato, 2018-12-15:
Fixed with PR https://github.com/arendst/Sonoff-Tasmota/pull/4629

Thanks for reporting :+1:

closed 2018-12-15 by ascillato2