arendst/Tasmota #4634: http timeout

s0170071, 2018-12-15:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : Shelly 1
- [ ] If a pre-compiled release or development binary was used, which one? :6.3 ota today 
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` : cant do.
```
STATUS 0 OUTPUT HERE
```
The web interface is super unresponsive. Times out on 9 out of 10 requests. Console doesn't work...
Info shows 

Program Version	6.3.0
Build Date & Time	2018-12-14T09:54:04
Core/SDK Version	STAGE/3.0.0-dev(c0f7b44)
Uptime	0T00:05:03
Flash write Count	51 at F8000
Boot Count	12
Restart Reason	Software/System restart
Friendly Name 1	Sonoff
	 
AP1 SSId (RSSI)	MNET (76%)
Hostname	shelly3-0312
IP Address	192.168.1.64
Gateway	192.168.1.1
Subnet Mask	255.255.255.0
DNS Server	192.168.1.1
MAC Address	CC:50:E3:50:01:38
	 
MQTT Host	192.168.1.52
MQTT Port	1883
MQTT Client &
 Fallback Topic	shelly3
MQTT User	DVES_USER
MQTT Topic	shelly3
MQTT Group Topic	sonoffs
MQTT Full Topic	cmnd/shelly3/
	 
Emulation	None
mDNS Discovery	Enabled
mDNS Advertise	Web Server
	 
ESP Chip Id	5243192
Flash Chip Id	1392879
Flash Size	2048kB
Program Flash Size	1024kB
Program Size	543kB
Free Program Space	204kB
Free Memory	12kB

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-12-15:
Unsupported core stage version.

closed 2018-12-15 by arendst

s0170071, 2018-12-15:
I did ota update from http://192.168.1.64/u1?o=http%3A%2F%2Fthehackbox.org%2Ftasmota%2F.
Shouldn't that be ok?

ascillato2, 2018-12-15:
> Program Version	6.3.0
> Build Date & Time	2018-12-14T09:54:04
> Core/SDK Version	STAGE/3.0.0-dev(c0f7b44)

Tasmota v6.3.0 is not fully compatible with the core that is in beta.

The link you provide is not under core 2.5.0. Seems that you have self compiled 6.3.0 with core 2.5.0

http://thehackbox.org/tasmota/release/ are Tasmota v6.3.0 under core 2.3.0
http://thehackbox.org/tasmota/ are latest Tasmota v6.3.0.17 but under core 2.3.0
http://thehackbox.org/tasmota/020402/ are latest Tasmota v6.3.0.17 but under core 2.4.2
http://thehackbox.org/tasmota/020500/ are latest Tasmota v6.3.0.17 but under core 2.5.0

v6.3.0.17 is the version that supports the core 2.5.0-beta

Remember that 2.5.0-beta is a core for testing. It has not been released.

s0170071, 2018-12-15:
When i update over the air the core gets updated as well, right?

ascillato, 2018-12-15:
Yes, of course. It is just one file all the firmware.

s0170071, 2018-12-15:
Oh. That just means my ota failed

ascillato, 2018-12-15:
Yes, sorry.

v6.3.0 with core 2.5.0 were not released by us.