arendst/Tasmota #4635: Mqtt#state type value in rules?

mzealey, 2018-12-15:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
Using rules you can track for example `Mqtt#connected` which is great, but I want to make a rule which behaves differently depending on whether or not Mqtt is currently connected.

**Describe the solution you'd like**
I can obviously track this via a variable but it seems overly complex when we are already tracking that in the C code. Can we add some rules like `Mqtt#state=0/1` for currently connected/disconnected and perhaps similar for wifi status to make it easier to test for these conditions?

andrethomas, 2018-12-15:
Hi

Could you elaborate on a specific use case for this feature?

Thanks

mzealey, 2018-12-15:
I have a thermostat connected to a boiler. Usually this is controlled via domoticz and pressing the button signals domoticz that it should turn on for a certain period of time. But if the connection to mqtt/domoticz is lost then I want the button to directly trigger the relay or set a timer or so. Basically while it has a mqtt connection it should be a dumb relay, but when it looses it it should start acting like a thermostat/boiler controller.

andrethomas, 2018-12-15:
I understand.

I think its also important to understand that it will still just be an event.

So it will only occur when the state of connection changes so you would basically get a duplication of event mqtt#connect + mqtt#state=1 and mqtt#disconnect + mqtt#state=0 respectively.

I'm not sure I see how this can be different from maintaining a variable.

@ascillato @arendst ?

ascillato2, 2018-12-15:
Closing this issue as it has been answered. If you have more comments or you need further assistant making your rules, please just ask. We are willing to help you. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-15 by ascillato2

andrethomas, 2018-12-15:
It may be possible to do this with telemetry status

```
20:10:35 MQT: tele/sonoff/STATE = {"Time":"2018-12-15T20:10:35","Uptime":"0T00:01:45","Vcc":2.997,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"NetStat":3,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SensorNet","BSSId":"B0:B9:8A:4C:57:F4","Channel":6,"RSSI":90}}
```

NetStat = 0 then both wifi and mqtt is down
NetStat = 1 wifi is up but mqtt is down
NetStat = 3 wifi is up and mqtt is up

Would this be more useful to rules @ascillato ?

ascillato2, 2018-12-15:
The triggers are already there:

```
Mqtt#Connected when MQTT is connected
Mqtt#Disconnected when MQTT is disconnected
Wifi#Connected when Wifi is connected (6.1.1c)
Wifi#Disconnected when Wifi is disconnected (6.1.1c)
```

If you have an extra var %netstat% would not make any difference. Nothing extra can be achieved with that. Would be the same as using the actual triggers.