arendst/Tasmota #4637: TH10 sensors report null

pererikl, 2018-12-16:
I've got a Sonoff TH10 with a AM2301 sensor and Tasmota 6.3.0

It's set up as "04 Sonoff TH" and "GPIO14 Sensor: 02 AM2301"
The result on the web interface is:
   AM2301 Temperature	        null°C
   AM2301 Humidity		null°%

Have i done anything wrong or is this a bug?

Jason2866, 2018-12-16:
Probably the plug isnt completly in the socket. Try with a little more strength

pererikl, 2018-12-16:
It worked! Tank you!

closed 2018-12-16 by pererikl

Emece2, 2018-12-19:
Hi, I had the same problem:
Sonoff TH10 with a AM2301 sensor and Tasmota 6.3.0 / 6.4.0
module 4
gpio14 2
And {"Time":"2018-12-19T09:22:43","AM2301":{"Temperature":null,"Humidity":null},"TempUnit":"C"}
Reviewing the power suply in the AM2301 the Vcc not arrives to the sensor due that the jack not fit completelly in. Some connector plastic sanding  , solve the problem

sblantipodi, 2018-12-19:
I'm investigating the problem.
I bought two identical DHT22 sensor from two different vendors.

Sensors seems identical but they have very very small differences in the resistances.
I noticed that the first sensor has many disconnection, now I swapped to the sensor with bigger resistance and I haven't had a disconnection since than.

Finger crossed.

dariusdev, 2019-01-24:
Try to restart, plug in and unplug and still the same situation:

AM2301 Temperature	null°C
AM2301 Humidity	null°%

How to check this sensor health in another way?

sblantipodi, 2019-01-25:
I had similar problems. There are a lot of am2301 build.
Some build are made with some resistors some other with some other resistors.

If they are build for 5v and you are using 3.3v esp8266 GPIO you could have that problems.
You have some options, 
- buy a different sensor with different resistor in the hope that you will solve the problem.
- buy a logic level converter to bump your 3.3V GPIO to 5v
- best solution buy a better sensor. Bosch BME280 for example.

dariusdev, 2019-01-28:
Problem was in the sensor. Not all contacts were properly soldered inside the sensor.  Now it works like a charm.

johnjoemorgan, 2019-11-17:
Same problem - mine was caused by the fine volcanic dust (lahar) we get locally 'blocking' the sensor.  Blew clean with a CO2 canister.

jonap78, 2020-05-11:
eI have the same problem with th16 on 8.2.0(tasmota) and am2301. Reports "null" as temperature and humidity. But when I try to connect am2301 to sonoff th16 with native firmware the sensor work properly. I think the problem in tasmota firmware. Does anybody solve the problem?

lalo-uy, 2020-05-11:
You have to use the sensors firmware. Standar fw no longer have the sensors
drivers.

jonap78, 2020-05-11:
Just upload to th16 new firmware
http://thehackbox.org/tasmota/release/tasmota-sensors.bin
Result - "null"

johnjoemorgan, 2020-05-12:
My th16 with Tasmota takes upto 10 minutes to go from 'null' to give C and
% humidity

johnjoemorgan, 2020-05-15:
Just updated to Tasmota 8.2.0 and noticed that "Dew Point" appeared in the console. Don't know if it came in with this version of Tasmota or earlier but it's there now.
[image]

'
 - platform: mqtt
    name: "Study Dew Point"
    state_topic: "castel_5478CE/tele/SENSOR"
    value_template: "{{ value_json['AM2301'].DewPoint }}"
    unit_of_measurement: "%"
'
    icon: fas:fa-dewpoint