arendst/Tasmota #4638: Migration Path from 5.14 to 6.3

laurentlemercier, 2018-12-16:
Hi Theo
I have upgraded a sonoff touch using your compiled releases. 
The touch was initially in a home compiled firmware (5.1.7 I think). I have upgraded first to 5.14 without any problem.
Then I tried to upgrade from 5.14 (classic version) to 6.3 (sensors version). It failed.
I tried to upgrade from 5.14 (classic) to 6.2.1 (classic) then from 6.2.1 (classic) to 6.3.0 (sensors), It worked.
Maybe this can interest other folks (and be precised in the migration path) ?
Best regards
Laurent
PS. : thanks for all the work you do/did. Each day I use this touch I have a thought for you :)

Jason2866, 2018-12-17:
Could you try upgrade path: 5.14 to 6.4.0

andrethomas, 2018-12-17:
Hi

It must be something with your device or flash.

See transcript upgrading from 5.14 directly to 6.4

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10373D, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-5949...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-5949.local with IP address 192.168.42.123
13:44:50 MQT: Attempting connection...
13:44:50 MQT: Connected
13:44:50 MQT: tele/sonoff/LWT = Online (retained)
13:44:50 MQT: cmnd/sonoff/POWER = 
13:44:50 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"5.14.0","FallbackTopic":"DVES_10373D","GroupTopic":"sonoffs"}
13:44:50 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5949","IPAddress":"192.168.42.123"}
13:44:50 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
13:45:25 CMD: otaurl http://thehackbox.org/tasmota/release/sonoff.bin
13:45:25 MQT: stat/sonoff/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin"}
13:45:30 CMD: upgrade 1
13:45:30 MQT: stat/sonoff/RESULT = {"Upgrade":"Version 5.14.0 from http://thehackbox.org/tasmota/release/sonoff.bin"}
13:45:32 HTP: Web server stopped
13:45:33 HTP: Web server active on sonoff-5949.local with IP address 192.168.42.123
13:45:35 HTP: Web server stopped
13:46:02 HTP: Web server active on sonoff-5949.local with IP address 192.168.42.123
13:46:02 MQT: Attempting connection...
13:46:02 MQT: Connected
13:46:02 MQT: tele/sonoff/LWT = Online (retained)
13:46:02 MQT: cmnd/sonoff/POWER = 
13:46:03 MQT: stat/sonoff/UPGRADE = Successful. Restarting
13:46:05 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff3608
@cp:0
ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10373D, GroupTopic sonoffs) Version 6.4.0(minimal)-2_4_2
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-5949...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-5949 with IP address 192.168.42.123
13:46:21 MQT: Attempting connection...
13:46:21 MQT: Connected
13:46:21 MQT: tele/sonoff/LWT = Online (retained)
13:46:21 MQT: cmnd/sonoff/POWER = 
13:46:21 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.4.0(minimal)","FallbackTopic":"DVES_10373D","GroupTopic":"sonoffs"}
13:46:21 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5949","IPAddress":"192.168.42.123"}
13:46:21 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
13:46:22 HTP: Web server stopped
13:47:01 HTP: Web server active on sonoff-5949 with IP address 192.168.42.123
13:47:02 MQT: Attempting connection...
13:47:02 MQT: Connected
13:47:02 MQT: tele/sonoff/LWT = Online (retained)
13:47:02 MQT: cmnd/sonoff/POWER = 
13:47:03 MQT: stat/sonoff/UPGRADE = Successful. Restarting
13:47:04 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
@cp:0
ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10373D, GroupTopic sonoffs) Version 6.4.0(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-5949...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-5949.local with IP address 192.168.42.123
13:47:25 MQT: Attempting connection...
13:47:25 MQT: Connected
13:47:25 MQT: tele/sonoff/LWT = Online (retained)
13:47:25 MQT: cmnd/sonoff/POWER = 
13:47:25 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.4.0(sonoff)","FallbackTopic":"DVES_10373D","GroupTopic":"sonoffs"}
13:47:25 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5949","IPAddress":"192.168.42.123"}
13:47:25 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
```

It's important to be patient during the upgrade process - you will note from the logs above that because of the firmware size of 6.4.0 the device automatically first upgrades to an intermediate minimal version and then the target version after that.

closed 2018-12-17 by andrethomas2