arendst/Tasmota #4639: MQTT breaking change regarding topics?

FauthD, 2018-12-16:
Hi, just want to mention that the order of topics have been changed in 6.3.0.17.
In Oct. timeframe (6.2 ) and before it used to be  %prefix%/%topic%/.
Now it is vice versa. %topic%/%prefix%/.
This breaks all MQTT apps like HA.

Is this on purpose or by accident? I saw there is different MQTT-lib now.
The comment text on the webserver is still the old  order.

Jason2866, 2018-12-16:
It is a feature introduced in newer version. See https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

ascillato2, 2018-12-16:
Try setoption19 0

ascillato, 2018-12-17:
Hi,

It is not a breaking change. Seems that in your old config you had SetOption59 set to 1. In the new version SetOption59 do this behaviour. In old versions do nothing.

SetOption59 | 0/1 | (6.3.0.14) Send tele/STATE in addition to stat/RESULT for commands: State, Power and any command causing a light to be turned on.
-- | -- | --

So, please go to the console and type:

```
setoption59 0
setoption19 0
```

DavidFW1960, 2018-12-17:
The HA doc here https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant says to set option 19 on?

I had to change the topic as per that doc and I now find it takes a few minutes after the restart of HA for the switch to sync again. I also added both automations from that doc but it still does this.

This in config is now returning the correct firmware version to my sensor and notifying if there is an available update.

'''
  - platform: mqtt
    name: "Coffee Maker Firmware"
    state_topic: "sonoff1/stat/STATUS2"
    value_template: 'v{{ value_json.StatusFWR.Version | replace("(sonoff)", "") }}'
  - platform: mqtt
    name: "Toothbrush Firmware"
    state_topic: "sonoff2/stat/STATUS2"
    value_template: 'v{{ value_json.StatusFWR.Version | replace("(sonoff)", "") }}'
  - platform: mqtt
    name: "Garage Door Firmware"
    state_topic: "sonoff5/stat/STATUS2"
    value_template: 'v{{ value_json.StatusFWR.Version | replace("(sonoff)", "") }}'
binary_sensor:
  - platform: template
    sensors:
      sonoff_update_available:
        value_template: >-
          {{ (states.sensor.sonoff_firmware_version_available.state > states.sensor.toothbrush_firmware.state) or (states.sensor.sonoff_firmware_version_available.state > states.sensor.coffee_maker_firmware.state) or (states.sensor.sonoff_firmware_version_available.state > states.sensor.garage_door_firmware.state)
              }}
```
I am using these automations:
```
- id: 'sonoff_firmware_installed'
  alias: "Sonoff Firmware Installed"
  initial_state: 'on'
  trigger:
    - platform: time
      at: '00:00:00'
    - platform: time
      at: '12:00:00'
    - platform: homeassistant
      event: start
  action:
  - service: mqtt.publish
    data:
      topic: sonoff1/cmnd/status
      payload: "2"
  - service: mqtt.publish
    data:
      topic: sonoff2/cmnd/status
      payload: "2"
  - service: mqtt.publish
    data:
      topic: sonoff5/cmnd/status
      payload: "2"
- id: 'Poweronstate'
  alias: "Power State Sonoff on HA Start-Up"
  initial_state: 'on'
  trigger:
    - platform: homeassistant
      event: start
  action:
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/power1"
      payload: ""
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/power2"
      payload: ""
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/Power3"
      payload: "" 
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/power4"
      payload: ""
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/state"
      payload: ""
- id: 'SonoffMQTTAuto'
  alias: "Enable MQTT discovery for all devices"
  initial_state: 'on'
  trigger:
    - platform: homeassistant
      event: start
  action:
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/SetOption19"
      payload: "1"
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/SetOption55"
      payload: "1"
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/SwitchRetain"
      payload: "0"
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/ButtonRetain"
      payload: "0"
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/PowerOnState"
      payload: "3"
  - service: mqtt.publish
    data:
      topic: "cmnd/sonoffs/PowerRetain"
      payload: "1"
```

Still doesn't update on restart of HA.

DavidFW1960, 2018-12-17:
I have also tried the exact automations in that doc with the same result.

andrethomas2, 2018-12-17:
Closing this issue as it has been answered by https://github.com/arendst/Sonoff-Tasmota/issues/4639#issuecomment-447694473

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-17 by andrethomas2