arendst/Tasmota #4640: Wemos D1 Mini, Sonoff 6.3.0 and AM2301 loses connection after some hours.

sblantipodi, 2018-12-16:
AM2301 loses connection after a couple of hours.
I'm using a Wemos D1 Mini and a DHT22 with the latest Sonoff.bin 6.3.0

I tried everythings and it seems that AM2301 loses connection after a couple of hours and it starts sending NaN values instead of temp and humidity values.

Resetting the sonoff does not help and I need to power off the sonoff by disconnecting the power and reconnect it to get correct readings again.

I'm testing this problem since weeks and I tried to use better cables, better power supply for the d1 mini and the problem still there.

I noticed that the default sleep time for the sonoff is 0.
With sleep set to 0 I loose my connection to the sensor approximately every 8-12 hours...

With a sleep set to 200 I loose my connection every 4 to 5 days but the problem still there and in this way is unusable and unreliable to me.

I connected my sonoff to my furnance and is driving my room temperature so this is a deal breaker for me.

I noticed that there are thousands of threads on the internet and even here on github of people asking for a solution to this problem.

Can someone from sonoff help us please?

Thanks for your great work guys!

Jason2866, 2018-12-16:
Release 6.3.0 is old. New release version 6.4.0 is comming soon.
The release version will be dev version 6.3.0.17. You can download from here
http://thehackbox.org/tasmota/sonoff.bin

sblantipodi, 2018-12-16:
> Release 6.3.0 is old. New release version 6.4.0 is coming soon.
> The release version will be dev version 6.3.0.17. You can download from here
> http://thehackbox.org/tasmota/sonoff.bin

@Jason2866 ok but I see no fix in the "new release" for this problem, so what's the point? thanks!

Jason2866, 2018-12-16:
Just try! There were over 1000 commits since 6.3.0. And some commits are affecting general issues

sblantipodi, 2018-12-16:
upgraded to the latest 6.0.17 as per your suggestion. still monitoring the issue and see if it happen again.

sblantipodi, 2018-12-16:
@Jason2866 is it safe to use http://thehackbox.org/tasmota/sonoff.bin instead of http://thehackbox.org/tasmota/release/sonoff.bin for "production use" like in my home with home assistant and a furnance connected to the sonoff?

Jason2866, 2018-12-16:
Yes, or wait a few days. V6.3.0.17 will become release v6.4.0. But nothing will be different.
Check wiki article Home Assistant because there are new features which needs attention!

sblantipodi, 2018-12-16:
I have just upgraded to the 6.3.0.17 version, I need a fast solution to my problem since as I said the sonoff is controlling my furnance :)

I'll keep you posted if the new firmware has solved the null readings from my temp sensor.

thanks for the answer and for the good work guys, we all appreciate your effort!

sblantipodi, 2018-12-16:
Ok arendst just release the 6.4.0 so I have just upgraded to the 6.4.0 💃

sblantipodi, 2018-12-16:
errata corridge, I am trying to upgrade to the 6.4 but I have seen that the upgrade failed from 6.3.0.17 to 6.4 with this message.

Upload Failed
Upload buffer miscompare

I need to wait the external OTA server release to address this problem.
https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#otamagic

waiting the 6.4 :)

Jason2866, 2018-12-16:
You can flash minimal version manual. After that upgrade to full sonoff.bin

sblantipodi, 2018-12-16:
@Jason2866 I really, really, appreciate your help. I upgraded as per your suggestion, know rocking 6.4.0 and testing it :) Thanks!

sblantipodi, 2018-12-17:
@Jason2866 Ok this morning happened again. 
Getting null readings from the sensor. No way to get back to normal readings until I removed the power to the esp. Neither a reboot of the sonoff solved the problem.

How can I do to make this considered an issue.?
If you search the net there are dozens of users with the same problem.

sblantipodi, 2018-12-17:
@ascillato2 I don't know how to solve the problem. As far as I see, it seems something related to the away mode of Home Assistant. The null readings start pretty always when I leave home and home assistant set the away mode.
Is there something that can trigger null readings from Home Assistant?

sblantipodi, 2018-12-17:
is it possible to make "a HARD reboot" via sonoff firmware?
I mean, something that remove the current to the sensor and then reapply it.
It could be useful in those cases when you want to get correct readings back and you don't have access to the esp power cables.

arendst, 2018-12-17:
If a restart of your device does not bring back the readings but the only solution is a power cycle, which in turn resets the sensor the problem is with the sensor, not with the software.

You might want to consider a decent sensor, like a I2C sensor or if you only need temperature try a DS18B20 although they are also known to be lost after some time ;-)

sblantipodi, 2018-12-17:
@arendst thanks for the support and for your excellent work here.
If you say this I will buy the I2C sensor. Now I don't have time so I will pass my holidays rebooting the esp :(

sblantipodi, 2018-12-17:
In any case there are too many singal loss happened as soon as I set away mode on.
In this holidays I will have time to investigate the problem before the switch to the I2C sensor :)
if I will find something interesting I will give some update here :)

closed 2018-12-17 by andrethomas2