arendst/Tasmota #4641: Add new template smart plug

riddik14, 2018-12-16:


riddik14, 2018-12-16:
I  have to try blitzwolf shp conf, it work with monitoring but rele go to blink. And key not work.
The plug have 2 led, 1 red and 1 blue. The blue is connect to serial the red to gpo0. I have burn flash tasmota and work with conf posted, but no monitoring

riddik14, 2018-12-16:
I have compiled fw with ide. Plug work with monitoring but Ampere are show high. Example 8w 16.6 A, why?

riddik14, 2018-12-16:
I have start calibration https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2

Powerset
Currentset
VoltageSet 
Now work good

andrethomas2, 2018-12-17:
@riddik14 Thanks for your information - happy to hear it is working. We will need to add this to the Wiki

Thank you for your contribution!

closed 2018-12-17 by andrethomas2