arendst/Tasmota #4645: Sonoff Basic reverting settings

brainthee, 2018-12-16:
**Describe the bug**
The device reverts to hard-coded settings after a crash and several reboots.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : **Sonoff Basic**
- [x] _Development/Compiler/Upload tools used_ : **Visual Studio Code & Platform IO**
- [x] _If a pre-compiled release or development binary was used, which one?_ : **Self-compiled version of release 6.3.0.11**
- [x] _You have tried latest release or development binaries?_ : **Latest as of two days ago**
- [x] _Provide the output of command_``status 0`` :
```
23:07:32 MQT: home/DVES_A86D05/STATUS = {"Status":{"Module":1,"FriendlyName":["Office Desk Lights Power"],"Topic":"DVES_A86D05","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:07:32 MQT: home/DVES_A86D05/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.XX.7:80/api/fw/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:03:30","StartupUTC":"2018-12-16T22:04:02","Sleep":0,"BootCount":17,"SaveCount":166,"SaveAddress":"F5000"}}
23:07:32 MQT: home/DVES_A86D05/STATUS2 = {"StatusFWR":{"Version":"6.3.0.11(sonoff)","BuildDateTime":"2018-12-15T15:27:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:07:32 MQT: home/DVES_A86D05/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":2,"LogHost":"192.168.XX.8","LogPort":5140,"SSId":["XXXXXX","XXXXXX"],"TelePeriod":10,"SetOption":["00028009","55818000","00000000"]}}
23:07:32 MQT: home/DVES_A86D05/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F9AE794","240383A1","23B617CE","00003BC0"]}}
23:07:32 MQT: home/DVES_A86D05/STATUS5 = {"StatusNET":{"Hostname":"DVES_A86D05-3333","IPAddress":"192.168.XX.136","Gateway":"192.168.XX.1","Subnetmask":"255.255.255.0","DNSServer":"194.168.4.100","Mac":"XXXXXX","Webserver":2,"WifiConfig":4}}
23:07:32 MQT: home/DVES_A86D05/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.XX.7","MqttPort":1883,"MqttClientMask":"DVES_A86D05","MqttClient":"DVES_A86D05","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:07:32 MQT: home/DVES_A86D05/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 16 22:07:32 2018","Local":"Sun Dec 16 23:07:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"09:09","Sunset":"16:56"}}
23:07:32 MQT: home/DVES_A86D05/STATUS10 = {"StatusSNS":{"Time":"2018-12-16T23:07:32"}}
23:07:32 MQT: home/DVES_A86D05/STATUS11 = {"StatusSTS":{"Time":"2018-12-16T23:07:32","Uptime":"0T00:03:30","Vcc":3.172,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"XXXXXX","Channel":11,"RSSI":56}}}
```

**To Reproduce**
Sonoff Basic is configured to use relay to power a 5V power supply. GPIO14 is configured as WS2812. Have ~ 70 LEDs configured.

On fresh flashing, works as expected. After about 24 hours, device appears to revert to config in config_override.h and loses it's WS2812 config. Reapplying the module configuration develops the following error in the logs after several self-reboots:

```
23:04:08 MQT: home/DVES_A86D05/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x401069a9 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```

**Expected behavior**
Not crash & maintain configuration

**Additional context**
Only been fairly recently (since 6.2 I think) that this has happened. Confirmed against several Sonoff Basic devices.

Jason2866, 2018-12-16:
No bug this is a feature. Reverts device to standards it recognizes watchdog resets and trys to
bring is stable condition to have a chance to do maintenaice.
So your compiled version has problems (faulty) or the connected stuff draws to much current

GianCann, 2018-12-17:
I too encountered a similar problem.
After some reboots due to power surges, the device returned to the 'Sonoff basic' state instead of staying on 'Sonoff Generic', losing the switch configuration.
All other configurations (MQTT, for example) have remained unchanged.
Is it possible to change the firmware (which file?) To force the default state of the configuration and of the pins?

Chip: Az-Delivery NodeMCU D1-Mini
https://www.az-delivery.de/products/nano-v3-mit-ch340-arduino-kompatibel?ls=en

Program Version	6.3.0.16(sonoff)
Build Date & Time	2018-12-07T16:30:38
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	0T01:16:10
Flash write Count	691 at 0xF5000
Boot Count	73

arendst, 2018-12-17:
The whole idea of getting a stable environment after several quick reboots relies on resetting reboot causes using a step down model. After a reboot within 10 seconds a step is taken from this process.

Disabling this feature defeats the self-restoring functionality of the device.

If you don't want it just remove lines 2658 until 2680 from file sonoff.ino.

badabing2005, 2019-06-18:
Hi,

Is it possible that with WifiConfig set to 0 and the Wifi Access Point going down, this bootloop detection would be triggered and everything reset to Sonoff Basic?

See what happened to me here:
https://groups.google.com/forum/#!topic/sonoffusers/aIeDfE7Mm_M