arendst/Tasmota #4648: Updating to 6.4.0 results in being unable to access web server of S20 units

Trippersgarage, 2018-12-17:
**Describe the bug**
_A clear and concise description of what the bug is._

**_Updating 6.4.0 to sonoff s20 results in being unable to access the sonoffs. Out of 3 S20 units, only one updated correctly, 2 are now unable to be accessed via web server_.**

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ +] _Searched the problem in issues and in the wiki_
- [ +] _Hardware used_ : **_Sonoff s20_**
- [+ ] _Development/Compiler/Upload tools used_ : **_OTA update_**
- [+ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ +] _You have tried latest release or development binaries?_ : **_Latest 6.4.0 release_** 
- [+ ] _Provide the output of command_``status 0`` :**_Unable_**
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
**_Updated to 6.4.0_**

**Expected behavior**
_A clear and concise description of what you expected to happen._
**_Expected it to update_**

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
**_Tried updating OTA my 3 sonoff s20. 1 updated without issue, 2 are now unable to be accessed. Unable to even ping the units, ping results in "host is down" error._**

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-17:
So it is not a general problem. Are the 2 others reachable via mqtt?
If yes try OTA update. Check OTA path with `OtaUrl` if not http://thehackbox.org/tasmota/release/sonoff.bin change it to this. 
After that do `Upgrade 1`

Marlor, 2018-12-17:
Had the same Problem.

2 Wemos D1 Mini v6.3.0
one update via v6.4.0 sonoff-classic.bin
one via OTA "http://thehackbox.org/tasmota/release/sonoff.bin"

both inaccessible (until reflash via serial).

Trippersgarage, 2018-12-17:
Reflashed via serial, units back up. No idea what caused the issue.

@Jason2866 Looks like it was not an isolated issue, refer to @Marlor, thou it was a different unit. Also, did not try mqtt as I do not have that setup, I'm using only the web server for my needs. And after a bunch of restarts and head banging, just went ahead to refresh via serial. Will report back if I notice any funkiness.

Naqua, 2018-12-17:
Same problem Sonoff POWR2 could not reach MQTT server at 6.4.0. Flakey webserver..
Flash back to 6.2.1 no problem.

Jason2866, 2018-12-17:
@Naqua 
Which core does your 6.2.1 use? 6.4.0 Precompiled?
If the device is upgraded over the time from older versions there are left old settings which can! make issues. You could try upgrading to 6.4.0 and do `reset 5` this erase everything and resets your device to default settings. Only wifi settings will survive, so your device connects

Marlor, 2018-12-17:
@andrethomas2 What has been answered? The issue remains.

Trippersgarage, 2018-12-22:
Update. One of the s20 has stopped connecting to my network reading it effectively useless. I’m seeing a bunch of posts by folks with similar issue and the current workaround seems to be to wipe the unit and do a wire flash with an older code base etc.

ascillato, 2018-12-22:
Please, to solve your issue read https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Trippersgarage, 2018-12-24:
I have read that many many times. Have reflashed one particular s20 according to the wiki. It’s gone beserk and at one point just kept turning on and off multiple times and seems to have fried my subwoofer. Another s20 is displaying weird behaviour where what I see on the web server does not correlate with the actual status; it’s shows the switch is off when it’s actually not. A third s20 keeps dropping connections. I’ve tried wiping the units, using the different code base as suggested etc. And one rfbridge that I cannot even access or reflash. 

I appreiacte the help and advise and this fantastic firmware is really cool. That said, this latest update seems to be seriously wonky and I’m not the only one who’s been experiencing issues. I have followed the wiki and tried the suggestions and the end result is a seemingly friend subwoofer, 3 wonky s20 units and one rfbridge that I cannot do anything about. 

Once again, I appreciate the awesome work done by Theo and the rest of the community. Perhaps I’m just too much of an idiot to figure this out myself but as it is , it seems like 6.4.0 has caused a bunch of issues for folks. 

I’m at my wits end here.

ascillato, 2018-12-24:
Please, to solve your issue try https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Trippersgarage, 2018-12-25:
Thank you for the replies. I will try to do this again and record each step as clearly as possible in hopes of finding out what exactly the issue is. Although I have suspected it’s the latest firmware update as these issues only started since then, it could possibly be something else in my setup that’s just not playing nice.

Once again thank you for the assistance and fingers crossed i can get to the bottom of this. 

Merry Christmas everyone!

Marlor, 2018-12-28:
I am now able to reproduce the error every time.
Arduino compiled Tasmota v6.3.0/v6.4.0/v6.4.1 -> OTA v6.4.0 or v6.4.1 will reset the configs and use the WifiManager:
[image]
Tomorrow im testing a fresch install of Arduino + Tasmota

Marlor, 2018-12-30:
Same thing on a new install.
Can someone else test/confirm this?

Arduino-1.8.8 + ESP Core 2.4.1 + Tasmota 6.3.0.
Only change SSID/PASS and Upload.
OTA update via webinterface and the configs will reset.

Jason2866, 2018-12-30:
You will get no reliable setup with core 2.4.1
Core 2.4.0 and 2.4.1 is known to have issues!
Use Core 2.3.0, 2.4.2 or 2.5.0 beta2

Marlor, 2018-12-30:
Sorry misswrote, i did use 2.4.2.
And just now tested 2.3.0, same result.