arendst/Tasmota #4652: Tasmota LWT status - Upper/Lower case letters - 'Online' in 6.4.0 - 'online' versions before

h4nc, 2018-12-17:
It looks like in the newest Version of tasmota 6.4.0 the LWT status is reported as 'Online' and 'Offline' instead of 'online' and 'offline'.
I don't really know if its a tasmota or a home assistant issue. But I think I should try it here first.

It causes problems with entities in home assistant.
Things like the `availability_topic` of mqtt-lights will be unavailable and there is no option for that to change it to 'Online' in HA. 
[](https://www.home-assistant.io/components/light.mqtt/)
It seems like for now there is no way but to uncomment this line to make the light work again (of course coudl also go back to 6.3.0).

digiblur, 2018-12-17:
I checked a couple of my previous configs in and I've always had them as 'Online' and 'Offline'

h4nc, 2018-12-17:
Thanks for checking your configs.

That's weird. Maybe I also always had 'Online'. The strange thing is that my LWT automations stopped working and I had to change them to 'Online'. They worked before the Sonoff-Upgrades.

h4nc, 2018-12-17:
Ok so I downgraded one of my sonoffs to 6.3.0 (both 6.3.0 and 6.4.0 from github, I didn't change something) 
In my screenshots you can see that there is indeed a upper/lower case letter difference for both versions. And because availability-topic only listens to online and offline (lower case) this will fail.

[image]
[image]

arendst, 2018-12-17:
Let me guess, Germany?

https://github.com/arendst/Sonoff-Tasmota/pull/4392

h4nc, 2018-12-17:
Yes, it's the german file, thanks.

arendst, 2018-12-17:
But this is the main reason why it has to stay as it is: https://github.com/arendst/Sonoff-Tasmota/issues/4388

h4nc, 2018-12-17:
Ok, I get that all languages should use 'Online' consistant. It's fine to change it.

So it seems like now it gets a home assistant issue? But if it was always Online for all the other languages, how did you use the 'availability_topic' in ha? As I said before it seems like it only listens to lower case online's ;-)

From here: https://www.home-assistant.io/components/light.mqtt/

> availability_topic
> (string)(Optional)The MQTT topic subscribed to receive availability (online/offline) updates.

I think sonoff devices are one of the most used mqtt devices in home assistant, so it wouldn't make sense to have those differences.

arendst, 2018-12-17:
I don't think Germany is the only country using HASS. The others never complaint so I guess there must be a simple HASS solution...

h4nc, 2018-12-17:
Yes, I also think so. I will take this to the ha community. Let's see what they say. Could also be possible that most guys don't use the availability_topic, as this isn't mandatory.

andrethomas2, 2018-12-17:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-17 by andrethomas2

h4nc, 2018-12-17:
Ok, so the hint from @digiblur was the problem solver. 

Had to add those lines:

```
    payload_available: "Online"
    payload_not_available: "Offline"
```
The default values are lower case.