arendst/Tasmota #4655: Extended logging to troubleshoot WiFi

pkaras3, 2018-12-17:
Hi,
I have some issues with one of the Sonoffs Basic and Sonoff Touch. They randomly disconnect from WiFi and sometimes it takes them hours to connect back and then it goes fine for couple of days.
Unfortunately, even with logging set to "More debug" the only debug info I get is as follows:
Dec 15 12:21:10 sonoff-wardrobe-vent-3971 ESP-WIF: Checking connection...
Dec 15 12:21:10 sonoff-wardrobe-vent-3971 ESP-WIF: Connected
Dec 15 12:21:32 sonoff-wardrobe-vent-3971 ESP-MQT: Attempting connection...
Dec 15 12:21:38 sonoff-wardrobe-vent-3971 ESP-MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
Dec 15 12:21:42 sonoff-wardrobe-vent-3971 ESP-WIF: Checking connection...
Dec 15 12:21:42 sonoff-wardrobe-vent-3971 ESP-WIF: Connected

On the other end I see in the Mosquito logs, that the session is dropped 

Can you extend the logging capabilities to troubleshoot the WiFi issues? Like: is this TCP/IP or physical layer, the Wifi signal strength during the issue, how is the network stack doing etc.

Regards,
Piotr

Jason2866, 2018-12-17:
Issue template not filled out. Without no help can be provided and issue will be closed

meingraham, 2018-12-17:
What Jason said... but have you tried level 4 logging (e.g., WebLog 4)?

pkaras3, 2018-12-17:
Hi Jason, this is more enchancement thank issue.

Syslog level is set to 4 already. This is what I get from it.

Jason2866, 2018-12-17:
ESP is a very limited in resources. Nothing will be implemented what is not a feature or needed.
This debugging feature will not be implemented because there is no known problem with wifi.
We can maybe help if you fill out completly the issue template.
It is a Open Source project you are invited to write the code you want and asked for.

ascillato2, 2018-12-17:
Hi,

In order to properly help you, please complete the troubleshooting template so as to have more information. Remember that you can also find us in the Tasmota Chat. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-17 by ascillato2