arendst/Tasmota #4659: Esp8266 core for Arduino version 2.3.0 and WifiManager - wdt reset load 0x4010f000, len 1384, room 16

araczkowski, 2018-12-17:
**Describe the bug**
platform = espressif8266@1.5.0 and WifiManager (by pressing button 4 times on Sonoff S20 or S26)
give me always an error:
```
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3df0
~ld
```
_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X] _Searched the problem in issues and in the wiki_
- [ X] _Hardware used_ : S20 or S26
- [ X] _Development/Compiler/Upload tools used_ : PlatformIO
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [X ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
19:54:13 CMD: status 0
19:54:13 RSL: gniazdo_1/stat/STATUS = {"Status":{"Module":8,"FriendlyName":["AIS dom iot"],"Topic":"gniazdo_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:54:13 RSL: gniazdo_1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"dom","OtaUrl":"http://ai-speaker.com/ota/dom/S20.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:08:15","StartupUTC":"2018-12-17T18:45:58","Sleep":0,"BootCount":7899,"SaveCount":15817,"SaveAddress":"F4000"}}
19:54:13 RSL: gniazdo_1/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0.1(sonoff)","BuildDateTime":"2018-12-17T19:38:17","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:54:13 RSL: gniazdo_1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"dom","LogPort":514,"SSId":["ais-dom","AI-SPEAKER"],"TelePeriod":300,"SetOption":["0A0A0009","558180C0","00000201"]}}
19:54:13 RSL: gniazdo_1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000415","1FDAE794","240383A0","23B617CE","00003BC0"]}}
19:54:13 RSL: gniazdo_1/stat/STATUS5 = {"StatusNET":{"Hostname":"dom_SONOFF_S20_3FFF2CF5-3865","IPAddress":"192.168.1.91","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:10:0F:19","Webserver":2,"WifiConfig":2}}
19:54:13 RSL: gniazdo_1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.222","MqttPort":1883,"MqttClientMask":"ais_iot_%06X","MqttClient":"ais_iot_100F19","MqttUser":"AIS_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:54:13 RSL: gniazdo_1/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 17 18:54:13 2018","Local":"Mon Dec 17 19:54:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:54"}}
19:54:13 RSL: gniazdo_1/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-17T19:54:13"}}
19:54:13 RSL: gniazdo_1/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-17T19:54:13","Uptime":"0T00:08:15","Vcc":3.177,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ais-dom","BSSId":"B4:0F:3B:44:83:F1","Channel":6,"RSSI":100}}}
```

**To Reproduce**
1. select platform = espressif8266@1.5.0
2. select WIFI_MANAGER as a default tool
#define WIFI_CONFIG_TOOL       WIFI_MANAGER
#define WIFI_CONFIG_NO_SSID    WIFI_MANAGER  
3. build
4. upload
5. monitor
press the button 4 times shortly logs you will see:

```
19:43:03 RSL: gniazdo_1/stat/RESULT = {"WifiConfig":"WifiManager selected"}
19:43:03 WIF: WifiManager aktywny 3 minuty
19:43:03 HTP: Serwer Web zatrzymany

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3df0
~ld19:54:13 CMD: status 0

```

**Expected behavior**
be able to use WifiManager with Arduino version 2.3.0 (stable WiFi)

**Screenshots**
[image]

**Additional context**
I think it old bug probably in Esp8266 core for Arduino version 2.3.0
the WifiManager works correct in Esp8266 core for Arduino version 2.4.2 but unfortunately the the WiFi there are known problems with WiFi stability on other versions of the  Esp8266 core for Arduino then 2.3.0

**Is ti possible to have booth WifiManager and  Esp8266 core for Arduino version 2.3.0?**

Thanks in advance 
Andrzej

andrethomas, 2018-12-17:
Binaries are built for both 2.3.0 and 2.4.2 and available for OTA.

The default ones at http://thehackbox.org/tasmota/release/ are 2.4.2 but you can download the 2.3.0 core based binaries for OTA from http://thehackbox.org/tasmota/release/020300/ - These are kept for posterity purposes only as we envisage to move entirely to 2.4.2 and eventually 2.5.0 if it proves to remain stable as it is now.

Jason2866, 2018-12-17:
Tested. Wifimanager works with core 2.3.0.
You can dl precompiled version with core 2.3.0 from here http://thehackbox.org/tasmota/release/020300/

araczkowski, 2018-12-17:
Thanks, let me check this... I will be back with feedback

araczkowski, 2018-12-17:
unfortunately no luck, what I did:
1. downloaded precompiled version with core 2.3.0 from here http://thehackbox.org/tasmota/release/020300/sonoff-PL.bin

2. esptool erase
```
andrzej@ultra:~/Projects$ esptool.py --port /dev/ttyUSB0 erase_region 0x0F4000 0x008000
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.7 seconds.
Hard resetting via RTS pin...
andrzej@ultra:~/Projects$ esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.2s
Hard resetting via RTS pin...
```

3. esptool flash
```
andrzej@ultra:~/Projects$ esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff-PL.bin 
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 517280 bytes to 353422...
Wrote 517280 bytes (353422 compressed) at 0x00000000 in 31.7 seconds (effective 130.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

4. terminal
```
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Projekt sonoff Sonoff (Topic sonoff, Awaryjny DVES_100F19, GroupTopic sonoffs) Wersja 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager aktywny 3 minuty

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld
```

@Jason2866 did you really tested this ?

PS
I think I'm not the only one, according to this, see HomeOps blog post
https://lobradov.github.io/FOSS-Firmware-comparison-configuration/
> I was never able to put Tasmota into “Admin AP” mode

Jason2866, 2018-12-17:
Try sonoff.bin not localized

araczkowski, 2018-12-17:
ok checking...

araczkowski, 2018-12-17:
I did all the same: download (this time not localized version), erase, flash 
and still the same issue:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_100F19, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_100F19, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_100F19, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-3865 with IP address 192.168.4.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld
```

of course I have the same on each S20 and S26 EU
and when I change the Esp8266 core for Arduino to version 2.4.2 the WifiManager works fine so it is not hardware issue 

@Jason2866  are you checking this on S20/S26?
Andy other ideas ?

Jason2866, 2018-12-17:
No checked it on a wemos mini with connected button...
Yes send wifi credentials via backlog command over serial

Jason2866, 2018-12-17:
No because i cant reproduce your issue. Why not give 2.4.2 a try? 
We have no feedback from Tasmota users with problems.
It frees ram and software serial is working better.

araczkowski, 2018-12-17:
I did try it some times ago and it was disaster - I rollback to 2.3.0
and I did it today again, I have 30 devices in my office and about 30 in my house.
(sockets, touches, basics, th with sensors, duals...)
First I thought that all is going well but about 1h hour after update I don't have connection with 30% of the devices and I'm trying to find a way to roll back again to 2.3.0
Other devices are appearing and disappearing periodically in my system...
And I have WiFI mesh in both locations are the devices has good WiFI connection.

When you are working on one device it can looks OK but in the real home automation system it is not.

> We have no feedback from Tasmota users with problems.
I don't think it is full true, I'm not the first time here and I saw a lot of feedback about WiFi stability problems on core above  2.3.0  but you are closing everything so fast without solution... 

But let's go back to the point .
I will be really happy to use 2.3.0 with WifiManager I will be glad for any help to achieve this.

Thanks

Jason2866, 2018-12-17:
I have around 60 devices about half a year rock solid running with 2.4.2
2.4.0 and 2.4.1 was bad...
So i know what i am recommend. But i dont use Mesh. My AP are all wired via switched managed
network and controlled via a Hardware Wlan controller

andrethomas, 2018-12-17:
The wifi stability issues you are referring to was related to bad thermal design on the sonoff basic and was mitigated by new process management from Tasmota 6.3.10 and higher.

There really is no reason not to run on 2.4.2 core, or 2.5.0 for that matter at this point.

araczkowski, 2018-12-17:
It's not that I didn't try... 
but anyway, do you have any idea how to solve the WifiManager problem on 2.3.0?

andrethomas, 2018-12-17:
I'm not sure what you mean, it works perfectly.

On boot I get this on the serial console
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10373D, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-5949 with IP address 192.168.4.1
```

Then, on my phone I scan for wifi and see the sonoff-xxx AP

[image]

After connecting to it I am presented with the captive portal to enter wifi credentials

[image]

After entering details and click save the following is produced on serial log

```
00:01:35 WIF: Hostname %s-%04d, SSId1 Wireless, SSId2 
00:01:36 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_10373D, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-5949...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-5949.local with IP address 192.168.42.123
```

This test was done on a Sonoff Basic.

So sad to say, but the problem is on your side and is not a firmware issue.

andrethomas2, 2018-12-17:
Closing this issue as it has been answered and does not appear to be a firmware problem, but a user configuration/setup problem.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-17 by andrethomas2