arendst/Tasmota #4662: pzem don't work 6.4.0

riddik14, 2018-12-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ : 
- [ x] _Development/Compiler/Upload tools used_ : release bin
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ x] _You have tried latest release or development binaries?_ : yes, 6.3.0 work good
- [x ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
/STATUS = {"Status":{"Module":1,"FriendlyName":["Ripostiglio"],"Topic":"luceripostiglio","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

pzem-004t don't receive data

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
ok

arendst, 2018-12-17:
Read the RELEASENOTES:
 * Fix hardware serial pin configuration. To keep using hardware serial swap current Rx/Tx pin configuration only (#4280)

riddik14, 2018-12-17:
21:41:23 MQT: stat/luceripostiglio/STATUS = {"Status":{"Module":1,"FriendlyName":["Ripostiglio"],"Topic":"luceripostiglio","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:41:23 MQT: stat/luceripostiglio/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.247:9999/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:52:20","StartupUTC":"2018-12-17T19:49:03","Sleep":50,"BootCount":360,"SaveCount":1842,"SaveAddress":"FA000"}}
21:41:23 MQT: stat/luceripostiglio/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:43:25","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:41:23 MQT: stat/luceripostiglio/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FASTWEB-W8NHB5",""],"TelePeriod":80,"SetOption":["0002A809","558180C0","00000029"]}}
21:41:23 MQT: stat/luceripostiglio/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000410","0FDAE794","000003A0","23B617CE","00000000"]}}
21:41:23 MQT: stat/luceripostiglio/STATUS5 = {"StatusNET":{"Hostname":"luceripostiglio-6158","IPAddress":"192.168.1.212","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:A3:18:0E","Webserver":2,"WifiConfig":5}}
21:41:23 MQT: stat/luceripostiglio/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.247","MqttPort":1883,"MqttClientMask":"Ripostiglio_%06X","MqttClient":"Ripostiglio_A3180E","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:41:23 MQT: stat/luceripostiglio/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 17 20:41:23 2018","Local":"Mon Dec 17 21:41:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:41","Sunset":"16:21"}}
21:41:23 MQT: stat/luceripostiglio/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:41:23 MQT: stat/luceripostiglio/STATUS10 = {"StatusSNS":{"Time":"2018-12-17T21:41:23","ENERGY":{"Total":199.641,"Yesterday":12.010,"Today":13.860,"Power":855,"Factor":0.91,"Voltage":233,"Current":4.020}}}
21:41:23 MQT: stat/luceripostiglio/STATUS11 = {"StatusSTS":{"Time":"2018-12-17T21:41:23","Uptime":"0T00:52:20","Vcc":3.192,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FASTWEB-W8NHB5","RSSI":74,"APMac":"B0:EA:BC:9A:5C:70"}}}

arendst, 2018-12-17:
Status is from 6.2.1 on core 2.3.0 !!!!?????

Show your GPIO config and tell what kind of PZEM you have.

arendst, 2018-12-17:
Works like a treat:

[image]

[image]

riddik14, 2018-12-17:
fw 6.4.0.1?

arendst, 2018-12-17:
Don't mind. Pls provide requested info.

riddik14, 2018-12-17:
you use fw 6.4.0.1, you have compiled. i have used official release 6.4.0. i try to flash with ide. thanks u

arendst, 2018-12-17:
Again, doesn't matter. PZEM has always worked. In 6.4.0 the GPIO's have changed.

That's why I asked you to provide your GPIO config and the type of PZEM you have.

andrethomas, 2018-12-17:
6.4.0.1is also official.

Please provide the requested information.

Thanks

riddik14, 2018-12-17:
ok,work, but stay quite. thank u

closed 2018-12-17 by riddik14

willyk0, 2018-12-31:
Similar problem here with 6.4.1 (official bin "sonoff" with core 2.4.2). I swapped the configuration as arendst said (Tx with Tx and Rx with Rx) and there is still no communication. If I switch hardware pins (TX->TX, RX-RX) it works, but its software serial which causes lots of exceptions... 

What I'm trying with my Sonoff Basic

GPIO1 Serial Out	PZEM0xx TX (62)
GPIO3 Serial In	PZEM004 RX (63)

Physycally, the pins are swapped (Sonoff RX -> PZEM TX, Sonoff TX -> PZEM RX).

So, to summarize:

Hardware pins "straight" -> Works with software serial configured with GPIO1 -> RX and GPIO3 -> TX, about 10 reboots/day (reason: exceptions).
Hardware pins "crossed" (to use hardware serial) -> Doesn't work with any combination...

I'm getting mad with this, can someone throw any light? 

Thanks in advance, and happy 2019 :)