arendst/Tasmota #4664: SetOption19 causes "RestartReason":"Software Watchdog" on Sonoff POW

jpalczewski, 2018-12-17:
**Describe the bug**
Today I upgraded a sonoff pow from 5.12 to 6.4 using a pre-built migration path(5.12->5.14->6.4). I have tried to enable homeassistant autodiscovery feature, which works(I have a fresh sonoff touch with self-compiled 6.4, which was sucessfully discovered and works)*. When I tried to `SetOption19 1`, a restart appears. 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff POW. 
- [x] _Development/Compiler/Upload tools used_ : upload via web page. 
- [x] _If a pre-compiled release or development binary was used, which one?_ : Latest tagged release, v6.4.0
- [x] _You have tried latest release or development binaries?_ : release.
- [x] _Provide the output of command_``status 0`` :
```
CMD: status 0
23:48:11 MQT: stat/multimedia-listwa/STATUS = {"Status":{"Module":6,"FriendlyName":["multimedia-listwa"],"Topic":"multimedia-listwa","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:48:11 MQT: stat/multimedia-listwa/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software Watchdog","Uptime":"0T00:08:46","StartupUTC":"2018-12-17T22:39:25","Sleep":50,"BootCount":99,"SaveCount":228,"SaveAddress":"F8000"}}
23:48:11 MQT: stat/multimedia-listwa/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:48:11 MQT: stat/multimedia-listwa/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"10.137.0.2","LogPort":514,"SSId":["anonimized","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","000180C0","00000021"]}}
23:48:11 MQT: stat/multimedia-listwa/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
23:48:11 MQT: stat/multimedia-listwa/STATUS5 = {"StatusNET":{"Hostname":"multimedia-listwa-6069","IPAddress":"10.137.0.117","Gateway":"10.137.0.1","Subnetmask":"255.255.0.0","DNSServer":"10.137.0.1","Mac":"2C:3A:E8:07:57:B5","Webserver":2,"WifiConfig":2}}
23:48:11 MQT: stat/multimedia-listwa/STATUS6 = {"StatusMQT":{"MqttHost":"10.137.0.2","MqttPort":1883,"MqttClientMask":"multimedia-listwa","MqttClient":"multimedia-listwa","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:48:11 MQT: stat/multimedia-listwa/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 17 22:48:11 2018","Local":"Mon Dec 17 23:48:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:54"}}
23:48:11 MQT: stat/multimedia-listwa/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:48:11 MQT: stat/multimedia-listwa/STATUS10 = {"StatusSNS":{"Time":"2018-12-17T23:48:11","ENERGY":{"TotalStartTime":"2018-12-17T23:28:11","Total":0,"Yesterday":0,"Today":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
23:48:11 MQT: stat/multimedia-listwa/STATUS11 = {"StatusSTS":{"Time":"2018-12-17T23:48:11","Uptime":"0T00:08:46","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"anonymized","BSSId":"anonymized","Channel":11,"RSSI":100}}}```

**To Reproduce**
1. input `SetOption19 1` into web console

**Expected behavior**
Wall of text and discoverable switch/sensor data in HA.

**Screenshots**
n/a

**Additional context**
I'm using a dockerized mosquitto/HA(recent, all of those were pulled today) with mqtt user/pass shared with sonoff touch. While configuring the other switch, I addressed something weird - manually configured didn't work, i.e.: a switch in HA caused a proper mqtt message, but there was something bad with LWT - HA didn't get any respond, so it thought that the message wasn't delivered(when irl it was) and reverted a switch internally, making HA unusable. When I tried a dynamic disovery - everything worked flawlessly. 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-18:
Maybe old settings which causes this issues. 
Try Reset 5 this commands resets device and erases all unused flash + settings area.
Only wifi credentials will survive. After that the devices reboots, and needs reconfiguring.
So it is a good idea to write down the calibration values.

ascillato2, 2018-12-18:
Hi

Sorry, I can't reproduce your issue. Please do as Jason told you or erase all flash with esptool.py as explained in the wiki and flash again from the precompiled bins in order to solve your issue. 

Thanks.

jjlawren, 2018-12-18:
My POW has the same behavior. I haven't used discovery on this device previously, but for context I recently upgraded it from 6.1.0 (self-built) to 6.4.0 (http://thehackbox.org/tasmota/sonoff.bin).

I also issued a `reset 5` and set it up from scratch. Still hangs and resets when issuing `setoption19 1` via the console. No discovery messages are published before restarting.

`status 0` for the device:
```
20:06:06 CMD: status 0
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff POW (Dryer)"],"Topic":"sonoff_POW_dryer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:03:11","StartupUTC":"2018-12-18T19:02:55","Sleep":50,"BootCount":5,"SaveCount":15,"SaveAddress":"F5000"}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["isomesh","indebuurt2"],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS5 = {"StatusNET":{"Hostname":"sonoff_POW_dryer-2584","IPAddress":"10.99.0.11","Gateway":"10.99.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.99.0.1","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS6 = {"StatusMQT":{"MqttHost":"10.99.0.55","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XXXXXX","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 19:06:06 2018","Local":"Tue Dec 18 20:06:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":1,"PowerHigh":1,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T20:06:06","ENERGY":{"TotalStartTime":"2018-12-18T19:58:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":117,"Current":0.000}}}
20:06:06 MQT: stat/sonoff_POW_dryer/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T20:06:06","Uptime":"0T00:03:11","Vcc":3.538,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"ssid","BSSId":"XX:XX:XX:XX:XX:XX","Channel":1,"RSSI":50}}}
```

Jason2866, 2018-12-18:
Do you have in router Auto channel select? 
Is your wifi a Mesh or Repeater network?

jpalczewski, 2018-12-18:
@Jason2866 so, `reset 5` didn't worked for me, I would say that webserver stability decreased(a lot of connection errors&sonoff didn't respond to ping), restarting by cutting power off maybe stabilized situation a bit, but situation from issue haven't been fixed. 6.4.0.1 didn't do the trick either, `status 0` after restart:
```
21:50:07 MQT: stat/multimedia-listwa/STATUS = {"Status":{"Module":6,"FriendlyName":["multimedia-listwa"],"Topic":"multimedia-listwa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:50:07 MQT: stat/multimedia-listwa/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:06:06","StartupUTC":"2018-12-18T20:44:01","Sleep":50,"BootCount":9,"SaveCount":30,"SaveAddress":"F9000"}}
21:50:07 MQT: stat/multimedia-listwa/STATUS2 = {"StatusFWR":{"Version":"6.4.0.1(1eb7f8e-sonoff)","BuildDateTime":"2018-12-18T12:01:46","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
21:50:07 MQT: stat/multimedia-listwa/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ssid","indebuurt2"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
21:50:07 MQT: stat/multimedia-listwa/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
21:50:07 MQT: stat/multimedia-listwa/STATUS5 = {"StatusNET":{"Hostname":"multimedia-listwa-6069","IPAddress":"10.137.0.117","Gateway":"10.137.0.1","Subnetmask":"255.255.0.0","DNSServer":"10.137.0.1","Mac":"2C:3A:E8:07:57:B5","Webserver":2,"WifiConfig":4}}
21:50:07 MQT: stat/multimedia-listwa/STATUS6 = {"StatusMQT":{"MqttHost":"10.137.0.2","MqttPort":1883,"MqttClientMask":"multimedia-listwa","MqttClient":"multimedia-listwa","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:50:07 MQT: stat/multimedia-listwa/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 20:50:07 2018","Local":"Tue Dec 18 21:50:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
21:50:07 MQT: stat/multimedia-listwa/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:50:07 MQT: stat/multimedia-listwa/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T21:50:07","ENERGY":{"TotalStartTime":"2018-12-18T20:54:48","Total":0.588,"Yesterday":0.000,"Today":0.588,"Power":24,"ApparentPower":29,"ReactivePower":16,"Factor":0.83,"Voltage":231,"Current":0.126}}}
21:50:07 MQT: stat/multimedia-listwa/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T21:50:07","Uptime":"0T00:06:06","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"ssid","BSSId":"bssid","Channel":11,"RSSI":100}}}
```

Tomorrow I'll bring that pow out from environment and flash it by serial, if it wouldn't work I would go into downgrading. 

Maybe it might be useful:  I have also a graylog instance, so there's dump from syslog:
```
"timestamp","source","message"
"2018-12-18T20:55:58.663Z","10.137.0.117","multimedia-listwa-6069 ESP-APP: (UTC) Tue Dec 18 20:56:12 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018"
"2018-12-18T20:55:59.669Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: Attempting connection..."
"2018-12-18T20:55:59.687Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: Subscribe to cmnd/multimedia-listwa/#"
"2018-12-18T20:55:59.689Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: Subscribe to cmnd/sonoffs/#"
"2018-12-18T20:55:59.714Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: tele/multimedia-listwa/INFO3 = {""RestartReason"":""Software Watchdog""}"
"2018-12-18T20:55:59.723Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: stat/multimedia-listwa/POWER = ON"
"2018-12-18T20:55:46.504Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: homeassistant/light/0757B5_LI_3/config =  (retained)"
"2018-12-18T20:55:46.507Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: homeassistant/switch/0757B5_RL_3/config =  (retained)"
"2018-12-18T20:55:46.481Z","10.137.0.117","multimedia-listwa-6069 ESP-CMD: SetOption19 1"
"2018-12-18T20:55:46.499Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: homeassistant/light/0757B5_LI_2/config =  (retained)"
"2018-12-18T20:55:46.510Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: homeassistant/light/0757B5_LI_4/config =  (retained)"
"2018-12-18T20:55:46.512Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: homeassistant/switch/0757B5_RL_4/config =  (retained)"
"2018-12-18T20:55:58.427Z","10.137.0.117","multimedia-listwa-6069 ESP-DNS: Initialized"
"2018-12-18T20:55:59.684Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: tele/multimedia-listwa/LWT = Online (retained)"
"2018-12-18T20:55:59.707Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: tele/multimedia-listwa/INFO1 = {""Module"":""Sonoff Pow"",""Version"":""6.4.0.1(1eb7f8e-sonoff)"",""FallbackTopic"":""multimedia-listwa"",""GroupTopic"":""sonoffs""}"
"2018-12-18T20:55:59.686Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: cmnd/multimedia-listwa/POWER ="
[... cut ...]
"2018-12-18T20:55:59.679Z","10.137.0.117","multimedia-listwa-6069 ESP-MQT: Connected"
```
It's unsorted, but there's a sign that ESP-MQT via "debug more" can arrive to another host & don't show in web console. I haven't watched mqtt messages directly, but in my case HA catches something and `sensor.multimedialistwa_energy_totalst` with "unavailable" state shows up. 

My WiFi env: WRT1200AC with OpenWRT & channel 11 selected manually, I also have a repeater on network(xiaomi) in roaming mode(sonoff touch works via it flawlessly).

Jason2866, 2018-12-18:
Lets see if a erase flash with esptool and flashing afterwards tasmota solves it.
If not could try to switch off temporary the xiaomi repeater?

jpalczewski, 2018-12-19:
And the winner is @ascillato2 - after erasing flash and uploading 6.4.0.1 `SetOption19 1` didn't cause a software reset. 

`status 0` for the record: 
```
19:26:03 MQT: multimedia-listwa/stat/STATUS = {"Status":{"Module":6,"FriendlyName":["multimedia-listwa"],"Topic":"multimedia-listwa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:26:03 MQT: multimedia-listwa/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:50","StartupUTC":"2018-12-19T18:20:13","Sleep":50,"BootCount":7,"SaveCount":21,"SaveAddress":"F7000"}}
19:26:03 MQT: multimedia-listwa/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0.1(8a7dd3e-sonoff)","BuildDateTime":"2018-12-19T01:01:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:26:03 MQT: multimedia-listwa/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"10.137.0.2","LogPort":514,"SSId":["ssid",""],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000200"]}}
19:26:03 MQT: multimedia-listwa/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:26:03 MQT: multimedia-listwa/stat/STATUS5 = {"StatusNET":{"Hostname":"multimedia-listwa","IPAddress":"10.137.0.117","Gateway":"10.137.0.1","Subnetmask":"255.255.0.0","DNSServer":"10.137.0.1","Mac":"2C:3A:E8:07:57:B5","Webserver":2,"WifiConfig":4}}
19:26:03 MQT: multimedia-listwa/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.137.0.2","MqttPort":1883,"MqttClientMask":"multimedia-listwa","MqttClient":"multimedia-listwa","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:26:03 MQT: multimedia-listwa/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 19 18:26:03 2018","Local":"Wed Dec 19 19:26:03 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
19:26:03 MQT: multimedia-listwa/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:26:03 MQT: multimedia-listwa/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-19T19:26:03","ENERGY":{"TotalStartTime":"2018-12-19T19:13:12","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":37,"ApparentPower":43,"ReactivePower":23,"Factor":0.84,"Voltage":231,"Current":0.188}}}
19:26:03 MQT: multimedia-listwa/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-19T19:26:03","Uptime":"0T00:05:50","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"ssid","BSSId":"bssid","Channel":11,"RSSI":100}}}
```
Also for the record - my sonoff is about a two-three years old, as far as I remember I followed the migration path from x.x(very old one, can't remember which one) to 5.12, and now from 5.12 to 6.40, but maybe that was some edge case of migrating. 

@Jason2866 I doubt that the repeater might be the cause, pow is located about 50cm from the router, and possibly helps a lot for sonoff touch which is located far away from router. 

Last thought for the record: While autodiscovery helps for mqtt stability in sonoff touch case, it don't help in sonoff pow case - same issue when I have , HA sends mqtt message for shutting down and can't receive/decode respond, so it switches state back internally(so I can turn off my sonoff when it is turned off, great), but on the other side dynamic discovery at least can parse pow stats in good manner. It might be not tasmota specifically issue(I need to investigate it further someday). 

Many thanks for all of you! Great project and great community.

closed 2018-12-19 by jpalczewski