arendst/Tasmota #4665: Shelly 2 Automatically turns ON/OFF  relay 2 with external switch connected

fsilva80, 2018-12-18:
Hello!
I have a Shelly2 with problems only on the second relay.
Here's my wiring diagram:
[image]

I already tried switching both wires from the switch and also for the lamps.
The problem is only on the second relay...

I also had this message when it appens:
SRC: Switch

But, as i said i switched the wiring and the problem stays in relay 2.

I'm using the latest Sonoff-Tasmota 6.4.0!

Here's the status 0:

```
03:33:40 CMD: STATUS 0
03:33:41 MQT: stat/shelly2_01/STATUS = {"Status":{"Module":47,"FriendlyName":["Interruptor entrada da sala","Interruptor fundo da sala"],"Topic":"shelly2_01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
03:33:41 MQT: stat/shelly2_01/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T07:08:37","StartupUTC":"2018-12-17T19:25:04","Sleep":50,"BootCount":10,"SaveCount":120,"SaveAddress":"F4000"}}
03:33:41 MQT: stat/shelly2_01/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
03:33:41 MQT: stat/shelly2_01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Lost Soul",""],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
03:33:41 MQT: stat/shelly2_01/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
03:33:41 MQT: stat/shelly2_01/STATUS5 = {"StatusNET":{"Hostname":"shelly2_01-3091","IPAddress":"192.168.1.72","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"CC:50:E3:13:4C:13","Webserver":2,"WifiConfig":4}}
03:33:41 MQT: stat/shelly2_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_134C13","MqttUser":"fsilva80","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:33:41 MQT: stat/shelly2_01/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 02:33:41 2018","Local":"Tue Dec 18 03:33:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
03:33:41 MQT: stat/shelly2_01/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:33:41 MQT: stat/shelly2_01/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T03:33:41","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-12-17T01:37:55","Total":0.015,"Yesterday":0.011,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
03:33:41 MQT: stat/shelly2_01/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T03:33:41","Uptime":"0T07:08:37","Vcc":3.556,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Lost Soul","BSSId":"00:26:44:38:86:5A","Channel":1,"RSSI":80}}}
```

At this moment there's nothing attached to relay 2.
And relay 2 stays off, no ghost switching...

Jason2866, 2018-12-18:
So at the moment no problem? Or do i missunderstand you last sentence wrong?
If there is no ghost switching if nothing connected than it has to be the external wiring.

fsilva80, 2018-12-18:
The wires that go from the switch to s1 and #2 are the same length.
As i said i already tried switching the wires and the problem remains in the second relay. It thasn't matter if the wires that come from the switch and the ones that go to the lights are on s1/o1 or s2/o2 the problem is always in relay 2/power2.

andrethomas, 2018-12-18:
Must be a hardware issue then.

Please rule out firmware issue by erasing flash and re-flashing 6.4.0 from scratch.

Thanks

itMax86, 2018-12-18:
I have similar problem with my sonoff 4ch pro r2, with 6.3 all work fine, after update I see a lot of ghost power on\off and frequent freeze (fisical button don't work, mqtt commands don't work, internal page sonoff's commands don't work) you must turn off/on your sonoff

how to downgrade FW without use serial programmer

Jason2866, 2018-12-18:
@itMax86 the same way you did the upgrade.
Did you ever erased the whole flash before you flashed Tasmota?
This can make this issues since this version is bigger and uses different flash adresses
Link for OTA version 6.3. http://thehackbox.org/tasmota/release/archive/6.3.0/sonoff.bin

pozi78, 2018-12-18:
Same problem here, upgraded to tasmota 6.4.0 and after some minutes first ghost switching
here is my syslog:

2018-12-18,21:58:59,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 CF 07 00 00 E1 08 88 0F 00 00 90 07 00 00 A9 11 00 00 6F 8F F4 C6 7E
2018-12-18,21:58:58,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 C8 07 00 00 E1 08 86 0F 00 00 97 07 00 00 A9 11 00 00 7D 8F C8 C6 5E
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 83 06 00 00 E1 08 21 0C 00 00 16 07 00 00 97 0E 00 00 98 95 EC C6 5F
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-MQT: home/shelly_lights/POWER1 = ON (retained)
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,"ESP-MQT: home/shelly_lights/RESULT = {""POWER1"":""ON""}"
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SRC: Switch
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-MQT: home/shelly_lights/POWER1 = OFF (retained)
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,"ESP-MQT: home/shelly_lights/RESULT = {""POWER1"":""OFF""}"
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SRC: Switch
2018-12-18,21:58:56,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 D0 07 00 00 E1 08 88 0F 00 00 93 07 00 00 A9 11 00 00 6F 8F BA C6 48
2018-12-18,21:58:55,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 D0 07 00 00 E0 08 8C 0F 00 00 9B 07 00 00 A9 11 00 00 52 8F AE C6 2A
2018-12-18,21:58:54,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 C8 07 00 00 E1 08 74 0F 00 00 88 07 00 00 A9 11 00 00 00 90 E0 C6 D9

and device status:

22:23:18 CMD: status 0
22:23:18 SRC: WebConsole from 172.17.1.1
22:23:18 RSL: Group 0, Index 1, Command STATUS, Data 0
22:23:18 MQT: home/shelly_lights/STATUS = {"Status":{"Module":47,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"shelly_lights","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:23:18 MQT: home/shelly_lights/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:27:01","StartupUTC":"2018-12-18T20:56:17","Sleep":50,"BootCount":5,"SaveCount":29,"SaveAddress":"F7000"}}
22:23:18 MQT: home/shelly_lights/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:23:18 MQT: home/shelly_lights/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":3,"LogHost":"172.17.0.2","LogPort":514,"SSId":["home_iot",""],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
22:23:18 MQT: home/shelly_lights/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:23:18 MQT: home/shelly_lights/STATUS5 = {"StatusNET":{"Hostname":"shelly_lights-0278","IPAddress":"172.17.1.28","Gateway":"172.17.0.1","Subnetmask":"255.255.0.0","DNSServer":"8.8.8.8","Mac":"CC:50:E3:55:A1:16","Webserver":2,"WifiConfig":4}}
22:23:18 MQT: home/shelly_lights/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.0.11","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_55A116","MqttUser":"mqtt_user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:23:18 MQT: home/shelly_lights/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 21:23:18 2018","Local":"Tue Dec 18 22:23:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
22:23:18 MQT: home/shelly_lights/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:23:18 MQT: home/shelly_lights/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T22:23:18","Switch1":"OFF","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-12-18T21:52:41","Total":0.268,"Yesterday":0.000,"Today":0.044,"Power":39,"ApparentPower":45,"ReactivePower":22,"Factor":0.87,"Frequency":51,"Voltage":227,"Current":0.197}}}
22:23:18 MQT: home/shelly_lights/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T22:23:18","Uptime":"0T00:27:01","Vcc":3.573,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"home_iot","BSSId":"7A:8A:20:21:A7:E8","Channel":11,"RSSI":100}}}

fsilva80, 2018-12-18:
> Must be a hardware issue then.
> 
> Please rule out firmware issue by erasing flash and re-flashing 6.4.0 from scratch.
> 
> Thanks

And how i do erase flash?

I believe i started flashing the original firmware with 6.3 anda after that i updated to 6.4 but i already had thi problem in 6.3...

andrethomas, 2018-12-18:
https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

pozi78, 2018-12-18:
Same problem here, upgraded to tasmota 6.4.0 and after some minutes first ghost switching
here is my syslog:

```
2018-12-18,21:58:59,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 CF 07 00 00 E1 08 88 0F 00 00 90 07 00 00 A9 11 00 00 6F 8F F4 C6 7E
2018-12-18,21:58:58,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 C8 07 00 00 E1 08 86 0F 00 00 97 07 00 00 A9 11 00 00 7D 8F C8 C6 5E
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 83 06 00 00 E1 08 21 0C 00 00 16 07 00 00 97 0E 00 00 98 95 EC C6 5F
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-MQT: home/shelly_lights/POWER1 = ON (retained)
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,"ESP-MQT: home/shelly_lights/RESULT = {""POWER1"":""ON""}"
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SRC: Switch
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-MQT: home/shelly_lights/POWER1 = OFF (retained)
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,"ESP-MQT: home/shelly_lights/RESULT = {""POWER1"":""OFF""}"
2018-12-18,21:58:57,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SRC: Switch
2018-12-18,21:58:56,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 D0 07 00 00 E1 08 88 0F 00 00 93 07 00 00 A9 11 00 00 6F 8F BA C6 48
2018-12-18,21:58:55,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 D0 07 00 00 E0 08 8C 0F 00 00 9B 07 00 00 A9 11 00 00 52 8F AE C6 2A
2018-12-18,21:58:54,Notice,172.17.1.28,user,shelly_lights-0278,ESP-SER: Received 06 19 C8 07 00 00 E1 08 74 0F 00 00 88 07 00 00 A9 11 00 00 00 90 E0 C6 D9

```

and device status:

```
22:23:18 CMD: status 0
22:23:18 SRC: WebConsole from 172.17.1.1
22:23:18 RSL: Group 0, Index 1, Command STATUS, Data 0
22:23:18 MQT: home/shelly_lights/STATUS = {"Status":{"Module":47,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"shelly_lights","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:23:18 MQT: home/shelly_lights/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:27:01","StartupUTC":"2018-12-18T20:56:17","Sleep":50,"BootCount":5,"SaveCount":29,"SaveAddress":"F7000"}}
22:23:18 MQT: home/shelly_lights/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:23:18 MQT: home/shelly_lights/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":3,"LogHost":"172.17.0.2","LogPort":514,"SSId":["home_iot",""],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
22:23:18 MQT: home/shelly_lights/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:23:18 MQT: home/shelly_lights/STATUS5 = {"StatusNET":{"Hostname":"shelly_lights-0278","IPAddress":"172.17.1.28","Gateway":"172.17.0.1","Subnetmask":"255.255.0.0","DNSServer":"8.8.8.8","Mac":"CC:50:E3:55:A1:16","Webserver":2,"WifiConfig":4}}
22:23:18 MQT: home/shelly_lights/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.0.11","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_55A116","MqttUser":"mqtt_user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:23:18 MQT: home/shelly_lights/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 21:23:18 2018","Local":"Tue Dec 18 22:23:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
22:23:18 MQT: home/shelly_lights/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:23:18 MQT: home/shelly_lights/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T22:23:18","Switch1":"OFF","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-12-18T21:52:41","Total":0.268,"Yesterday":0.000,"Today":0.044,"Power":39,"ApparentPower":45,"ReactivePower":22,"Factor":0.87,"Frequency":51,"Voltage":227,"Current":0.197}}}
22:23:18 MQT: home/shelly_lights/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T22:23:18","Uptime":"0T00:27:01","Vcc":3.573,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"home_iot","BSSId":"7A:8A:20:21:A7:E8","Channel":11,"RSSI":100}}}

```

andrethomas, 2018-12-18:
The shelly 2 inputs appear to notoriously susceptible to interference. The firmware itself cannot cause ghost switching afaik.

pozi78, 2018-12-18:
> The shelly 2 inputs appear to notoriously susceptible to interference. The firmware itself cannot cause ghost switching afaik.

thats bad news, such a good device with that big probrem.
There is a way to minimize this when using low voltage, but i dont know if there is a way when you are using mains.

Low  pass filter:
https://www.youtube.com/watch?v=aq8_os6g13s&t=195s

I tried with last version of tasmota (6.3) setting Switchdebounce to maximum (1000) and the problem reduced a lot, but i kept having ghost switching. Maybe working with these parameters could resolve the problem via software, idk.

ascillato2, 2018-12-19:
> Maybe working with these parameters could resolve the problem via software, idk.

Let's try without the internal pull up resistors of the esp8266. Please select the GENERIC module and choose:

    RELAY1: GPIO04
    RELAY2: GPIO05
    SWITCH1n: GPIO12
    SWITCH2n: GPIO14

And try again if that ghost switching happens

fsilva80, 2018-12-20:
I erased the flash and re-flashed 6.4.0
Also checked the wiring and saw that the wire attached to the relay 1 on the double switch was a bit loose...
It only as passed 40min but so far so good...
Lets see.. before doing that...

pozi78, 2018-12-20:
I am using Switchdebounce 500 and Buttondebounce 500 since yesterday even the only parameter that afects to my installation is the switch one because i have toggle switches. No ghost switching until now (Almost 20 hours). Im going to try reducing to 250ms each parameter this night and keep you informed.

Thanks

pozi78, 2018-12-26:
i keep having ghost switching, updated to the latest version but have one ghost switch just 5 min ago.
Using switchdebounce and buttondebounce 500 makes no difference
So sad

fsilva80, 2018-12-26:
> i keep having ghost switching, updated to the latest version but have one ghost switch just 5 min ago.
> Using switchdebounce and buttondebounce 500 makes no difference
> So sad

Have you tried this already?

> @itMax86 the same way you did the upgrade.
> Did you ever erased the whole flash before you flashed Tasmota?
> This can make this issues since this version is bigger and uses different flash adresses
> Link for OTA version 6.3. http://thehackbox.org/tasmota/release/archive/6.3.0/sonoff.bin

fsilva80, 2018-12-26:
Did you use the Esptool to erase the flash or you just did a reset in tasmota?
If not, try It...

pozi78, 2018-12-27:
no, only the reset in tasmota. i'll try a full flash with espeasy

arendst, 2018-12-27:
@pozi78 pls try the latest development version 6.4.1.1 released yesterday.

It contains a fix for switch input spikes which are most probably the cause of Shelly problems. 

Let me know if it fixed your problem so I can close shelly issues.

arendst, 2018-12-27:
See #4724 too.

pozi78, 2018-12-28:
Here is what i discovered:
I erased flash with esptool and then flashed 6.4.1.1

The electrical installation in the room where i have the shelly 2 controlling 2 circuits of lights shares neutral and line with the wall plug where my computer is connected with 4 screen and some other equipment. 

As soon as i turn on my pc, shelly2 stopped responding to switch 2 toggles, but switch 1 worked ok. If i turn off 2 of the screens, no matter wich ones, shelly 2 starts responding again to switch 2 toggles. If i turn them on again, shelly 2 stops responding to switch2. WTF

Now i reflashed via OTA 6.4.1 again, and shelly responds to both switches with all the screens turned on, but there is ghost switching. 

I don't know if this behaviour in 6.4.1.1 is a bug or is part of the debouncing code added

arendst, 2018-12-28:
Thx for testing.

It reveals that when the switch is open a 50Hz signal is induced. The latest fix masks all pulses over 20Hz so it never detects when the switch is open.

I need to solve a hardware design fault of the Shelly in software. For now go back to 6.4.1 and live with the gosts switching while I look to the sky for inspiration....

pozi78, 2018-12-28:
Thx to you fur such a great software and maintenance, testing is the least we should do.

Is there any problem flashing forward and backward 6.4.1 and 6.4.1.1 via OTA? Or i should erase flash each time i switch versions.

arendst, 2018-12-28:
Between those two versions there is no problem in going OTA forward and backward.

arendst, 2018-12-28:
Just rewrote the Switch handler in line with the original Shelly code. Pls give it a try and let me know if it works (better).

pozi78, 2018-12-28:
updated to 6.4.1.2 5 min ago. 
Switch 1 & 2 work as expected. 
Will inform about ghost switching

Thx

pozi78, 2018-12-30:
Informing:

Uptime | 1T14:39:34

I think not a single ghost switch and everething working as expected.
I say that i think because since i updated to 6.4.1.2 i cant find any "ESP-SRC: Switch" in the logs to catch ghost switching and just realizaed that when i toggle physical switches they are reflected as "ESP-SRC: MQTT" in the logs, both weblog and syslog.

Thx