arendst/Tasmota #4666: MQTT topic configuration does not work

welbo97, 2018-12-18:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
MQTT topic defaults to 
`%topic%/%prefix%/` 
and can not be changed to `%prefix%/%topic%` in web interface.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : iFAN02
- [x ] _Development/Compiler/Upload tools used_ : PlatformIO, esptool to erase flash
- [x ] _If a pre-compiled release or development binary was used, which one?_ : Master 6.4.0
- [x ] _You have tried latest release or development binaries?_ : no, compiled my own
- [x ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
22:18:14 MQT: media_fan/stat/STATUS = {"Status":{"Module":44,"FriendlyName":["Media Room Fan"],"Topic":"media_fan","ButtonTopic":"0","Power":11,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:14 MQT: media_fan/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://a.avolarcorp.com/tasmota/linving_fan.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:35","StartupUTC":"2018-12-18T04:12:39","Sleep":50,"BootCount":2,"SaveCount":13,"SaveAddress":"F7000"}}
22:18:14 MQT: media_fan/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-17T22:06:13","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:18:14 MQT: media_fan/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"10.1.1.1","LogPort":514,"SSId":["divot","Pearsall"],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000200"]}}
22:18:14 MQT: media_fan/stat/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FCAE794","240383A1","23B617CE","00003BF1"]}}
22:18:14 MQT: media_fan/stat/STATUS5 = {"StatusNET":{"Hostname":"media_fan-3611","IPAddress":"10.1.1.151","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"84:0D:8E:42:CE:1B","Webserver":2,"WifiConfig":4}}
22:18:14 MQT: media_fan/stat/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.avolarcorp.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_42CE1B","MqttUser":"hauser1","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:18:14 MQT: media_fan/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 04:18:14 2018","Local":"Mon Dec 17 22:18:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:20","Sunset":"17:35"}}
22:18:14 MQT: media_fan/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-17T22:18:14"}}
22:18:14 MQT: media_fan/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-17T22:18:14","Uptime":"0T00:05:35","Vcc":3.495,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","FanSpeed":3,"Wifi":{"AP":1,"SSId":"divot","BSSId":"FE:EC:DA:81:05:76","Channel":6,"RSSI":62}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_ Always reproducable

**Expected behavior**
_A clear and concise description of what you expected to happen._ MQTT topic changed after save and restart

**Screenshots**
_If applicable, add screenshots to help explain your problem._ n/a

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

witten, 2018-12-18:
FWIW, I'm also seeing this same behavior with 6.4.0 release sonoff.bin flashed with esptool.py onto an ESP12 device configured as module type "18 Generic". Previously with 6.3.0 release sonoff.bin, this issue did not occur and I could set the topic on this device just fine.

mhuca, 2018-12-18:
@welbo97 @witten 

There has been significant updates to the home assistant integration. One of the changes was enforcing %topic%/%prefix%

If you don't us the home assistant auto discovery you can disable this with

SetOption19 0
SetOption55 0
SetOption59 0
restart 1

more information can be found in the wiki:

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

welbo97, 2018-12-18:
Good enough. Thanks.

closed 2018-12-18 by welbo97