arendst/Tasmota #4670: 2.4.2 core based firmwares can't resolve DNS names.

benzino77, 2018-12-18:
**Describe the bug**
Binaries based on 2.4.2 core, are not able to connect to mqtthost, loghost and ntphost when specified by name and works only when IP is specified. There is no problem with binaries compiled against 2.3.0 core. All binaries downloaded from http://thehackbox.org

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : wemos d1 mini
- [x] _Development/Compiler/Upload tools used_ : esptool
- [x] _If a pre-compiled release or development binary was used, which one?_ : 6.4.0 based on 2.4.2 core, 6.4.0 based on 2.3.0 core and 6.4.0.1 (dev) based on 2.4.2 core
- [x] _You have tried latest release or development binaries?_ : yes
- [x] _Provide the output of command_``status 0`` :
```
release 6.4.0 core 2.3.0
00:00:00 Project sonoff Test12 (Topic test12-wm01, Fallback DVES_0C250E, GroupTopic sonoffs) Version 6.4.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 myiots in mode 11N as test12-wm01-1294...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on test12-wm01-1294.local with IP address 10.10.30.110
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/test12-wm01/LWT = Online (retained)
00:00:06 MQT: cmnd/test12-wm01/POWER = 
00:00:06 MQT: tele/test12-wm01/INFO1 = {"Module":"Sonoff Basic","Version":"6.4.0(release-sonoff)","FallbackTopic":"DVES_0C250E","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/test12-wm01/INFO2 = {"WebServerMode":"Admin","Hostname":"test12-wm01-1294","IPAddress":"10.10.30.110"}
00:00:06 MQT: tele/test12-wm01/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/test12-wm01/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/test12-wm01/POWER = OFF
08:28:55 MQT: tele/test12-wm01/STATE = {"Time":"2018-12-18T08:28:55","Uptime":"0T00:00:15","Vcc":2.788,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myiots","BSSId":"B8:27:EB:2E:AF:F9","Channel":1,"RSSI":100}}
08:28:58 CMD: status 0
08:28:58 MQT: stat/test12-wm01/STATUS = {"Status":{"Module":1,"FriendlyName":["Test12"],"Topic":"test12-wm01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:28:58 MQT: stat/test12-wm01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:18","StartupUTC":"2018-12-18T07:28:40","Sleep":50,"BootCount":5,"SaveCount":10,"SaveAddress":"FA000"}}
08:28:58 MQT: stat/test12-wm01/STATUS2 = {"StatusFWR":{"Version":"6.4.0(release-sonoff)","BuildDateTime":"2018-12-16T15:33:27","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:28:58 MQT: stat/test12-wm01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"log.lipowa54.local","LogPort":514,"SSId":["myiots",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
08:28:58 MQT: stat/test12-wm01/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
08:28:58 MQT: stat/test12-wm01/STATUS5 = {"StatusNET":{"Hostname":"test12-wm01-1294","IPAddress":"10.10.30.110","Gateway":"10.10.30.3","Subnetmask":"255.255.255.0","DNSServer":"10.10.30.3","Mac":"60:01:94:0C:25:0E","Webserver":2,"WifiConfig":4}}
08:28:58 MQT: stat/test12-wm01/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lipowa54.local","MqttPort":2883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C250E","MqttUser":"myiots","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:28:58 MQT: stat/test12-wm01/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 07:28:58 2018","Local":"Tue Dec 18 08:28:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
08:28:58 MQT: stat/test12-wm01/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T08:28:58"}}
08:28:58 MQT: stat/test12-wm01/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T08:28:58","Uptime":"0T00:00:18","Vcc":2.788,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myiots","BSSId":"B8:27:EB:2E:AF:F9","Channel":1,"RSSI":100}}}

release 6.4.0 core 2.4.2
00:00:00 Project sonoff Test12 (Topic test12-wm01, Fallback DVES_0C250E, GroupTopic sonoffs) Version 6.4.0(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 myiots in mode 11N as test12-wm01-1294...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on test12-wm01-1294.local with IP address 10.10.30.110
00:00:07 MQT: Attempting connection...
00:00:12 MQT: Connect failed to mqtt.lipowa54.local:2883, rc -2. Retry in 10 sec
00:00:22 MQT: Attempting connection...
00:00:27 MQT: Connect failed to mqtt.lipowa54.local:2883, rc -2. Retry in 10 sec
00:00:31 CMD: status 0
00:00:31 RSL: stat/test12-wm01/STATUS = {"Status":{"Module":1,"FriendlyName":["Test12"],"Topic":"test12-wm01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:31 RSL: stat/test12-wm01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:21","StartupUTC":"","Sleep":50,"BootCount":4,"SaveCount":9,"SaveAddress":"FB000"}}
00:00:31 RSL: stat/test12-wm01/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:00:31 RSL: stat/test12-wm01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"log.lipowa54.local","LogPort":514,"SSId":["myiots",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:00:31 RSL: stat/test12-wm01/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
00:00:31 RSL: stat/test12-wm01/STATUS5 = {"StatusNET":{"Hostname":"test12-wm01-1294","IPAddress":"10.10.30.110","Gateway":"10.10.30.3","Subnetmask":"255.255.255.0","DNSServer":"10.10.30.3","Mac":"60:01:94:0C:25:0E","Webserver":2,"WifiConfig":4}}
00:00:31 RSL: stat/test12-wm01/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lipowa54.local","MqttPort":2883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C250E","MqttUser":"myiots","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:31 RSL: stat/test12-wm01/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:31 1970","Local":"Thu Jan 01 00:00:31 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:00:31 RSL: stat/test12-wm01/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:31"}}
00:00:31 RSL: stat/test12-wm01/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:31","Uptime":"0T00:00:21","Vcc":3.041,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myiots","BSSId":"B8:27:EB:2E:AF:F9","Channel":1,"RSSI":100}}}

rel 6.4.0.1 core 2.4.2
00:00:00 Project sonoff Test12 (Topic test12-wm01, Fallback DVES_0C250E, GroupTopic sonoffs) Version 6.4.0.1(a232267-sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 myiots in mode 11N as test12-wm01-1294...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on test12-wm01-1294.local with IP address 10.10.30.110
00:00:07 MQT: Attempting connection...
00:00:12 MQT: Connect failed to mqtt.lipowa54.local:2883, rc -2. Retry in 10 sec
00:00:22 MQT: Attempting connection...
[... 5 lines cut ...]
00:00:46 RSL: stat/test12-wm01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:32","StartupUTC":"","Sleep":50,"BootCount":3,"SaveCount":8,"SaveAddress":"F4000"}}
00:00:46 RSL: stat/test12-wm01/STATUS2 = {"StatusFWR":{"Version":"6.4.0.1(a232267-sonoff)","BuildDateTime":"2018-12-17T23:01:14","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
[... cut ...]
esptool.py --port /dev/ttyUSB0 erase_flash 
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 /tmp/Tasmota/2.4.2-rel/sonoff.bin      
esptool.py --port /dev/ttyUSB0 erase_flash 
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 /tmp/Tasmota/2.4.2-dev/sonoff.bin 

**Expected behavior**
Binaries based on 2.4.2 core should be able to connect to MQTTHOST, LOGHOST and NTPHOST via names not only IP 

**(Please, remember to close the issue when the problem has been addressed)**
I will, but I'm sure... nevermind

andrethomas, 2018-12-18:
Can't reproduce. 2.3.0 and 2.4.2 resolve fine for me so it may be something to do with your dns server (the one allocated by dhcp) or the mqtt broker's domain service is not working.

benzino77, 2018-12-18:
I t  works fine with 2.3.0 core and not working with 2.4.2 core - I'm sure it **is not** a problem with DNS server.
I have checked with two different localizations: Office and Home (different WiFi devices). 
I have also checked different Wemos D1 mini device - so this **is not** a problem with device also.

andrethomas, 2018-12-18:
Right, but I cannot reproduce the problem - Let see if someone else can.

benzino77, 2018-12-18:
@andrethomas I assume you do not have DNS names of your MQTTHOST, NTPHOST and LOGHOST ending with `*.local`?
This is the problem (I think). When I changed MQTTHOST to `mqtt.withoutlocal.iot` (add this entry to my DNS) 2.4.2 core based firmware is able to connect (it is still not able to connect to LOGHOST and NTPHOST which names ends with `*.local`).

... and have no idea what comes to my mind to check the name without `*.local` at the end...

Jason2866, 2018-12-18:
Search for mDNS. .local is used for that and not supported in Tasmota.
As far as i remember Theo said if it works okay, if not he will not change it.
mDNS in Tasmota is only used (and supported) to discover devices not more.

benzino77, 2018-12-18:
@Jason2866  Even if I disable mDNS, and compile the sources it doesn't work:
```
#undef USE_DISCOVERY
#undef WEBSERVER_ADVERTISE 
#undef MQTT_HOST_DISCOVERY
```

Jason2866, 2018-12-18:
Could be, yes. Just dont use .local There will be no fix.
[image]

benzino77, 2018-12-18:
Still don't get it why it is working perfect on 2.3.0 and doesn't on 2.4.2... disable mDNS on device (by compile directives) doesn't change the situation, even if this is no more "related to mDNS" because support for that is not included in the firmware.

What is a plan for future releases? It will be based on 2.4.2+ or 2.3.0 or on both?

benzino77, 2018-12-18:
.. .to clarify: all my `*.local` names are provided by my DNS server (no magic mDNS service/provider which answer to mDNS requests: "where is mqtt broker" or "where is ntp server" or "where is syslog server").

andrethomas, 2018-12-18:
@benzino77 unless you are explicitly using mDNS you should avoid the .local TLD as it is reserved for and used by mDNS (RFC6762 & RFC6763)

I do not use mDNS and like you I have a DNS server with the hostnames configured for devices for simplicity sake, but I use a different TLD to avoid conflict with RFC6762/RFC6763 because I do actually have Apple devices on my network.

So if you're not using mDNS then just use a different TLD for the same purpose - In my case, I use sensornet so my mosquito mqtt server is on hostname mqtt.sensornet.

jziolkowski, 2018-12-18:
Basically, you only use subdomain in a LAN environment when it's different than .local and/or different than the domain the device is in.
When you use only the hostname, your DNS client should automatically add the domain to the query.

benzino77, 2018-12-18:
So it looks like 2.3.0 doesn't fit to standards (according to RFC documents you provided regarding *.local Top Level Domain) :)

Anyway, I know how to workaround this problem (I have full control of DNS entries in my environments, so I can create a new TLD) - this part is solved.

How about the question:

> What is a plan for future releases? It will be based on 2.4.2+ or 2.3.0 or on both?

Jason2866, 2018-12-18:
The release is on 2.4.2. Version 2.3.0 is compiled because it is still possible...
In future core 2.5.0 will be the base.

benzino77, 2018-12-18:
@jziolkowski I use form: host_or_service_name.some_name.local. E.g.: mqtt.lipowa54.local or ntp.lipowa54.local etc. So there is a "subdomian" before "host" part.

Anyway it works with 2.3.0 and it doesn't with 2.4.2.

closed 2018-12-18 by benzino77

andrethomas, 2018-12-18:
> So it looks like 2.3.0 doesn't fit to standards (according to RFC documents you provided regarding *.local Top Level Domain) :)

Partially correct because when a .local hostname needs at all times to be resolved the core should have used the multicast method defined in the RFC's I mentioned - This means that the request does not even go to your DNS server but rather to a broadcast MAC address to which the respective mDNS configured device should respond "Hey yeah that's me.. here's my ip"

Obviously 2.3.0 was not as strict and when not receiving a response to a mDNS request it reverted to your DNS server on your lan which also puts into question why your DNS is responding to the .local TLD because in theory  this should also not happen. So they likely implemented mDNS in 2.3.0 with a potential failover to normal DNS in the event that a mDNS multicast was not answered.

I don't know how to explain it any more simple than that.

> What is a plan for future releases? It will be based on 2.4.2+ or 2.3.0 or on both?

@arendst will answer, but my best guess is 2.4.2 and 2.5.0.

andrethomas, 2018-12-18:
Wait, actually I can explain it in a simpler way.

You were using the .local TLD in contrast to the standards defined by the RFC's - You got away with it because the 2.3.0 core allowed you to do so.

Welcome to 2.4.2 :)

andrethomas, 2018-12-18:
> I haven't noticed that there can be something wrong with my approach.

Its not impossible to get it to work, you just need to get your mqtt server to respond to mDNS multicast frames but to be honest its a lot easier to just use a different TLD.

Windows can use and respond to mDNS (Apple's version of NetBios) if the Bonjour service provided to Microsoft by Apple is enabled on Windows. So yes, in most cases it should work for normal machine to machine but I'm guessing  the problem here is that your mqtt service is not on a device that will respond to mDNS multicast frames.

Essentially mDNS does not use a DNS server at all.

The client device sends a multicast frame and those devices that are mDNS enabled will see this frame and respond to it accordingly. It has nothing to do with traditional DNS whatsoever.

NetBios works the same on Windows - this is why you can see other computers in your "Network Neighborhood" even though you have not configured an entry for them on the DNS server.

Are we done now? :)

benzino77, 2018-12-18:
We ware done 4 posts earlier...

arendst, 2018-12-18:
As you're done here I still want to mention that the goal is to go forward so 2.4.2+ is the way to go.

n0dyjeff, 2019-01-01:
I have the same issue with Core 2.4.2 and a ".local" TLD. I understand the explanation and that one "workaround" is to change TLD to something other than ".local". Unfortunately, some time ago I (stupidly) set up my entire network based on a Window Active Directory Domain named "davis.local". It would be a huge undertaking to change the domain at this point. The problem is compounded by the fact that my SYSLOG and MQTT server is on a different LAN segment, so it's not even "link-local" to the sonoff device.

Does anyone know of a workaround that would keep me from having to reconfigure my entire network to another domain? For now, I'm sticking with Core 2.3.0, but I'd rather not be stuck there forever...

Hmmm. I sort of answered my own question by realizing that I could hard-code IP addresses for the servers, rather than using domain names. That's a possibility, but I'm still looking for a solution that allows me to continue using DNS. Any help?

andrethomas, 2019-01-01:
Given that the .local TLD is supposed to be used for mDNS and that no conventional DNS server should respond to .local hostname resolution requests - If it does it is doing so in contrast to RFC6762 & RFC6763.

The mere fact that the 2.3.0 core reverts to conventional DNS to resolve a .local hostname is because mDNS was not properly implemented in 2.3.0 - You can always check if they made the same mistake in 2.5.0 but my advice would be to steer clear of .local hostnames unless you are explicitly using mDNS.

jziolkowski, 2019-01-01:
Again, you're trying to find a soultion to a problem which is not a problem, but a consequence of poor design choices upon creating your network. It's like asking how to avoid paying tickets when you choose to cross red lights.

benzino77, 2019-01-01:
I'm not a WIN expert, but AD can handle more than one domain (additional suffixes to be precise) or a new domain in the forest?
You can than create new entry davis.notlocal or something, pointing to the same IP address.

m0rph2k, 2019-01-01:
Hi

My solution with a Windows DNS Server. 
1) Add new Primary Zone eg. mqtt-sonoff
2) New A Record to mqtt Server ip
3) Add Tasmota Mqtt Host without *.local eg. mqtt-sonoff

n0dyjeff, 2019-01-02:
> Hi
>
> My solution with a Windows DNS Server.
>
> * Add new Primary Zone eg. mqtt-sonoff
> * New A Record to mqtt Server ip
> * Add Tasmota Mqtt Host without *.local eg. mqtt-sonoff

Hi, 

Thanks for the helpful response. I will give that a shot.  

In the longer term, I'm also looking into using an mDNS reflector on my
router and installing avahi-daemon on my server so that it will respond
to .local mdns requests. This seems like it will take some effort, but
is still less than redoing network to get rid of .local addresses. 

Thanks again, 

Jeff

 

Links:
------
[1]
https://github.com/arendst/Sonoff-Tasmota/issues/4670#issuecomment-450756343
[2]
https://github.com/

n0dyjeff, 2019-01-02:
Hi, 

Thanks for the helpful response. I will give that a shot.  

In the longer term, I'm also looking into using an mDNS reflector on my
router and installing avahi-daemon on my server so that it will respond
to .local mdns requests. This seems like it will take some effort, but
is still less than redoing network to get rid of .local addresses. 

Thanks again, 

Jeff

 

Links:
------
[1]
https://github.com/arendst/Sonoff-Tasmota/issues/4670#issuecomment-450745887
[2]
https://github.com/

KDMcMullan, 2019-11-02:
In case it's of any use, I was getting DNS: failed on each of my devices. I found that this behaviour completely disappeared by doing setoption55 0. I subsequently found that setoption55 1 did NOT cause the fault to return. I don't know if this is relevant.

Jason2866, 2019-11-03:
mDNS in older core has faults and do make weird issues....
This is the reason why mDNS is by default off.

Jason2866, 2023-03-03:
@FelR We are very happy with removing mDNS from Tasmota. No more bug hunting for strange issues coming from mDNS. No come back...