arendst/Tasmota #4671: Battery suggestion

helmar74, 2018-12-18:
Hi,

I'm planning to run the Tasmota fork from stefan bode on a Wemos D Mini, connected with a HC-SR04.
([https://github.com/stefanbode/Sonoff-Tasmota])
The Device should be battery powered.
As a HC-SR04 is connected, I need a 5V battery or battery pack. The battery should have round about 10000mAh
I tried several usb power packs, but unfourtnately all powerpacks I tried had a "auto power down" mode. So it was powering down, when the wemos mini entered deepsleep mode...

Has anyone an suggestion which battery, usb powerpack or battery pack would meet my requirements?

Thanks a lot

Helmar

Jason2866, 2018-12-18:
Not a easy answered question. Please visit Discord for this

ascillato2, 2018-12-18:
Closing this issue as it is not a Tasmota software issue. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-18 by ascillato2