arendst/Tasmota #4677: Energy today, total,...reporting

vitapower82, 2018-12-18:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Atom, platformio,FTDI...
- [x] Hardware used : Sonoff pow r2
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? : ?
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
21:31:21 CMD: status 0
21:31:21 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:31:21 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T08:56:29","StartupUTC":"2018-12-16T11:34:52","Sleep":250,"BootCount":5,"SaveCount":59,"SaveAddress":"FA000"}}
21:31:21 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:31:21 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vita@home",""],"TelePeriod":30,"SetOption":["00208009","560980C0","00000000"]}}
21:31:21 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
21:31:21 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6889","IPAddress":"192.168.178.100","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:45:7A:E9","Webserver":2,"WifiConfig":4}}
21:31:21 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.18","MqttPort":1883,"MqttClientMask":"Sonoff-Licht-Vorne","MqttClient":"Sonoff-Licht-Vorne","MqttUser":"big.vitali@gmx.de","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:31:21 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 20:31:21 2018","Local":"Tue Dec 18 21:31:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:39","Sunset":"16:54"}}
21:31:21 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:31:21 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T21:31:21","ENERGY":{"TotalStartTime":"2018-12-16T11:49:09","Total":3.0461,"Yesterday":0.9205,"Today":1.9130,"Power":1.4,"ApparentPower":8.5,"ReactivePower":8.4,"Factor":0.17,"Voltage":240,"Current":0.035}}}
21:31:21 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T21:31:21","Uptime":"2T08:56:29","Vcc":3.171,"POWER":"ON","Wifi":{"AP":1,"SSId":"vita@home","BSSId":"08:96:D7:89:B9:7B","Channel":13,"RSSI":52}}}

**(Please, remember to close the issue when the problem has been addressed)**

I use christmas light with consumption of 1,4W approx. 10 hours per day.
It makes 14 Wh per day or 0,014 kWh (approx.)
Tasmota showing values in kWh (I'm not sure if it is correct, because in wiki is written Wh) like this:
measured at 19:00 -- 0,6149kWh
measured at 20:00 -- 1,2136kWh

--> makes consumption of arround 600W

Sonoff pow r2
Sonoff-Tasmota 6.3.0

andrethomas, 2018-12-18:
Can you post a screenshot of the main web ui page where it shows the values when the device is in the ON position?

vitapower82, 2018-12-18:
[image]

21:31 -- Today":1.9130
22:54 -- Today":3.1101
 fully not realistic with 1,4W

I sa that several peoples claimed this issue, but I don't see solution

andrethomas, 2018-12-18:
Did you calibrate the device? Really low power factor you got going there...

vitapower82, 2018-12-18:
I calibrate voltage and power. Something more to do?

andrethomas, 2018-12-18:
Did you do it with a normal resistive load like an incandescent light bulb?

vitapower82, 2018-12-18:
no, all old lamps are away. 
tested voltage with multimeter & power with halogen lamp

andrethomas, 2018-12-18:
Halogen should give you a power factor of 1 if calibration is correct.

The problem here is that the apparent power (voltage x current) = 8.5, so that means 8.5 Watts of power but the fact that there is a low power factor suggests that it is connected to a load that uses switch mode power regulation which causes the voltage curve and current curve to be almost entirely out of sync - This is why you have a low power factor.

So the question is whether or not the load factor is indeed correct as 8.5 x 24 hours = ~ 0.2KW per day so that also does not explain the stored values.

I suggest connecting to the halogen again and see if you have a power factor of 1 after calibration.

vitapower82, 2018-12-18:
ok, will do tomorrow

andrethomas, 2018-12-18:
Thanks, appreciate the feedback.

ascillato2, 2018-12-19:
Hi

Have you managed to solve your issue?

vitapower82, 2018-12-19:
before calibration of power with a lamp, power factor was 0,93-0,94.
After set the factor to 1, same bad results (energy total, yesterday, today I did reset.)

Energy of 0,2kWh after 2 minutes with 1,4W power
[image]

ascillato, 2018-12-19:
Please, try connect another thing like your washer machine or something like that and please report back your power factor. If your light has a very poor switching power source, may be it is making a lot of electrical noise and make the pow to read bad. A lot of electrical noise is a limitation of the power measurement chip.

vitapower82, 2018-12-19:
good proposal to connect "washer machine ". socket is outside and sonoff in garage. :-)

But you are right, I have to eliminate the point with possible noises from LED light.

andrethomas, 2018-12-19:
A washing machine will also have some reactive effect - The best is to see if you can get a power factor of 1 with a resistive load - use a heater or toaster or a kettle.

andrethomas, 2018-12-19:
Oh wait, LED Christmas lights will definitely give you a low power factor. I think the problem is that this is not factored in with the usage or so it would seem.

Perhaps try to reset the values which were stored and monitor over a 1 hour period.

Relevant commands are from wiki

```
EnergyReset1 0 // ((P)Re)Set Energy Today in Wh
EnergyReset2 0 // ((P)Re)Set Energy Yesterday in Wh
EnergyReset3 0 // ((P)Re)Set Energy Total in Wh
```

vitapower82, 2018-12-19:
something is defenitelly with sonoff.
I conncted directly second sonnof after "broken" one. Second works with report of energy perfect.
First works in tha same time as before =not correct.

so it is not noise desturbing from LED light.
Could it be some problems during flashing (use same version of tasmota, flashing was same way) or sensors in Sonoff are down ?

ascillato, 2018-12-19:
> Could it be some problems during flashing

Sorry, no. Seems to be, as you said, a hardware issue. Sorry.

andrethomas, 2018-12-19:
There's no harm in doing a complete erase and reflash - but its more likely a hardware problem.

ps.. we're still waiting for a photo of the Christmas lights :)

ascillato, 2018-12-19:
Photo, Photo, Photo, Photo, Photo, Photo, Photo, Photo, Photo :rofl:

vitapower82, 2018-12-19:
nothing special :)
last chance to flash again on weekend.

Thanks guys!!

closed 2018-12-19 by ascillato2

vitapower82, 2018-12-21:
1. erase "with nodemcu-flasher-master"
2. reflash
3. OTA update to Tasmota 6.4.0
4. EnergyResetX 0

-->crazy, but work fine since 7 hours

andrethomas, 2018-12-21:
@vitapower82 That's good to read. Possibly just a bad initial flash then - this is not uncommon depending on your flashing setup - we've seen this often on Discord Chat.