arendst/Tasmota #4678: HA Autodiscovery misses pressure Unit Of Measure

dcandi, 2018-12-18:
Pressure sensor is reported without unit of measure in the MQTT HA discovery packet. 
HA shows the entity history as a bar, like for a switch, instead of a graph, like for temperature sensors.

**Versions**
HW: nodeMCU board with BME280 sensor
Tasmota: 6.4.0 release binary (sonoff.bin)
HA: 0.84.2

**Status 0 output**
```
23:00:36 CMD: status 0
23:00:37 MQT: sonoff/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:00:37 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:22:40","StartupUTC":"2018-12-18T21:37:57","Sleep":50,"BootCount":10,"SaveCount":70,"SaveAddress":"F6000"}}
23:00:37 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:00:37 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["candi-fi",""],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000220"]}}
23:00:37 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
23:00:37 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1901","IPAddress":"192.168.1.238","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:01:67:6D","Webserver":2,"WifiConfig":4}}
23:00:37 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.65","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_01676D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:00:37 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 22:00:37 2018","Local":"Tue Dec 18 23:00:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
23:00:37 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T23:00:37","BME280":{"Temperature":19.8,"Humidity":27.6,"Pressure":990.3},"PressureUnit":"hPa","TempUnit":"C"}}
23:00:37 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T23:00:37","Uptime":"0T00:22:40","Vcc":3.001,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":40,"Color":"0,50,102","HSBColor":"211,100,40","Channel":[0,19,40],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"candi-fi","BSSId":"5C:E2:8C:A9:21:80","Channel":8,"RSSI":78}}}
```

**To Reproduce**
[setup hass and tasmota for autodiscovery](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant)
Wire up a BME280 sensor to the module. 
Configure module for D3 = I2C SDA and D4 = I2C SCL
setoption55 1
setoption19 1
reboot module

The pressure packet output is:
```
23:08:59 MQT: homeassistant/sensor/01676D_BME280_Pressure/config = {"name":"Sonoff BME280 Pressure","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","val_tpl":"{{value_json['BME280'].Pressure}}","uniq_id":"01676D_BME280_Pressure","device":{"identifiers":["01676D"],"name":"Sonoff","model":"Generic","sw_version":"6.4.0(sonoff)","manufacturer":"Tasmota"}, "~":"sonoff/tele/"} (retained)
```
[image]

**Expected behavior**
Instead the packet shall contain "unit_of_meas":
```
23:08:59 MQT: homeassistant/sensor/01676D_BME280_Pressure/config = {"name":"Sonoff BME280 Pressure","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","unit_of_meas":"hPa","val_tpl":"{{value_json['BME280'].Pressure}}","uniq_id":"01676D_BME280_Pressure","device":{"identifiers":["01676D"],"name":"Sonoff","model":"Generic","sw_version":"6.4.0(sonoff)","manufacturer":"Tasmota"}, "~":"sonoff/tele/"} (retained)
```

ascillato2, 2018-12-18:
@emontnemery please, can you take care of this one? Thanks.

dcandi, 2018-12-19:
@emontnemery I got it. Tomorrow I'll work out how to make a pull request.

emontnemery, 2018-12-19:
@ascillato2 I've been discussing with @dcandi on discord, I think he's on it :)

ascillato2, 2018-12-19:
PR #4692

closed 2018-12-19 by ascillato2

fluppie, 2018-12-19:
Cool, I wanted to report this bug as well :).
[image]

Normal behaviour:
[image]

Counters have the same issue, I report the CPM value of a Geiger counter:
[image]

And while where at it, the Lux values of a BH1750 as well:
[image]

dcandi, 2018-12-19:
wow! did not know you could attach a geiger counter to WemosD1! Which hardware is that?
Btw if you don't use autodiscovery you can setup the sensor into configuration **with their unit of measure** and you get the graph.