arendst/Tasmota #4679: Tasmota Sonoff TH MQTT tele status cannot retrive

eadin79, 2018-12-18:
Hello,
I'm using Node-red and Mosquitto for Sonoff TH and SI7021 sensor.
Power ON/OFF and temprature humidity I can get by MQTT commands, but I cannot retrive tele/sonoff1/STATUS ... what is wrong in my request to mqtt broker? or why tasmota sonoff is not aswering?
I tried cmnd/sonoff1/STATUS 0.... still nothing.

PS: I'm newbie ,but I learn.

eadin79, 2018-12-18:
[image]
wih first 2 stat and tele works someting more, but wth tele/sonoff1/uptime or anything else not except wildcard #

ascillato2, 2018-12-18:
Hi,

Seems that your config is not correct. 
As we need a lot more information to properly help you, please address this to the Tasmota Support Chat. Github issues are for bug reporting with the provided tenplates. Thanks

closed 2018-12-18 by ascillato2

eadin79, 2018-12-18:
a link to Tasmota Support chat please ...

ascillato, 2018-12-18:
In the readme (main webpage of Sonoff-Tasmota) are the links to the chat, wiki and forum.

Chat: https://discord.gg/Ks2Kzd4

garymcleanuk, 2018-12-19:
> [image]
> wih first 2 stat and tele works someting more, but wth tele/sonoff1/uptime or anything else not except wildcard #
> even with timestamp sending 0 to cmnd/sonoff1/status I got nothing back :-)
> 
> what I did wrong?
> [image]

your Command is incorrect when trying to grab the status. You need to PUBLISH thatn subscrive to start with and you need to send payload Zero or 4 if you want status 4 Try this:

mosquitto_pub -h 192.168.1.7 -p 1883 -u admin -P test -t cmnd/sonoff1/status -m "0"