arendst/Tasmota #4680: PZEM-004T RX TX Ports Swapped

xNuno, 2018-12-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

After upgrade from 6.3.0 to 6.4.0 I had to change TX and RX ports on the config without changing nothing at hardware level.

On 6.3.0 I had:
[image]

On 6.4.0 I changed to:
[image]

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X ] _Searched the problem in issues and in the wiki_
- [ X ] _Hardware used_ :  NodeMCU ESP8266 ESP-12E for arduino CP2102
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ X ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
00:13:59 MQT: stat/sonoff_power_meter/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff Power Meter"],"Topic":"sonoff_power_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:04","StartupUTC":"2018-12-19T00:08:55","Sleep":50,"BootCount":57,"SaveCount":119,"SaveAddress":"F4000"}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SilvaN",""],"TelePeriod":30,"SetOption":["00C08009","558180C0","00000000"]}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS5 = {"StatusNET":{"Hostname":"sonoff_power_meter-4096","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:3C:30:00","Webserver":2,"WifiConfig":5}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C3000","MqttUser":"SilvaN","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 19 00:13:59 2018","Local":"Wed Dec 19 00:13:59 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+00:00","Sunrise":"07:39","Sunset":"15:54"}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS10 = {"StatusSNS":{"Time":"2018-12-19T00:13:59","ENERGY":{"TotalStartTime":"2018-11-27T19:45:57","Total":189.977,"Yesterday":8.446,"Today":0.027,"Power":250,"ApparentPower":350,"ReactivePower":245,"Factor":0.71,"Voltage":235,"Current":1.490}}}
00:13:59 MQT: stat/sonoff_power_meter/STATUS11 = {"StatusSTS":{"Time":"2018-12-19T00:13:59","Uptime":"0T00:05:04","Vcc":3.098,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"SilvaN","BSSId":"50:64:2B:92:C6:1D","Channel":8,"RSSI":66}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_

Just play with TX and RX ports...

**Expected behavior**
_A clear and concise description of what you expected to happen._

It should work as it's hardware connected. RX -> TX and TX -> RX

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-19:
You are right. No bug. Wanted change. See changelog.

ascillato2, 2018-12-19:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-19 by ascillato2

meingraham, 2018-12-20:
What version of TASMOTA?

digiblur, 2018-12-20:
About 10 different versions it feels like and different cores as well as I tried the various other pins on other generic boards. That's why I am thinking it is toast.

xNuno, 2018-12-20:
Yes it worked after login on web and swap the config like I show on picture. Don't forget to power off, wait 5s and power on again both devices, pzem and nodemcu or whatever you are using.