arendst/Tasmota #4681: Water flow sensor reading

ccfiel, 2018-12-19:
I have tried a water flow sensor yf-s201 but I have a problem when I install it to the water system and compare the reading of the manual meter with the water flow sensor there reading has a very big variance  example in the manual reading it consume 20 liters but In the tasmota and yf-s201 sensor it just consume 5 liters.

my computation on how to compute liters is ( pulse / 450)  because the data sheett of this sensor says Pulses per Liter: 450

Am I doing it right? or I miss something?

[image]

ascillato, 2018-12-19:
Have you set properly the command counterdebounce and command countertype?

ccfiel, 2018-12-19:
hello @ascillato ,

I already edit my post for template about the counter type
```
05:55:16 CMD: countertype1
05:55:17 MQT: stat/sonoff-2371/RESULT = {"CounterType1":0}
```
the value is zero which means pulse Counter should I change it to 1 (pulse Timer)?

and the counterdebounce is zero for now
```
05:55:54 CMD: counterdebounce
05:55:54 MQT: stat/sonoff-2371/RESULT = {"CounterDebounce":0}
```

any idea how to to get the right counter debounce?

Cheers,
Chris

--

05:55:54 CMD: counterdebounce
05:55:54 MQT: stat/sonoff-2371/RESULT = {"CounterDebounce":0}

hhaim, 2018-12-19:
One question regarding this. I have a different water meter and it works great. 
I don't want to reset the count with tasmota rules ( as I might miss a tele/status) 
The counter value is zero after reset, is there a way to make it persistence ?

Jason2866, 2018-12-19:
See rules https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#rule-commands

ccfiel, 2018-12-19:
@hhaim 

What is your water meter model? This is the rule I made to make it persistence

```
Rule1 on COUNTER#C1>0 do backlog Add1 %value%; counter1 0 endon
Rule2 on Time#Minute|15 do mem1 %var1% endon
Rule3 on system#boot do var1 %mem1% endon
```

hhaim, 2018-12-19:
Nice!  Is there a price to save it every sec? Is it saved to flash (with limited write) or to the eeprom? If there is a limit I could have a better solution without saving it. 
I have a water meter from Arad with EV pulses

ascillato2, 2018-12-19:
@ccfiel 

Please, try the `counterdebounce `command to eliminate the noise and adjust it until your measured value is ok. Remember also that this water-flow sensor is super cheap and not precise.

Please, let us know how that goes. Thanks

closed 2018-12-19 by ascillato2

hhaim, 2018-12-19:
regarding the  flash I agree.  
However I think it is better **not** to use the clear rule in every tele.  
The reason is that you might miss a message (Broker is down) better to use diff from an incremented counter (Tasmota save it as 64bit )  and in case of reboot compare the boot_counter. 
So controller will always **add** the diff if the boot_count is the same. in case it is not the same reset the reference and start other. this way the probably to loose something is only when device is in reset.

ascillato, 2018-12-19:
Your EEPROM is not a Harddrive. Flash is specified for up to 100000 erase cycles. So that aproach will wear your flash out in about 3 weeks.

hhaim, 2018-12-19:
I'm not suggesting to save the info to flash. just not to have the rule that clear the counter

ascillato, 2018-12-19:
Ok, what you want to do is different than what this issue was about.

You can use rules to store in vars last 5 pulses/min when mqtt disconnects, and then send them when mqtt connects. Is not an easy rule but it is doable. You can check the wiki for examples in rules.

Remember that if you use the mem instead vars you will wear the flash if you use them a lot of times per day. That is why, for this should be better to use vars.

ccfiel, 2018-12-20:
@ascillato 

What I tested is I have a 1 liter container and get the pulse count but it's not stable this is what I get.

counterdebounce  |  0
countertype           |  0
sensor pin             |  d2 (gpio4)

High Pressure (pulse count/1 liter)   |   Low Pressure (pulse count/1 liter)
375                                                    |    314
412                                                    |    330
363                                                    |    268
350                                                    |    258
362                                                    |    339

=================================================================

counterdebounce  |  100
countertype           |  0
sensor pin             |  d2 (gpio4)

High Pressure (pulse count/1 liter)   |   Low Pressure (pulse count/1 liter)
131                                                    |    288
129                                                    |    298
127                                                    |    287
132                                                    |    232
133                                                    |    263

=================================================================
```
Program Version	6.4.0(sensors)
Build Date & Time	2018-12-16T14:35:56
Core/SDK Version	2_4_2/2.2.1(cfd48f3)
Hardware: Wemos D1 mini
```

```
04:46:42 CMD: status 0
04:46:42 MQT: stat/sonoff-7804/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff-7804","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:46:42 MQT: stat/sonoff-7804/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:17:57","StartupUTC":"2018-12-20T03:28:45","Sleep":50,"BootCount":41,"SaveCount":132,"SaveAddress":"F8000"}}
04:46:42 MQT: stat/sonoff-7804/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sensors)","BuildDateTime":"2018-12-16T14:35:56","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
04:46:42 MQT: stat/sonoff-7804/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["bai-wela-wifi",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
04:46:42 MQT: stat/sonoff-7804/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
04:46:42 MQT: stat/sonoff-7804/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7804-7804","IPAddress":"192.168.88.142","Gateway":"192.168.88.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.88.1","Mac":"80:7D:3A:3D:5E:7C","Webserver":2,"WifiConfig":4}}
04:46:42 MQT: stat/sonoff-7804/STATUS6 = {"StatusMQT":{"MqttHost":"iot.metromatik.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3D5E7C","MqttUser":"device","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:46:42 MQT: stat/sonoff-7804/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 20 03:46:42 2018","Local":"Thu Dec 20 04:46:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:55"}}
04:46:42 MQT: stat/sonoff-7804/STATUS10 = {"StatusSNS":{"Time":"2018-12-20T04:46:42","COUNTER":{"C1":2418},"ANALOG":{"A0":8}}}
04:46:42 MQT: stat/sonoff-7804/STATUS11 = {"StatusSTS":{"Time":"2018-12-20T04:46:42","Uptime":"0T00:17:57","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"bai-wela-wifi","BSSId":"FC:EC:DA:AD:58:D0","Channel":11,"RSSI":90}}}`
![20181220_113545](https://user-images.githubusercontent.com/1788360/50262556-db03dd80-044c-11e9-923c-b31ea6f6921e.jpg)

```

> @ccfiel
> 
> Please, try the `counterdebounce `command to eliminate the noise and adjust it until your measured value is ok. Remember also that this water-flow sensor is super cheap and not precise.
> 
> Please, let us know how that goes. Thanks

hhaim, 2018-12-21:
@ccfiel I have this [Arad with EV output for each Litter](https://arad.co.il/assets/M-August-2017.pdf]) at my home primary water entrance.
I get exacly 1 pulese for every litter that the manual counter reports (1000 pulses for one cubic meter) . The EV and manual counter are in sync.  It's cost about ~50-100$. I'm not sure what is your usecases but if you require accurate mesurment this is the type of meter you need. 
I'm using it for leak monitor consumption report.

hhaim, 2019-01-27:
A hass custom component for reading the counter and sync to the right value in case of power up/down without saving the counter to flash. It sync using the up-time field. 
No need for sending diff rule suggested by @ascillato (which could be not accurate in case of missing mqtt packets)
 
the component. 
[tasmota_counter](https://github.com/hhaim/hass/blob/master/custom_components/sensor/tasmota_counter.py)

[example](https://github.com/hhaim/hass/blob/master/configuration.yaml#L283)

Example of usage (under sensor)
```
 - platform: tasmota_counter
    name: water_total
    stopic: water_out
    counter_id: 1
    max_valid_diff: 2000
    unit_of_measurement: 'l'
    icon: mdi:water-pump
    expire_after: 300
    value_template: "{{ (4885 + (value))|int }}"
```

ortegafernando, 2023-07-19:
> regarding the flash I agree. However I think it is better **not** to use the clear rule in every tele. The reason is that you might miss a message (Broker is down) better to use diff from an [...]

Sorry, is it 64bits or 32bit? 
https://github.com/arendst/Tasmota/blob/3da5124348958d9cc7f4669fc27edb3f15b7d014/tasmota/tasmota.ino#L150C12-L150C20

arendst, 2023-07-21:
32-bit

hhaim, 2023-07-23:
> Sorry, is it 64bits or 32bit? https://github.com/arendst/Tasmota/blob/3da5124348958d9cc7f4669fc27edb3f15b7d014/tasmota/tasmota.ino#L150C12-L150C20

I've looked into my hass  tasmota counter component code, it assumes the counter is firmware counter is 32bit.