arendst/Tasmota #4685: Sonoff Tasmota update

wroblewski6, 2018-12-19:
I have 5.0.12 tasmota how can i update Ota and not loose any Of my settings 
 
I have read read me but have no clue because upgrade button do nothing ?

wroblewski6, 2018-12-19:
URL ota is http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin

andrethomas, 2018-12-19:
First, you need to upgrade to 5.14.0 from https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff.bin (do not do this using OTA, **you need to do this one manually from the web ui**)

Once you are running 5.14.0 you can OTA by url to release 6.4.0 from http://thehackbox.org/tasmota/release/sonoff.bin

Be advised that the latter requires a two-step automatic upgrade, first to the intermediate minimal and then to the new firmware itself so do not interrupt the device during this time - usually would take up to 1-2 minutes depending on your internet speed.

wroblewski6, 2018-12-19:
When I try to update in web ui 

I got 

Upload Failed

Upload buffer miscompare

wroblewski6, 2018-12-19:
What is worse after this the sonoff is totally not visible in my lan and do not work on goio14 press . Now I am worried to do it on next sonoff

andrethomas, 2018-12-19:
Try uploading the minimal first and see what happens

wroblewski6, 2018-12-19:
And how to do it ?? By usb ?? Because I can not see them in my lan and it is very difficult for me to connect it by usb

wroblewski6, 2018-12-19:
Can I do it by lab some how ??

wroblewski6, 2018-12-19:
Lab=lan

wroblewski6, 2018-12-19:
Please be so kind and help me

Jason2866, 2018-12-19:
Is it reachable via mqtt?

ascillato2, 2018-12-19:
Hi,

Seems that your upgrade has left some incorrect values in your flash.

Please, erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/.

If you need further assistance with this, please find us in the Tasmota Chat. Thanks.

Support Information
See Wiki for more information.
See Chat for more user experience.

closed 2018-12-19 by ascillato2