arendst/Tasmota #4686: Wifi Issues with 6.4.0

cbennett6008, 2018-12-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
Issues with MQTT Disconnects on 6.4.0 with 2.4.2 core

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : D1 Mini, Sonoff Basic V2, Shelly1, Shelly 2, Sonoff S31
- [x] _Development/Compiler/Upload tools used_ : FlashESP8266.exe (just a GUI for esptool.exe -vv -cd nodemcu -cb 115200 -cp COM5 -ca 0x00000 -cf firmware.bin) - source here: https://github.com/BattloXX/ESPEasyFlasher/blob/master/FlashESP8266/Form1.cs

- [x] _If a pre-compiled release or development binary was used, which one?_ : 6.4.0 with 2.4.2 core from github
- [x] _You have tried latest release or development binaries?_ :
- [x] _Provide the output of command_``status 0`` :
```
06:36:18 CMD: status 0
06:36:18 SRC: WebConsole from 192.168.0.107
06:36:18 RSL: Received Topic /status, Data Size 1, Data 0
06:36:18 RSL: Group 0, Index 1, Command STATUS, Data 0
06:36:18 MQT: stat/garage_door/STATUS = {"Status":{"Module":18,"FriendlyName":["Garage-Door"],"Topic":"garage_door","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"garagestate","SwitchMode":[0,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
06:36:18 MQT: stat/garage_door/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T15:55:14","StartupUTC":"2018-12-18T19:41:04","Sleep":50,"BootCount":8,"SaveCount":27,"SaveAddress":"F9000"}}
06:36:18 MQT: stat/garage_door/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
06:36:18 MQT: stat/garage_door/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"192.168.248.10","LogPort":514,"SSId":["Bennett-IoT",""],"TelePeriod":300,"SetOption":["00008229","558180C0","00000000"]}}
06:36:18 MQT: stat/garage_door/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
06:36:18 MQT: stat/garage_door/STATUS5 = {"StatusNET":{"Hostname":"garage_door","IPAddress":"192.168.248.15","Gateway":"192.168.248.1","Subnetmask":"255.255.248.0","DNSServer":"192.168.0.40","Mac":"60:01:94:B1:1A:C7","Webserver":2,"WifiConfig":4}}
06:36:18 MQT: stat/garage_door/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.248.10","MqttPort":1883,"MqttClientMask":"garage_door","MqttClient":"garage_door","MqttUser":"hassio","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
06:36:18 MQT: stat/garage_door/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 19 11:36:18 2018","Local":"Wed Dec 19 06:36:18 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 03:00:00 2018","Timezone":99,"Sunrise":"02:39","Sunset":"10:54"}}
06:36:18 MQT: stat/garage_door/STATUS10 = {"StatusSNS":{"Time":"2018-12-19T06:36:18","Switch2":"OFF"}}
06:36:18 MQT: stat/garage_door/STATUS11 = {"StatusSTS":{"Time":"2018-12-19T06:36:18","Uptime":"0T15:55:14","Vcc":2.994,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Bennett-IoT","BSSId":"36:A4:3C:0D:00:56","Channel":1,"RSSI":68}}}

```

**To Reproduce**
I flashed from 6.3.0 bin to 6.4.0 bin OTA.   Everything was fine on 6.3.0  After flashing all my devices began having repeated disconnects from MQTT.  I flashed most of my devices with 6.4.0 with the older core, and the problem has gone away on those devices.

**Expected behavior**
A lack of disconnects

**Screenshots**
N/A

**Additional context**
Have been troubleshooting in discord, #issues, 
Reset 5 and reconfigure did not work
The test device experienced no disconnects when it was on my test bench, underneath an Access point (for three hours)
I have also disabled the IoT SSID on my second access point, and set the remaining AP to a fixed channel, instead of auto, still experiencing disconnects on my test device (the test AP is about 15 feet from the AP but on top of a garage door opener.

Normally my wifi is a pair of Ubiquity AP's with an always connected local controler, with meshing enabled. both are hardwired.

**(Please, remember to close the issue when the problem has been addressed)**

JochenFriedrich, 2018-12-19:
I can confirm this. I had frequent WIFI disconnects with the newer core on 4 of my devices. 2 were a bit more stable, but still had sporadic disconnects. With the older core, all 6 are running stable again. I also use Ubiquity AP's on my site running the latest firmware and controller.

I also noticed that the devices did not disconnect, when running a "ping -f" against the ESP. This helped a lot for OTA "upgrade" to the older core and saved me from unmounting all my ESP based wall switches.

araczkowski, 2018-12-19:
I had the same with my devices, and I'm back to 2.3.0 too
the best WiFi is on 2.3.0, I wish to stay with 2.3.0 core forever but unfortunately I have a problem with WifiManager  reported here: #4659 unresolved and closed :(

@JochenFriedrich @cbennett6008 
are you able to have WifiManager on old coree 2.3.0?

spider7611, 2018-12-19:
Noticed this too on Sonoff Rf Bridge after update to 6.4.0.

ascillato2, 2018-12-19:
Hi,

Seems that your upgrade has left some incorrect values in your flash.

Please, erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/.

If you need further assistance with this, please find us in the Tasmota Chat. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

ascillato2, 2018-12-19:
@localhost61

Hi,

Every HomeAutomation software has aleady that built-in feature using the mqtt LWT message. You can check there and no need to configure anything. For example, in Home Assistant is just clicking on the name of the device an it will pop up a window with the previous states of the device including off-line.

localhost61, 2018-12-19:
OK, I use OpenHab, it's seems a bit more complicated :-), it seems we have to create our own rules.

JochenFriedrich, 2018-12-19:
Yes, the console view showed the reconnects.
A normal ping also stopped after 14s and restarted after 30s.
A flood ping (ping -f), however, maked the disconnects disappear.
The mosquitto log shows:

1545172813: New connection from 10.200.5.X on port 1883.
1545172813: New client connected from 10.200.5.X as DVES_85355D (c1, k10, u'malm2').
1545172828: Client DVES_85355D has exceeded timeout, disconnecting.
1545172828: Socket error on client DVES_85355D, disconnecting.
1545172837: New connection from 10.200.5.X on port 1883.
1545172837: New client connected from 10.200.5.X as DVES_85355D (c1, k10, u'malm2').
1545172853: Client DVES_85355D has exceeded timeout, disconnecting.
1545172853: Socket error on client DVES_85355D, disconnecting.

And so on.

Even a reset 5 and reconfig didn't help. Switching to a different SSID non either.

Jason2866, 2018-12-19:
It seems this disconnects only occours with Ubiquiti UniFi devices.

localhost61, 2018-12-19:
On my side I confess that I removed support of almost all sensors out of the one I use, that is DS18B20, AM2301 and SDM120, with no emulation, no KNX and so on. So I should have plenty of RAM.

andrethomas, 2018-12-19:
What I can say is that we did observe mqtt disconnects in cases where WiFi equipment is configured to automatically choose a wifi channel - This causes the 2.4.2 core to initiate an entire disconnect and re-connect causing mqtt to time out and reconnect.

The 2.3.0 core was less picky insofar that the TCP timeout was perhaps set to an unreasonably high value.

Jason2866, 2018-12-19:
There are some options that can be the reason for the disconnects with ubiquiti
https://dl.ubnt.com/guides/UniFi/UniFi_Controller_V5_UG.pdf
See Page 24-26!

Jason2866, 2018-12-19:
You can try latest core 2.5.0 beta. It seems stable
http://thehackbox.org/tasmota/020500/

ascillato, 2018-12-19:
If core 2.5.0 don't solve your issue, you can try sleep 0 to disable the sdk feature of light_modem_sleep.

cbennett6008, 2018-12-19:
flash_erase did not help.  (that's good news for those of us with devices mounted inside the wall. :) )
I enabled UAPSD on the SSID this is connecting to and I feel like it helped, but didn't solve it completely.
the 2.5.0 firmware has been clean without disconnects for 2 hours now, I'll continue to monitor

andrethomas, 2018-12-19:
Thanks for the feedback - It will be good news if it holds.

h-tro, 2018-12-19:
I had a similar problem.
It works for me now, see # 4688

salopette, 2018-12-19:
I also have problems with the connection with 6.4.0

ascillato, 2018-12-19:
Please, try the latest Tasmota v6.4.0 with core v2.5.0: http://thehackbox.org/tasmota/020500/

ascillato, 2018-12-19:
Hi,

For anyone having wifi problems, please try:

* Erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/.

For anyone having mqtt disconnections, please try:

* Erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/020500/.

For anyone having mqtt disconnections with core 2.5.0, please try in the console:

* `sleep 0`

andrethomas, 2018-12-19:
Just don't try sleep 0 on a sonoff basic R1 - you will have a new kind of problem to deal with :)

cbennett6008, 2018-12-19:
An OTA update to precompiled bins from: http://thehackbox.org/tasmota/020500/ resolved this for me, I'm at 6.5 hours without an MQTT disconnect now, thats' twice longer than previous longest stretch.

closed 2018-12-19 by cbennett6008

martikainen87, 2019-01-02:
I recently upgraded from 6.3.x to 6.4.0 and started getting error logs in my MQTT server, found this thread and upgraded with the precompiled bin posted above by cbennet6008. It didn't help so I used esptool.py and executed these commands and flashed them with the bin file from above (6.4.1.4

```
esptool.py --port COM6 erase_region 0x0F4000 0x008000
esptool.py --port COM6 erase_flash
esptool.py --port COM6 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
```

**MQTT Logs**

```
1546456412: Socket error on client balkong_9F067A7A, disconnecting.
1546456412: New connection from 192.168.1.41 on port 1883.
1546456412: New client connected from 192.168.1.41 as balkong_9F067A7A (c1, k10, u'mqtt').
```

**Logs in sonoff console**
```
20:13:24 MQT: balkong/tele/STATE = {"Time":"2019-01-02T20:13:24","Uptime":"0T08:55:17","Vcc":3.541,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Martikainen","BSSId":"FC:EC:DA:8A:71:7C","Channel":6,"RSSI":62}}
20:13:32 DNS: Initialized
20:13:32 MQT: Attempting connection...
20:13:32 MQT: Connected
20:13:32 MQT: balkong/tele/LWT = Online (retained)
20:13:32 MQT: balkong/cmnd/POWER = 
20:18:24 MQT: balkong/tele/STATE = {"Time":"2019-01-02T20:18:24","Uptime":"0T09:00:17","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Martikainen","BSSId":"FC:EC:DA:8A:71:7C","Channel":6,"RSSI":66}}

```

**Status 0**

```
20:41:28 CMD: status 0
20:41:28 MQT: mattias_sanglampa/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["mattias_sanglampa"],"Topic":"mattias_sanglampa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:31","StartupUTC":"2019-01-02T19:31:57","Sleep":50,"BootCount":7,"SaveCount":25,"SaveAddress":"FB000"}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(447ec62-sonoff)","BuildDateTime":"2019-01-01T15:14:58","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Martikainen",""],"TelePeriod":300,"SetOption":["400A8009","558180C0","00000200"]}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":456,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS5 = {"StatusNET":{"Hostname":"mattias_sanglampa-7811","IPAddress":"192.168.1.54","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:F8:9E:83","Webserver":2,"WifiConfig":4}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"mattias_sanglampa_%06X","MqttClient":"mattias_sanglampa_F89E8383","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 02 19:41:28 2019","Local":"Wed Jan 02 20:41:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-02T20:41:28"}}
20:41:28 MQT: mattias_sanglampa/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-02T20:41:28","Uptime":"0T00:09:31","Vcc":3.516,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Martikainen","BSSId":"78:8A:20:49:85:9B","Channel":11,"RSSI":64}}}
```

I did this for 8 Sonoff Basic's ( I've had these running since the beginning of tasmota) and 3 sonoff POW. I made sure to use unique Client ID's under the MQTT settings page by adding **%06X** to the end of the ID. Right now 3 basic's and 2 POW's are running without issues, the rest are having these disconnects.

I'm running 3 ubiquiti unifi access points, 3 and the sonoff's haven't changed AP as far as I can see (did a printscreen earlier today and they are all still on the same AP as before) All AP's are hardwired, sharing the same SSID's but on different channels. And according to the sonoff console log they are using the same channels all the time, and the channels are only active on one AP.

I'm using **setoption19 1** on all devices to get discovery to home assistant, and using **setoption30 1** on 6 basic's as they have lightbulbs connected.

I tried setting **sleep 0** on the POW i'm having issue with, that might have worked, enough time haven't passed yet to be sure.
But setting it on the basic model leaves the device unreachable (just like Andrethomas told us not to do :D )

ascillato, 2019-01-02:
Please, try https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

martikainen87, 2019-01-05:
> Please, try https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

reflashed my Basic's and sonoff SV's and the've been running without any errors for 2 days now. My POW's are still running the core 2.5.0 version without issues, but they needed to be configured with"sleep 0"