arendst/Tasmota #4688: Problem with Tasmota 6.4 after update with file upload

h-tro, 2018-12-19:
Yesterday I flashed two new Gosund Sp111s with Tasmota 6.4. Both work well with Tasmota 6.4
Today I flashed an existing Gosund SP111 with Tasmota 6.3 via file upload menu first with Tasmota minimal.bin version 6.4 and then to Tasmota 6.4.
Thereafter, the Web-Gui works only delayed. It seems to be a wifi problem (see log file).

The same problem occurs with my Shelly1.

When I go back to version 6.3 via file upload everything is OK again.

[Log.txt](https://github.com/arendst/Sonoff-Tasmota/files/2695090/Log.txt)

ascillato2, 2018-12-19:
Hi Sorry. The log does not inform any wifi issue.

Seems that your upgrade has left some incorrect values in your flash.

Please, erase all flash using esptool.py and flash again by wire (as explained in the wiki)

If you need further assistance with this, please find us in the Tasmota Chat. Thanks.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-19 by ascillato2

h-tro, 2018-12-19:
Hello,
my log shows 2 of many wifi- connects and disconnects.
In the log of my access point are also many connects and disconnects with tasmota 6.4 to see.
When I go back to Tasmota 6.3, the wifi connection is stable again.

ascillato, 2018-12-19:
ok,

So please, erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/.

h-tro, 2018-12-19:
A normal software restart is not enough.
After I have disconnected **all** devices (Gosund SP111 and Shelly1) from the power grid after OTA update (bin-file upload), Tasmota 6.4 works normally (no wifi-disconnects).