arendst/Tasmota #4694: Sonoff Basic new layout auto turn on\off with gpio2 connected

itMax86, 2018-12-20:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
This happen only if GPIO2 is configured as switch.
In console you can see a quickly on\off command and you can hear relay do it
This happen with light turned on or off

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : SonOff Basic new layout 
- [X] _Development/Compiler/Upload tools used_ : OTA
- [X] _If a pre-compiled release or development binary was used, which one?_ : upload last fw via ota from http://thehackbox.org, binary 2.4.2
- [X] _You have tried latest release or development binaries?_ : YES
- [X] _Provide the output of command_``status 0`` :
```
08:58:30 MQT: stat/LuceTavolo/STATUS = {"Status":{"Module":1,"FriendlyName":["LuceTavolo"],"Topic":"LuceTavolo","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:58:30 MQT: stat/LuceTavolo/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:30","StartupUTC":"2018-12-20T07:52:00","Sleep":50,"BootCount":58,"SaveCount":358,"SaveAddress":"F7000"}}
08:58:30 MQT: stat/LuceTavolo/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:55:52","Boot":7,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
08:58:30 MQT: stat/LuceTavolo/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wi-kurto-fi",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
08:58:30 MQT: stat/LuceTavolo/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000410","0FDAE794","000383A0","23B617CE","00003BC0"]}}
08:58:30 MQT: stat/LuceTavolo/STATUS5 = {"StatusNET":{"Hostname":"LuceTavolo-7516","IPAddress":"192.168.1.113","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:DF:5D:5C","Webserver":2,"WifiConfig":5}}
08:58:30 MQT: stat/LuceTavolo/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.230","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DF5D5C","MqttUser":"itMax86","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:58:30 MQT: stat/LuceTavolo/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 20 07:58:30 2018","Local":"Thu Dec 20 08:58:30 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:55"}}
08:58:30 MQT: stat/LuceTavolo/STATUS10 = {"StatusSNS":{"Time":"2018-12-20T08:58:30","Switch1":"OFF"}}
08:58:30 MQT: stat/LuceTavolo/STATUS11 = {"StatusSTS":{"Time":"2018-12-20T08:58:30","Uptime":"0T00:06:30","Vcc":3.533,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"wi-kurto-fi","BSSId":"FC:EC:DA:11:54:DC","Channel":11,"RSSI":60}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Use a new Sonoff basic, configure gpio2 as switch, connect a switch

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-20:
This is not a basic specific issue. If you wire extern switches you have to take care. See
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

ascillato, 2018-12-20:
Have you added a pull up resistor?

ascillato2, 2018-12-20:
Have you configured as switch1 or switch1n?

itMax86, 2018-12-20:
normal switch, configured as switch1

no pullup resistor, only switch connected to gpio2 and gnd
switch connected with about 15cm of wire

ascillato2, 2018-12-20:
Hi,

From https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs Gpio2 is:

GPIO2 | Boot mode select TX1 | 3.3V (boot only) | Don’t connect to ground at boot time Sends debug data at boot time
-- | -- | -- | --

So, if you have that gpio connected to GND, could make your device to hung due to short circuit.

Please, check GPIO Hardware restrictions here: https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#restrictions

Also, if that 15cm of wire is very close to AC wires, it will introduce noise into your ESP8266. Please, search in issues that the ghost switching issue has been addressed several times. Several users had reported different types of solutions to that.

eric-void, 2018-12-21:
Just for confirmation.

New sonoff basic has only GPIO2 pin available to connect external switches, right?
But, as stated in https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#restrictions, "GPIO2 can’t be low at boot, so you can’t connect a switch to it".

So: new sonoff basic can't have an external switch connected to its available pins. Right?

itMax86, 2018-12-21:
for now the safety system is connect a button in parallel to integrated button!

ascillato, 2018-12-21:
Using a button on gpio0 worked for you? You did not have ghost switching with that pin?

If so, it means that you need a pull up resistor for gpio2. Gpio0 has a pull up resistor.

wongnam, 2018-12-22:
@itMax86 How can use the switch on GPIO0, in case the switch is closed, how to restart or boot the sonoff successfully.
GPIO0 can only use with Button, not Switch.

ascillato2, 2018-12-22:
@itMax86

Hi, 

When you have time, please tell us if a pull up on gpio2 solves your issue. Thanks.

closed 2018-12-22 by ascillato2

mprins, 2018-12-29:
posting the image here for reference
[image]

ascillato, 2019-01-13:
Sorry, there isn't. It is a hardware limitation.

seritos, 2019-01-13:
Ok, Thanx for your fast reply.