arendst/Tasmota #4697: Wifi issues after upgrade to 6.4.0

xNuno, 2018-12-20:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

After the upgrade to 6.4.0 I see a lot of wifi connection drops...I have around 20 sonoff sw arround the house and they lost connection randomly... I did not see this happen with 6.3.0 so maybe something changed... they have more than 70% of signal most of the cases...

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : NodeMCU ESP8266 ESP-12E for arduino CP2102
- [ ] _Development/Compiler/Upload tools used_ :
- [ X ] _If a pre-compiled release or development binary was used, which one?_ : Latest from this repository
- [ X ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS = {"Status":{"Module":28,"FriendlyName":["Sonoff"],"Topic":"sonoff_sw_bedroom_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.182:80/data/firmwares/sonoff.bin","RestartReason":"Power on","Uptime":"1T20:05:01","StartupUTC":"2018-12-18T22:17:25","Sleep":50,"BootCount":53,"SaveCount":692,"SaveAddress":"F7000"}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SilvaN",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000001"]}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_sw_bedroom_1-4770","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:A1:B2:A2","Webserver":2,"WifiConfig":2}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A1B2A2","MqttUser":"SilvaN","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 20 18:22:26 2018","Local":"Thu Dec 20 19:22:26 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:55"}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS10 = {"StatusSNS":{"Time":"2018-12-20T19:22:26"}}
19:22:26 MQT: stat/sonoff_sw_bedroom_1/STATUS11 = {"StatusSTS":{"Time":"2018-12-20T19:22:26","Uptime":"1T20:05:01","Vcc":3.430,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SilvaN","BSSId":"50:64:2B:92:C6:1D","Channel":8,"RSSI":88}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_

Upgrade from 6.3.0 to 6.4.0

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

[image]

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

cHunter789, 2018-12-20:
I see the same issue with my sonoff basic. Wemos and Sonoff S26 are working fine. Strange.

xNuno, 2018-12-20:
Yep...mine are Sonoff T1 1CH and Sonoff T1 2CH...the S20 I have didn't saw the issue as well but they are not used much ...

dolenec, 2018-12-20:
Same for Sonoff TH10, TH16 and POW2..
After restoring back to 6.3.0 problems gone..

mjoshd, 2018-12-20:
I am also having the issue on multiple devices after upgrading to 6.4.0... Sonoff Basic, S31, iFan02, Wemos Mini all went into reconnect loops as shown in the OP Screenshot. After reverting to 6.3.0 the problem disappears. Happy to help in any way I can... just let me know what info is needed.

xNuno, 2018-12-20:
All my devices with http://thehackbox.org/tasmota/020500/

Let's see how it goes...

[image]

xNuno, 2018-12-20:
This it's the same behaviour...

[image]

cHunter789, 2018-12-20:
For me 6.4.0.1 is the same issue (with sonoff basic). Back to 6.3

xNuno, 2018-12-20:
Back to 6.3.0 as well until further news...

ascillato2, 2018-12-20:
Hi,

This issue is duplicated to https://github.com/arendst/Sonoff-Tasmota/issues/4686

Please, try the following as explained in that issue:

------------------------

For anyone having wifi problems, please try:

* Erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/.

For anyone having mqtt disconnections, please try:

* Erase all flash using esptool.py and flash again by wire (as explained in the wiki) using the precompiled bins from http://thehackbox.org/tasmota/020500/.

* For anyone having mqtt disconnections with core 2.5.0, please try in the console:

`sleep 0`

If any of the previous solutions don't work in your network please use Tasmota v6.4.0 with core 2.3.0:
http://thehackbox.org/tasmota/release/020300/

----------------

Thanks

ascillato2, 2018-12-20:
@xNuno 

Please, try the previous solutions. Thanks.

casperklein, 2018-12-21:
"sleep 0" fixed it for me. Tested with Sonoff S20 and Basic.

ascillato, 2018-12-21:
More **Wifi Troubleshooting** and **Explanations** can be found in the **wiki** at: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

wongnam, 2018-12-21:
All my Sonoff devices is working properly with latest TASMOTA FW with core 2.4.2 now.

I'd like to share my case study.
* Symptom: last time my Sonoff device was running FW ver6.3.0.17-**core 2.3.0**  i fashed them to version 6.3.17-core2.4.2 over OTA that 1/3 of my Sonoff device cannot reach by webserver page.
* Solution: Flash them by Serial interface to version 6.3.17-core2.4.2 to bring them to work. after that my Sonoff devices can use OTA flash them to any version  with **core 2.4.2** without any issue.

My Sonoff devices include: Sonoff_Basic; Sonoff Touch-US; Sonoff T-1US-2C, 3C.  Sonoff Dual R2; Sonoff Bridge; Sonoff S22.

mrbgfromv, 2018-12-21:
Compiled tasmota 6.4.0.1 with core 2.5.0, sleep 50:
No disconnects so far.

mjoshd, 2018-12-21:
After some trial and error I was able to finally update all of my devices to the official 6.4.0 (2.4.2 core) github release without uninstalling, disassembling or connecting via wire. A few of my Sonoff devices are installed in places that would have required a substantial effort to remove and/or disassemble.

I tried using various firmware versions from thehackbox but the only one which didn't exhibit the issue was the version compiled against core 2.3.0 and I really wanted to continue using the official FW from the github release page.

The method I used to mitigate the issue takes advantage of the fact that many things get erased when you change the CFG_HOLDER variable.

1. I downloaded a copy of "Sonoff-Tasmota-master.zip" and opened it in PlatformIO
2. I compiled two versions of "sonoff-minimal", both with my WiFi credentials entered
     a. one version with the default CFG_HOLDER     e.g. "original+wifi.bin"
     b. one version with a modified CFG_HOLDER     e.g. "modified+wifi.bin"
3. I used the web GUI to create configuration back-ups of each affected device 
4. I flashed each one in the following order
     a. stock "sonoff-minimal.bin" from [version 6.3.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v6.3.0)
     b. my "modified+wifi.bin"
     c. my "original+wifi.bin"
     d. stock "sonoff.bin" from [version 6.4.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v6.4.0)
5. I then restored the backed up configuration to each device and 🎉 viola! 🎉 no more disconnects, no config/setoption changes, and still using officially released firmware 😃 

Hope this helps some folks.

andrethomas, 2018-12-21:
@mjoshd Good job, and thanks for documenting your process!

spider7611, 2018-12-22:
I disassembly and connected with wire and flashed with esptool. Setting it up and working a while, but I noticed the wifi is very poor 8%. Another sonoff basic with 6.3 tasmota 2.3.0 core same far from my router having 78%. So later with the new firmware I can't reach this sonoff basic. 😕

KoLSMS, 2018-12-22:
> All my devices with http://thehackbox.org/tasmota/020500/
> 
> Let's see how it goes...
> 
> [image]

Hello,

how have you generated this overview site of your Sonoffs???

xNuno, 2018-12-22:
It's a plug-in for hassio. Maybe you can have it as standalone.

mcc05, 2018-12-22:
I've just upgraded to  6.4.0.2, seem to be much better on the wifi front

narfel, 2018-12-22:
@xNuno you can with apache/php7: https://github.com/reloxx13/TasmoAdmin

Maciej2, 2018-12-22:
Hi, after I flashed my Sonoff T2 first time to version 6.4.0 and core 2.3.0 i could upgrade to version 6.4.0 with core 2.4.2
I have another problem with the new firmware-with the previous version i was able to hide SSID of my WiFi network but after flash the device looses connection strait after hiding WiFi SSID and switches itself to Wifi Manager mode

Jason2866, 2018-12-22:
What is the goal you want to reach hiding SSID?
To make it more secure? Haha That is snake oil. Every script kiddie can find a hidded SSID....

ascillato, 2018-12-22:
@xNuno

Have you managed to solve your wifi issue?
Have you tried the previous solutions?
Thanks.

ascillato2, 2018-12-22:
For anyone having wifi or mqtt issues, please check the wifi troubleshooting on the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Thanks everyone for all the reports and testings.

closed 2018-12-22 by ascillato2

xNuno, 2018-12-22:
@ascillato2, not solved. I'm on 6.3.0 without issues. In my opinion remove 20 sw from wall and open 3 s20 plugs to flash directly is not a good solution... We should focus on solve this other way.
Let me know if I can help more.

andrethomas, 2018-12-22:
@xNuno Just a thought - 6.3.0 can run a little hot - have you tried setting sleep to 50 and leave it for 10 minutes or so and then try OTA to 6.4.0? We had this problem with the previous versions of Sonoff Basic's wherein the terminal dissipation of the basic was not good and the extra heat generated by the ESP8266/ESP8285 caused some wifi problems.

Just a shot in the dark because 6.3.0.10 and higher solved the thermal issues for the Sonoff Basic modules - I am not sure how the S20 deals with the heat issues we found in earlier firmware issues as I do not have an S2x to test with.

xNuno, 2018-12-22:
Hi @andrethomas , I can try if you say that's not good stay with the current 6.3.0...

Will give feedback later.

Jason2866, 2018-12-22:
The S20 doesnt have a heat problem. Version 6.3.0 will run without a problem

xNuno, 2018-12-22:
So far no reboots...

[image]

JF62, 2018-12-23:
6.4.0.3 with 2.5 is the only Version wich works fine for me in a Mesh / multi Accesspoints Environment. All others are heavily unresponsive.

jupzup, 2018-12-29:
I had the same problems with 6.4.1.1 multiple issues like wifi & mqtt disconnects on several hardware ( SX20, Basic, Dual, Blitzwolf and also Shelly 1 ). Now im back to 6.3.0.10 and everything is ok.

mcc05, 2018-12-29:
Yep I have same issues on multiple sonoff basics. I think if off back to 6.3.0.10 also...

andrethomas, 2018-12-29:
2.4.2 seems a little picky on some network setups - 2.5.0 seems more forgiving...

cHunter789, 2018-12-30:
Bad news, looks like it's a problem with some errors in flash (like ascillato wrote). For me, after flash erase and flashing with 6.4.1_2.4.2 by serial console, sonoff basic start working OK for at least 12h...

IxsharpxI, 2018-12-30:
i was able to fix it on mine by using 2.5.0 core bin files. But that’s interesting maybe ill try the original again although you think at least one of the 3 would have flashed right

andrethomas, 2018-12-30:
@IxsharpxI bad flashes via serial is quite possible with some usb serial boards - we have seen this with ones which do not provide sufficient current @ 3.3V - I myself had this problem a number of times and consequently modified by CP2102 board to provide 3.3V via a LD1117-3.3 800mA regulator from the 5V line and out on the same 3.3V pin provided by the CP2012 module.

[image]

The same solution may work with others also - most of the usb serial converters cannot deliver the 200mA+ spikes demanded by the ESP8266 so its possible that this may cause bad data to be written to the flash (since it is serial data)

I have not tried but just adding a 47uF capacitor between the 3.3V and ground pins of the usb serial adapter may also improve results.

hhaim, 2019-02-17:
I’ve tried 6.4.1.17 and only with 2.3.0 core the WiFi works.

nhondong, 2019-02-26:
Same Problem here after upgrade to 6.4.1
If i ping the device from my pc, the WIFI connection is stable. 
just a few minutes (~5 min) after i stopped pinging, the device disconnect and reconnect to WLAN.
Even devices nearby the access point. So this is not a interference problem.
with 5.11 it has worked without any problems.