arendst/Tasmota #4699: HASS autodiscovery pick incorrect state topic

Gaff, 2018-12-20:
HASS autodiscovery pick incorrect state topic

Specifically for the `state_topic` it's picking `.../tele/STATE`. This topic does not get republished on a state change. It used to pick `.../stat/RESULT` which does get republished.

This looks like an unintentional error, see here:

https://github.com/arendst/Sonoff-Tasmota/blame/62c3dfa27b954946dc3a3999c6efbbea1ec5e9ae/sonoff/xdrv_12_home_assistant.ino#L254

Gaff, 2018-12-20:
Though I'm not sure I understand fully - `.../stat/RESULT` can contain other things (e.g. the result of any console command). Perhaps `.../stat/POWER` would be the right thing to pick? Would need to be de-jsonify the config (which I think is ok?)

mhuca, 2018-12-20:
@Gaff 

There has been significant updates to the home assistant integration. One of the changes was enforcing %topic%/%prefix%
If you don't us the home assistant auto discovery you can disable this with
SetOption19 0
SetOption55 0
SetOption59 0
restart 1
more information can be found in the wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Gaff, 2018-12-20:
Ok - I see that `%prefix%/%topic%/` is not compatible with the abbreviations / base topic (option55).

However if I've set MQTT to use `%prefix%/%topic%/` then there's no point in sending `%topic%/%prefix%/` to HASS - it just leaves the users very confused 😢 

Perhaps we could enforce `%topic%/%prefix%/` for anyone who enables option19? Or make it default? Or make option19 fail if it's not set thusly? Anything is better than where it is now!

mhuca, 2018-12-20:
if you are not using discovery then SetOption19 0 will clear the retain message to enable hass discovery.

SetOption59 0 will disable Send tele/STATE in addition to stat/RESULT for commands: State, Power and any command causing a light to be turned on.

Gaff, 2018-12-20:
your comment should go on my other issue, this is just for the `state_topic` issue :)

mhuca, 2018-12-20:
@Gaff 
I'm using the HASS auto discovery and the changes were a pain to have to go back and update to handle the changes in topic. However, these new changes have made adding new devices to HASS extremely easy.

I now have 50+ devices with Tasmota in HASS now with this configuration... its well worth the transition.

Gaff, 2018-12-20:
Just to be clear I want HASS discovery! I don't particularly care about `%profile%/%topic%/` vs `%topic%/%profile%/`. However I would like it to "just work" - or at least fail gracefully.

Right now it sends a meaningless config to HASS. That can't be right.

araczkowski, 2018-12-21:
Hi @ascillato2 @Gaff @mhuca 

I think that this misunderstanding came from the description, in:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

if the GroupTopic is **sonoffs** (by default)
than to automatically discover the device we should now publish the SetOption19 1 to the topic:
> **sonoffs**/cmnd/SetOption19 

but in the description we have the samples from the past:
> cmnd/**sonoffs**/SetOption19 

it was a puzzle for me too, another one ware why I can't change the "FullTopic" 
now when I read this topic the concept is more clear

Gaff, 2018-12-21:
Ok I tried this on a fresh device and I can't reproduce - sorry. I think my other device must have had some corrupted settings or something :(

It is a little surprising that the MQTT topic changes from prefix/topic to topic/prefix when you switch option19 on, but it does just work so most people won't see it.

closed 2018-12-21 by Gaff