arendst/Tasmota #4700: HASS Discovery doesn't respect MQTT topic

Gaff, 2018-12-20:
I've set my MQTT Topic to "sonoff-pow1234" and my Full topic to `%prefix%/%topic%/`

However the HASS relay discovery messsage has things like:

    "command_topic":"sonoff-pow1234/cmnd/POWER"

It should be:

   "command_topic":"/cmnd/sonoff-pow1234/POWER"

mhuca, 2018-12-20:
@Gaff 

There has been significant updates to the home assistant integration. One of the changes was enforcing %topic%/%prefix%
If you don't us the home assistant auto discovery you can disable this with
SetOption19 0
SetOption55 0
SetOption59 0
restart 1
more information can be found in the wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Gaff, 2018-12-20:
Ok - I see that `%prefix%/%topic%/` is not compatible with the abbreviations / base topic (option55).

However if I've set MQTT to use `%prefix%/%topic%/` then there's no point in sending `%topic%/%prefix%/` to HASS - it just leaves the users very confused 😢 

Perhaps we could enforce `%topic%/%prefix%/` for anyone who enables option19? Or make it default? Or make option19 fail if it's not set thusly? Anything is better than where it is now!

ascillato2, 2018-12-20:
@Gaff

Let's gather all your problems with Hass Autodiscovery in only one issue https://github.com/arendst/Sonoff-Tasmota/issues/4699

closed 2018-12-20 by ascillato2