arendst/Tasmota #4702: Cannot disable HASS discovery on SonoffPow (option19)

Gaff, 2018-12-21:
**Describe the bug**

Cannot disable HASS discovery (option19) on SonoffPow R2 

**To Reproduce**

1. Configure module to be SonoffPow R2
2. Setoption19 1
3. HASS announces as expected
4. Setoption19 0

The device will reboot, but when it comes back up option19 will still be enabled.

closed 2018-12-21 by ascillato2

ascillato, 2018-12-21:
Hi, 

@reey

have you tried the proposed solution?

ascillato, 2018-12-21:
Please, explain reoccurs. You erase all flash then you flash latest Tasmota. Then you do a setoption19 0, it works and then you do again and don't work?

reey, 2018-12-21:
Yeah, so I had some trouble yesterday and wanted to do a setoption19 0 that didn' work out, so I erased the flash and flashed the binary from development branch again.
Today I wanted to do a setoption19 1 and it doesn't work..

ascillato, 2018-12-21:
Hehe, the "it doesn't work" expression is too much general. Please, give us more precise info. What that means? What do you see in the console? Your sonoff reboots? Etc. Etc.

I just want to help you troubleshoot your problem. In my devices it works just fine with latest Tasmota v6.4.0.2 and Hassio 0.84.5

reey, 2018-12-21:
The console is not replying. The device reboots and I'm getting the output of a normal boot..
But if I just do a SetOption19 I'm seeing that the option wasn't changed..

I'm also running Hassio 0.84.5.

ascillato, 2018-12-21:
Ok. Your device is crashing before saving that option. So, that means that you have a firmware issue. Please, can you erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from development branch?

reey, 2018-12-21:
So I don't know why, but for some reason the SetOption19 1 has now worked..
I just repeatedly entered it and after more than 10 times it worked..
But I know have the same problem when I do a SetOption19 0..