arendst/Tasmota #4704: some compiler warnings show some stupid code

Mondeo20, 2018-12-21:
Hi,

I compile the source and it give me warnings:

...arduino/sonoff/support_wifi.ino:252:67: warning: statement has no effect [-Wunused-value]

if (best_network_db < -WIFI_RSSI_THRESHOLD) { best_network_db +WIFI_RSSI_THRESHOLD; }

Which is true, makes no sense. Can somebody saying if this line is needed and if yes what is the correct line.

Here the next code which makes no sense:

xdrv_19_ps16dz_dimmer.ino:156:82:

if (ps16dz_byte_counter || (!ps16dz_byte_counter && serial_in_byte == 'A'));
ps16dz_rx_buffer[ps16dz_byte_counter++] = serial_in_byte;

And again, from my point of view stupid code: xsns_05_ds18x20.ino:264:68:

for (ds18x20_sensors = 0; ds18x20_sensors < DS18X20_MAX_SENSORS; ds18x20_sensors) {

There a lot of more warning mainly caused by data type conversions and some more where possible somebody has to lock more in details.

Mondeo20, 2018-12-21:
what is about the 2 other code lines (in xdrv_19_ps16dz_dimmer.ino and xsns_05_ds18x20.ino)?

arendst, 2018-12-21:
The perform as they should.