arendst/Tasmota #4705: mysterious code

Mondeo20, 2018-12-21:
Sorry if I'm back again, but I asked for 3 code mysteries which one is solved now. The other are still open from my point of view:

1.   in  xdrv_19_ps16dz_dimmer.ino
if (ps16dz_byte_counter || (!ps16dz_byte_counter && serial_in_byte == 'A')); <------ 
ps16dz_rx_buffer[ps16dz_byte_counter++] = serial_in_byte;

The semicolon after the id statement lead to the situation that the if statement will never used means the next line will always executed. If this is true I would propose to comment out the if statement to have clean code!

2. in xsns_05_ds18x20.ino
for (ds18x20_sensors = 0; ds18x20_sensors < DS18X20_MAX_SENSORS; ds18x20_sensors) {
                                                                                                                                              ^----------
Loop counter are never changed which can led to a endless loop!

arendst, 2018-12-21:
I'm not the owner of the ps16dz dimmer driver but he may think you're right ;-)

Regarding xsns_05_ds18x20 I was wondering too until you start to search for somethin like
```
      ds18x20_sensors++;
```
which happen to be present at line 278.

The break at 266 will stop the loop.

ascillato, 2018-12-21:
Please, use the Tasmota Support Chat for general questions. Github issues are for bug reporting. Thanks

closed 2018-12-21 by ascillato2

chaosmaster, 2018-12-23:
@Mondeo20 
> 1. in  xdrv_19_ps16dz_dimmer.ino
>        if (ps16dz_byte_counter || (!ps16dz_byte_counter && serial_in_byte == 'A')); <------
>        ps16dz_rx_buffer[ps16dz_byte_counter++] = serial_in_byte;
>
>
> The [...]

Good catch!
That statement is supposed to make sure commands always start with "A" 
The PS-16-DZ uses "AT+"-style commands.
I created a [PR](#4720) with a fix.