arendst/Tasmota #4708: HomeAssistant Discovery incomplete in lengthy mode

joelhaasnoot, 2018-12-22:
**Describe the bug**
_A clear and concise description of what the bug is._
When running Home Assistant auto discovery turned on (SetOption19 1) without 'abbreviations and base topic' turned on (SetOption55 0), discovery on Home Assistant works, but fails to create editable entities in Home Assistant. Looking at the 'announce' body, several keys are missing in the lengthy mode, most notably _unique_id_. Without unique_id an entity in the Home Assistant registry isn't created, and while the device appears, it's not possible to edit the entity_id in Home Assistant. 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic / S20
- [x] _Development/Compiler/Upload tools used_ : Various, both OTA updates and VS Code with Platform IO
- [x] _If a pre-compiled release or development binary was used, which one?_ : 
- [x] _You have tried latest release or development binaries?_ : Yes, latest version
- [x] _Provide the output of command_``status 0`` :
```
02:05:32 MQT: sonoff-5/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff-5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:05:32 MQT: sonoff-5/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T03:44:27","StartupUTC":"2018-12-21T21:21:05","Sleep":50,"BootCount":6,"SaveCount":35,"SaveAddress":"F9000"}}
02:05:32 MQT: sonoff-5/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-20T22:18:06","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
02:05:32 MQT: sonoff-5/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TheInterwebs",""],"TelePeriod":300,"SetOption":["400A8009","558180C0","00000220"]}}
02:05:32 MQT: sonoff-5/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","240383A0","23B617CE","00003BC0"]}}
02:05:32 MQT: sonoff-5/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5-1319","IPAddress":"192.168.1.30","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:15:xx:xx","Webserver":2,"WifiConfig":4}}
02:05:32 MQT: sonoff-5/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.80","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_152527","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:05:32 MQT: sonoff-5/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 22 01:05:32 2018","Local":"Sat Dec 22 02:05:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
02:05:32 MQT: sonoff-5/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-22T02:05:32"}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
- Enable HomeAssistant Discovery in Console: SetOption19 1

**Expected behavior**
_A clear and concise description of what you expected to happen._
- Device is created in Home Assistant with a unique_id so that the entity can a) be edited (wiki mentions the cog/wheel icon) and b) show sup under Integrations -> MQTT -> Devices. Discovery should also work without the SetOption55 on (unless there are hardware/size limitations we're running into making this impossible ofcourse, but would be happy to hear so)

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
Comparing the two context bodies without and with SetOption55 enabled:
_Normal_ (SetOption55 0)
````
{"name":"Sonoff","command_topic":"sonoff-5/cmnd/POWER","state_topic":"sonoff-5/tele/STATE","value_template":"value_json.POWER","payload_off":"OFF","payload_on":"ON","availability_topic":"sonoff-5/tele/LWT","payload_available":"Online","payload_not_available":"Offline"}
````
Concise (SetOption55 1)
````
{"name":"Sonoff","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"152527_LI_1","device":{"identifiers":["152527"],"name":"Sonoff","model":"Sonoff Basic","sw_version":"6.4.0(sonoff)","manufacturer":"Tasmota"}, "~":"sonoff-5/"}
````

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-22:
Hi,

I can reproduce your finding. Thanks for reporting. 

@emontnemery @blakadder any ideas on this? This can be changed or only with setoption55 1 will be available the unique id? If so, we can add this restriction to the wiki. Thanks

emontnemery, 2018-12-22:
This is not a bug and it's already stated on the wiki that `setoption55 1` is necessary for integration with home assistant.

@arendst suggested to make `setoption55 1` the default in January, so let's do it soon.

joelhaasnoot, 2018-12-22:
Thanks for the quick responses and feedback, didn't realize there was a max length. Making SetOption55 the default makes a lot of sense. (currently actually the condensed is longer than the original :) )

One other point related to this is that SetOption55 1 doesn't resend the discovery message, while SetOption 19 does. This means that the tip "Tip: Make all Tasmota devices be automatically discovered" as listed on the wiki doesn't work. See this console listing, the updated config message isn't sent till after you reboot the device or run SetOption19 again. At minimum this is a documentation defect, but flipping the order of the options should work I think.
````
10:43:06 CMD: SetOption19 1
10:43:06 MQT: homeassistant/switch/152527_RL_1/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_1/config = {"name":"Sonoff","command_topic":"sonoff-5/cmnd/POWER","state_topic":"sonoff-5/tele/STATE","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"sonoff-5/tele/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_2/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_2/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_3/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_3/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_4/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_4/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_5/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_5/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_6/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_6/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_7/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_7/config =  (retained)
10:43:06 MQT: homeassistant/switch/152527_RL_8/config =  (retained)
10:43:06 MQT: homeassistant/light/152527_LI_8/config =  (retained)
10:43:06 MQT: sonoff-5/stat/RESULT = {"SetOption19":"ON"}
10:43:12 CMD: SetOption55 1
10:43:12 MQT: sonoff-5/stat/RESULT = {"SetOption55":"ON"}
````

blakadder, 2018-12-22:
I would recommend making SetOption55 1 default from now as well. Having it manually entered is only a hindrance and I don't see a reason for it to be deactivated since its supported from hass 0.81 and its on 0.84 now, going 0.85 early january.

arendst, 2018-12-22:
@emontnemery Considering the amount of happy HASS users wanting SetOption55 set I think the easiest way to go forward is you make a PR any time soon without SetOption55 and just assume SetOption55 is set.

If users then encounter problems all we have to say is update HASS to 0.81 and up ;-)

emontnemery, 2018-12-22:
@arendst I read sloppily. You actually mean to remove SetOption55 option?