arendst/Tasmota #4714: Please add REST base report for SonOff POW measurements

halacs, 2018-12-22:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
I already have a PHP script what collects measurements from other sources via REST. It would be nice to integrate my boiler controlled by a SonOff POW. I don't want to add a new component to my system just because of this (MQTT). REST report would ensure that an attacker cannot modify my Sonoff from the Internet.

**Describe the solution you'd like**
POW would periodically send its measurements as it does in case of MQTT but via REST. Web UI would have a configurable URL to define where to POST the measurements in json format.

**Describe alternatives you've considered**
Using MQTT as transport protocol then use mqttwarn to forward the received information into the database what the REST use. This solution would need extra effort in the operation.

Thanks!

Jason2866, 2018-12-22:
Tasmota is designed for use as a mqtt device. Web frontend isnt the core goal. Just a nice to have
Do to the fact of the limited resources of the device your wish will probably not implemented from Theo.

arendst, 2018-12-22:
Just pull the info using HTTP. See wiki commands how to.

closed 2018-12-22 by arendst

halacs, 2018-12-22:
Then would it be acceptable to add a "freeze" MQTT command somehow? The goal of this would be to make impossible to control the device via MQTT. Device would be able to send telemetry data but receive any command.

ascillato, 2018-12-22:
You can use rules, events and websend command. Search in issues. This was already addressed. Thanks

arendst, 2018-12-22:
You can turn off MQTT on the fly. See wiki commands.