arendst/Tasmota #4717: rules timer not executing

shanekuz, 2018-12-23:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-23:
You hve to check. `"ARM"`
[image]

ascillato2, 2018-12-23:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-23 by ascillato2

manikdutta, 2019-02-12:
Rules on timer not working..
[image]

ascillato, 2019-02-12:
Hi,

Please, stop spamming issues. You have an answer in the issue you have opened.

shanekuz, 2019-02-12:
I don’t know what you mean I have not spammed anything I raised an issue and have not done anything since

ascillato, 2019-02-12:
@shanekuz

Sorry, that was not meant for you.

Was for @manikdutta that he/she is posting his/her issue repeteadly in several issues.