arendst/Tasmota #4721: Tasmota 6.4.0.3 2.5 Huge Problem with Sonoff read ECONNRESET

ymurawski, 2018-12-23:
Hi guys, today i flashed all of my devices (Sonoff Touch and Basic) to 6.4.0.3 with Core 2.5 and i have now huge problems. Every device is now restarting random time. 
i already tried to flash back to 6.3 with core 2.3 but still the same problem. 

My log file of iobroker shows:

sonoff.0 | 2018-12-23 21:24:59.615 | info | Client [Sonoff-Esszimmer-1-1ch] connected with secret 1545596699615_9052
-- | -- | -- | --
sonoff.0 | 2018-12-23 21:24:59.357 | info | Client [Sonoff-Esszimmer-1-1ch] connection closed: closed
sonoff.0 | 2018-12-23 21:24:51.800 | info | Client [Sonoff-Gaestebad-2ch] reconnected. Old secret 1545596605971_3441. New secret 1545596691800_6886
sonoff.0 | 2018-12-23 21:24:51.492 | info | Client [Sonoff-Kueche-3-3ch] connected with secret 1545596691492_7251
sonoff.0 | 2018-12-23 21:24:51.483 | info | Client [Sonoff-Kueche-3-3ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:24:47.790 | info | Client [Sonoff-Flur-1-1ch] connected with secret 1545596687789_8489
sonoff.0 | 2018-12-23 21:24:47.026 | info | Client [Sonoff-Flur-1-1ch] connection closed: closed
sonoff.0 | 2018-12-23 21:24:37.149 | info | Client [Sonoff-Wohnzimmer-1-2ch] connected with secret 1545596677147_7673
sonoff.0 | 2018-12-23 21:24:37.142 | info | Client [Sonoff-Wohnzimmer-1-2ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:24:35.300 | info | Client [Sonoff-Schlafzimmer-2ch] connected with secret 1545596675298_5913
sonoff.0 | 2018-12-23 21:24:35.294 | info | Client [Sonoff-Schlafzimmer-2ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:29.133 | info | Client [Sonoff-Flur-1-1ch] connected with secret 1545596609133_7535
sonoff.0 | 2018-12-23 21:23:29.128 | info | Client [Sonoff-Flur-1-1ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:28.563 | info | Client [Sonoff-Esszimmer-1-1ch] connected with secret 1545596608562_4589
sonoff.0 | 2018-12-23 21:23:28.558 | info | Client [Sonoff-Esszimmer-1-1ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:25.971 | info | Client [Sonoff-Gaestebad-2ch] connected with secret 1545596605971_3441
sonoff.0 | 2018-12-23 21:23:25.705 | info | Client [Sonoff-Kinderzimmer-1ch] reconnected. Old secret 1545596528639_1790. New secret 1545596605705_5740
sonoff.0 | 2018-12-23 21:23:25.564 | info | Client [Sonoff-Hauswirtschaftsraum-1ch] connected with secret 1545596605563_5007
sonoff.0 | 2018-12-23 21:23:25.557 | info | Client [Sonoff-Hauswirtschaftsraum-1ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:25.266 | info | Client [Sonoff-Badezimmer-1ch] connected with secret 1545596605261_8060
sonoff.0 | 2018-12-23 21:23:25.255 | info | Client [Sonoff-Badezimmer-1ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:20.305 | info | Client [Sonoff-Kueche-3-3ch] connected with secret 1545596600304_8340
sonoff.0 | 2018-12-23 21:23:20.293 | info | Client [Sonoff-Kueche-3-3ch] connection closed: Error: read ECONNRESET
sonoff.0 | 2018-12-23 21:23:20.150 | info | Client [Sonoff-Gaestebad-2ch] connection closed: closed

Any ideas how to fix that ?

ascillato2, 2018-12-24:
Hi,

@ymurawski, @GumbyMan82

Seems that you have a bad flashing problem. 

You should erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from release. 

Please, check the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk for explanations
Also check issue: https://github.com/arendst/Sonoff-Tasmota/issues/4697

closed 2018-12-24 by ascillato2

ymurawski, 2018-12-30:
So please reactivate it again. 

What i tried so far...

- complete flash delete 
- flash version 6.4.1.1 with core 2.5
- completely new pc with new install of iobroker

I activated Hue emulation because of alexa
Could this make this error ??
Maybe because iobroker detects it as hue bridge and also via the sonoff adapter ?