arendst/Tasmota #4723: Precompiled FW with MQTT Password will expose in HTML in plain text

wongnam, 2018-12-24:
If i delete the password hidden mark in MQTT password field it will expose the precompile MQTT password in plain text and even you can read it in HTML source as in my screenshot.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ :  Sonoff SV
- [x] _Development/Compiler/Upload tools used_ : VSC + PlatformIO
- [x] _If a pre-compiled release or development binary was used, which one?_ : Tasmota 6.4.0.3 core 2.4.2
- [x] _You have tried latest release or development binaries?_ :

**To Reproduce**
* try to delete the password back dot.
* View it in HTML source code

**Expected behavior**
* It should be hidden.

**Screenshots**
[image]

[image]

andrethomas, 2018-12-24:
@wongnam Thanks for reporting :)

ascillato2, 2018-12-24:
Fixed by Theo

Thank you very much for reporting :+1:

closed 2018-12-24 by ascillato2