arendst/Tasmota #4728: 6.4.0.3 update stops at 17% upload

JF62, 2018-12-25:
As described in the header each update stops after 17%.
Independent of the  version I tried to upload (minimal, basic,...). 
6.4.0.3 is compiled with core 2.5.0 (STAGE/3.0.0-dev(c0f7b44)).
Any idea how to solve this (esptool works, but I have many build in Sonoffs which are hard to flash by wire.

ascillato, 2018-12-25:
Hi,

Sorry, I can't reproduce that. 

This is clearly a wifi issue from your network setup in your router. Too many devices? Wifi interference? Channel selection in auto? Wifi repeaters?

JF62, 2018-12-25:
Hi,
Channel is fixed at 6.
Repeaters: yes

But after a flash with esptool to 6.3 with core 1.3 a update will work to 6.4.0.3. And with this (core 2.5) same error again in upload with any other .bin

JF62, 2018-12-25:
I can try, but I need repeaters because of distance to the main router.
Are you sure this is the issue? Is there a change or a known bug in core 2.5?

ascillato2, 2018-12-25:
There is no known bug that prevents the upload. I can't reproduce your issue on any of the cores (2.3.0, 2.4.2, 2.5.0).

Try the precompiled bins (there are the 3 cores version available for download) as explained in the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

closed 2018-12-25 by ascillato2

ascillato, 2018-12-25:
Ok. So the issue is in the main router. Do you have more than 30 wifi devices?

17% seems to be a wifi micro cut. Sometimes is due to too many wifi devices.

Have you tried to reboot your router?

JF62, 2018-12-25:
I have more the 30 devices - will try a reboot and come back

ascillato, 2018-12-25:
Ok. Have you tried an phone app like wifi analyzer to see if there is no other wifi network in your same channel?

JF62, 2018-12-25:
Sure there are others but what is the reason? I could not tell all guys in my  neighborhood nit to use their wlan ;-)

ascillato, 2018-12-25:
Hehehe, of course not. The test with the wifi analizer app is to choose the less crowded channel.

If you have too many wifi networks with too many wifi devices, all in the same channel, all them have to share it using just one portion of time each one. If the time available for every device is too little, this disconnections will happen. If all the time your OTA is being cancelled at 17% seems that it is occurring a timeout and the upload is being cancelled.

Please, try the wifi analyzer for choosing another less crowded channel. This reported problem is getting more and more common.

JF62, 2018-12-25:
Good to know - I get it running for some devices - some not. Is there a chance to change the timeout?

ascillato2, 2018-12-25:
The timeout of the wifi connection is in the router. AFAIK it can not be changed. It is a known limitation for 2.4ghz wifi networks to have disconnections when it is too crowded. That is why it was invented the wifi 5ghz. Btw 5ghz is not supported by esp8266 devices.

JF62, 2018-12-25:
Still no chance to update some Sonoff Basics. Tested many things including reducing the number of installed WLAN clients to less then 10. Use of fix channel. deactivation of Mesh. No idea what I can do more except update via wire (very bad solution :-( )

arendst, 2018-12-26:
Provide the output of command ststas 0

JF62, 2018-12-26:
Status 0 of a updated and still updateable device:
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS = {"Status":{"Module":1,"FriendlyName":["Wohnzimmer vorne"],"Topic":"Wohnzimmer-vorne","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.10.180:80/data/firmwares/sonoff-full.bin","RestartReason":"Software/System restart","Uptime":"0T08:20:00","StartupUTC":"2018-12-26T00:27:52","Sleep":50,"BootCount":29,"SaveCount":46,"SaveAddress":"FB000"}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS2 = {"StatusFWR":{"Version":"6.4.1.1(basic)","BuildDateTime":"2018.12.25 18:36:45","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["JF WLAN","JF WLAN"],"TelePeriod":300,"SetOption":["00000009","558180C0","000000C1"]}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS4 = {"StatusMEM":{"ProgramSize":452,"Free":548,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000407","0F082780","04008000","00000006","000001C0"]}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS5 = {"StatusNET":{"Hostname":"Wohnzimmer-vorne","IPAddress":"192.168.10.65","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.10","Mac":"xx:xx:xx:95:C1:12","Webserver":2,"WifiConfig":5}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.20","MqttPort":1883,"MqttClientMask":"Wohnzimmer-vorne","MqttClient":"Wohnzimmer-vorne","MqttUser":"Sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 26 08:47:52 2018","Local":"Wed Dec 26 09:47:52 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"08:25","Sunset":"16:29"}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS10 = {"StatusSNS":{"Time":"2018-12-26T09:47:52","Switch1":"ON"}}
09:47:52 MQT: stat/Wohnzimmer-vorne/STATUS11 = {"StatusSTS":{"Time":"2018-12-26T09:47:52","Uptime":"0T08:20:00","Vcc":3.448,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"JF WLAN","BSSId":"E0:28:6D:7B:C9:32","Channel":6,"RSSI":94}}}

Status 0 of a non updateable device:

09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS = {"Status":{"Module":1,"FriendlyName":["Wohnzimmer-hinten"],"Topic":"Wohnzimmer-hinten","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://www.rhenum.eu/SPW/firmware_6411_25.bin","RestartReason":"Software/System restart","Uptime":"0T08:18:27","StartupUTC":"2018-12-26T00:27:59","Sleep":50,"BootCount":24,"SaveCount":45,"SaveAddress":"FB000"}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS2 = {"StatusFWR":{"Version":"6.4.0.3(basic)","BuildDateTime":"2018.12.23 18:04:25","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["JF WLAN","JF WLAN"],"TelePeriod":300,"SetOption":["00008009","558180C0","000000C0"]}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS4 = {"StatusMEM":{"ProgramSize":452,"Free":548,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000407","0F082780","04008000","00000006","000001C0"]}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS5 = {"StatusNET":{"Hostname":"Wohnzimmer-hinten","IPAddress":"192.168.10.64","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.10","Mac":"xx:xx:xx:A7:D1:AA","Webserver":2,"WifiConfig":5}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.20","MqttPort":1883,"MqttClientMask":"Wohnzimmer-hinten","MqttClient":"Wohnzimmer-hinten","MqttUser":"Sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 26 08:46:26 2018","Local":"Wed Dec 26 09:46:26 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"08:25","Sunset":"16:29"}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS10 = {"StatusSNS":{"Time":"2018-12-26T09:46:26","Switch1":"ON"}}
09:46:26 MQT: stat/Wohnzimmer-hinten/STATUS11 = {"StatusSTS":{"Time":"2018-12-26T09:46:26","Uptime":"0T08:18:27","Vcc":3.442,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"JF WLAN","BSSId":"E0:28:6D:7B:C9:32","Channel":6,"RSSI":100}}}

arendst, 2018-12-26:
Both outputs seem fine although I'm not sure if 6.4.0.3 would work fine with stage core.

As there seems to be enough memory available for a single sweep update did you try to update using the gui where you select the file yourself? So not using OTA.

Where do you see the upload failed message?

If using OTA open the gui console, enter weblog 3 and perform an OTA update using command upload 1 and notice the messages shown. Wait until the screen stops updating after a minute..

JF62, 2018-12-26:
Everytime I tried both (OTA and direct upload). Direct upload stops at 17%. OTA shows:

10:46:48 CMD: upload 1
10:46:48 SRC: WebConsole from 192.168.10.183
10:46:48 RSL: Gruppe 0, Index 1, Befehl UPLOAD, Daten 1
10:46:48 MQT: stat/Wohnzimmer-hinten/RESULT = {"Upload":"Version 6.4.0.3 from http://xxxxxxxxxx/tasmota/firmware_6411_25.bin"}
10:46:48 CFG: in Flash gespeichert am FB, zählen 49, Bytes 3584
10:46:50 HTP: Web-Server angehalten
10:46:50 UPL: http://xxxxxxxxxx/tasmota/firmware_6411_25.bin
10:46:59 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64
10:47:00 HTP: Web-Server angehalten
10:47:00 UPL: http://xxxxxxxxxx/tasmota/firmware_6411_25.bin
10:47:09 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64
10:47:10 HTP: Web-Server angehalten
10:47:10 UPL: http://xxxxxxxxxx/tasmota/firmware_6411_25.bin
10:47:19 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64
10:47:20 HTP: Web-Server angehalten
10:47:20 UPL: http://xxxxxxxxxx/tasmota/firmware_6411_25.bin
10:47:29 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64
10:47:30 HTP: Web-Server angehalten
10:47:31 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64

After this ther is a restart:

00:00:00 CFG: aus Flash geladen am FB, zählen 49
00:00:00 SRC: Restart
00:00:00 Projekt sonoff Wohnzimmer-hinten Version 6.4.0.3(basic)-STAGE
00:00:00 WIF: Verbindungsversuch...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Network 0, AP-, SSId FRITZ!Box 7490, Channel 1, BSSId 34:81:C4:FB:08:50, RSSI -87, Encryption 1
00:00:04 WIF: Network 1, AP-, SSId HP-Print-08-Officejet Pro 8610, Channel 1, BSSId 48:0F:CF:CE:E0:08, RSSI -92, Encryption 1
00:00:04 WIF: Network 2, AP1, SSId JF WLAN, Channel 6, BSSId E0:28:6D:7B:C9:32, RSSI -35, Encryption 1
00:00:04 WIF: Network 3, AP-, SSId Fritzis Gastzugang, Channel 6, BSSId E2:28:6D:7B:C9:32, RSSI -35, Encryption 1
00:00:04 WIF: Network 4, AP1, SSId JF WLAN, Channel 6, BSSId 7C:FF:4D:E2:9E:64, RSSI -66, Encryption 1
00:00:04 WIF: Network 5, AP1, SSId JF WLAN, Channel 6, BSSId 7C:FF:4D:D2:28:06, RSSI -76, Encryption 1
00:00:04 WIF: Network 6, AP-, SSId Fritzis Gastzugang, Channel 6, BSSId CE:CE:1E:18:C3:AF, RSSI -79, Encryption 1
00:00:04 WIF: Network 7, AP-, SSId Fritzis Gastzugang, Channel 6, BSSId 7E:FF:4D:D2:28:06, RSSI -76, Encryption 1
00:00:04 WIF: Network 8, AP-, SSId WLAN-544524, Channel 1, BSSId E0:60:66:47:40:DB, RSSI -92, Encryption 1
00:00:04 WIF: Network 9, AP-, SSId Telekom_FON, Channel 1, BSSId B4:30:52:97:5A:9E, RSSI -93, Encryption 0
00:00:04 WIF: Network 10, AP1, SSId JF WLAN, Channel 11, BSSId 7C:FF:4D:E2:A1:F2, RSSI -60, Encryption 1
00:00:04 WIF: Network 11, AP-, SSId Fritzis Gastzugang, Channel 11, BSSId 7E:FF:4D:E2:A1:F2, RSSI -61, Encryption 1
00:00:04 WIF: Network 12, AP1, SSId JF WLAN, Channel 11, BSSId 5C:49:79:E7:96:72, RSSI -54, Encryption 1
00:00:04 WIF: Network 13, AP-, SSId Fritzis Gastzugang, Channel 11, BSSId 5E:49:79:E7:96:72, RSSI -54, Encryption 1
00:00:04 WIF: Network 14, AP-, SSId PS4-95A2ACB22F4D, Channel 11, BSSId E8:61:7E:14:E3:BF, RSSI -89, Encryption 1
00:00:04 WIF: Network 15, AP-, SSId WLAN-UU5MDX, Channel 1, BSSId B4:30:52:97:5A:9D, RSSI -91, Encryption 1
00:00:04 WIF: verbinden mit AP1 JF WLAN in Modus 11N wie Wohnzimmer-hinten...
00:00:04 WIF: Verbindungsversuch...
00:00:05 WIF: verbunden
00:00:05 HTP: Web-Server aktiv bei Wohnzimmer-hinten mit IP-Adresse 192.168.10.64
00:00:07 MQT: Verbindungsversuch...
00:00:09 MQT: verbunden
00:00:09 MQT: tele/Wohnzimmer-hinten/LWT = Online (beibehalten)
00:00:09 MQT: cmnd/Wohnzimmer-hinten/POWER = 
00:00:09 MQT: abonniere cmnd/Wohnzimmer-hinten/#
00:00:09 MQT: abonniere cmnd/sonoffs/#
00:00:09 MQT: abonniere cmnd/Wohnzimmer-hinten_fb/#
00:00:09 MQT: tele/Wohnzimmer-hinten/INFO1 = {"Module":"Sonoff Basic","Version":"6.4.0.3(basic)","FallbackTopic":"cmnd/Wohnzimmer-hinten_fb/","GroupTopic":"sonoffs"}
00:00:09 MQT: tele/Wohnzimmer-hinten/INFO2 = {"WebServerMode":"Admin","Hostname":"Wohnzimmer-hinten","IPAddress":"192.168.10.64"}
00:00:09 MQT: tele/Wohnzimmer-hinten/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/Wohnzimmer-hinten/RESULT = {"POWER":"OFF"}
00:00:09 MQT: stat/Wohnzimmer-hinten/POWER = OFF
00:00:09 APP: (UTC) Wed Dec 26 09:47:45 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
10:47:48 APP: Anzahl Startvorgänge 27
10:47:53 MQT: tele/Wohnzimmer-hinten/STATE = {"Time":"2018-12-26T10:47:53","Uptime":"0T00:00:15","Vcc":3.444,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"JF WLAN","BSSId":"E0:28:6D:7B:C9:32","Channel":6,"RSSI":100}}
10:47:53 MQT: tele/Wohnzimmer-hinten/SENSOR = {"Time":"2018-12-26T10:47:53","Switch1":"ON"}
10:49:23 HTP: Befehl
10:49:23 SRC: WebCommand from 192.168.10.20
10:49:23 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
10:49:23 MQT: stat/Wohnzimmer-hinten/STATUS = {"Status":{"Module":1,"FriendlyName":["Wohnzimmer-hinten"],"Topic":"Wohnzimmer-hinten","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[... cut ...]
10:49:23 MQT: stat/Wohnzimmer-hinten/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 26 09:49:23 2018","Local":"Wed Dec 26 10:49:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"08:25","Sunset":"16:29"}}
10:49:23 MQT: stat/Wohnzimmer-hinten/STATUS10 = {"StatusSNS":{"Time":"2018-12-26T10:49:23","Switch1":"ON"}}
10:49:23 MQT: stat/Wohnzimmer-hinten/STATUS11 = {"StatusSTS":{"Time":"2018-12-26T10:49:23","Uptime":"0T00:01:45","Vcc":3.445,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"JF WLAN","BSSId":"xx:xx:xx:7B:C9:32","Channel":6,"RSSI":100}}}

arendst, 2018-12-26:
Whow! what an abundance of AP's!!! And some with same channel and RSSI close to each other.

For direct upload did you ever try a different browser?

I'm afraid the number of AP's with same channel are making the wifi landscape unreliable.

JF62, 2018-12-26:
But this was fine until 6.3 - with 6.4 the problem starts up

arendst, 2018-12-26:
I'm quite sure you didn't use the current stage version with 6.3 as at that time that version didn't exist. 

Using stage is dangerous as the stage core is changing every day and may make certain parts of Tasmota unusable (like probably this upload feature).