arendst/Tasmota #4729: Core 2.4.2 as default?

ihavenonick, 2018-12-25:
Ab 6.4.0.3 ist der Core 2.4.2 wieder default und alle damit verbundenen Probleme.
HUE Emulation funktioniert nicht
Websend läuft nur sporadisch wenn man damit ein anderes Tasmota Gerät schalten möchte.
Ein Beispiel:
`Rule1 on Power1#State=1 do websend [192.168.100.60:80,admin:joker] POWER1 ON endon on Power1#State=0 do websend [192.168.100.60:80,admin:joker] POWER1 OFF endon`

ascillato, 2018-12-25:
Hi, sorry I can't understand your post. Please, write in English with the template completed. Thanks

closed 2018-12-25 by ascillato2

ihavenonick, 2018-12-25:
From Tasmota version 6.4.0.3 the Core 2.4.2 is default again and all related problems.
Websend runs only sporadically if you want to switch it with another Tasmota device.
An example:
`Rule1 on Power1#State=1 do websend [192.168.100.60:80,admin:joker] POWER1 ON endon on Power1#State=0 do websend [192.168.100.60:80,admin:joker] POWER1 OFF endon`

ascillato, 2018-12-25:
Please, see the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

There you have the links for other cores also. Thanks

ihavenonick, 2018-12-25:
OK, thanks for the link. Is it correct that in the Core 2.5.0 patch no longer the parsing.cpp is necessary?

ascillato, 2018-12-25:
Correct. Core 2.5.0 has Alexa bug fixed and also light_modem_sleep bug fixed.

ascillato, 2018-12-25:
Great! Thanks for sharing your tests. Very appreciated.