arendst/Tasmota #4730: Make variable/mem comparison a trigger for rules.

AalianKhan, 2018-12-25:
Hello, I was just thinking comparing variables could be a useful trigger. for example `on var1=var2 do ruletimer1 300` i know there is a way around it by using events but it takes a lot of space compared to this method.

AalianKhan, 2018-12-26:
@ascillato2 Thanks for replying.
If that is the case can you set a `Time#Minute`. that way it will check it every one minute if they are equal or not.

AalianKhan, 2018-12-26:
Or can you do that with the BREAK function?

ascillato2, 2018-12-27:
Comment from @AalianKhan, copied from https://github.com/arendst/Sonoff-Tasmota/issues/4737

> Hello,
> I want to compare two variables to see if they are equal or not. but as @ascillato said "Tasmota must know 'when' to trigger the comparison". So to work around this issue you can add the rule like this
> on Time#Minute var1=var2 do ruletimer1 300
> This way Tasmota knows when to trigger the rule.
> You can also do this Events but they can get really big and confusing really fast.
> 
> Thank you.
> Note I made this issue before but it was really open ended and I didn't tell the problem in detail so I hope this is better to understand.

ascillato2, 2018-12-27:
Hi,

Sorry I didn't explain better before.

We want to help you and also we want to improve Tasmota, so, would be better that you explain what is the case that you need this feature, May be, there is already implemented a way to satisfy your requirements and solve your use-case scenario. If not, we can evaluate to add this or another feature to solve your issue if it is needed for several users and if it is doable in a Tasmota with a small ram and flash usage.

The rule you proposed: `on Time#Minute var1=var2 do ruletimer1 300 endon` has 1 trigger and 1 conditional concatenated. It is kind of complex to add all that in Tasmota with small ram and flash fingerprint.

It is already a way to compare 2 variables using events. An event is a trigger, so you can do a conditional with a trigger. To run the event you have to tell tasmota when, that can be Time#Minute if you prefer:

```
on Time#Minute do event checkvar=%var2% endon
on event#checkvar=%var1% do ...... endon
```

but that depends obviously in what you want to achieve. So, the main question is, what is the use-case scenario you need this?

AalianKhan, 2018-12-27:
Thanks for the detailed explanation.
I am making a thermostat that has 3 relays. 1 for heating 1 for cooling and 1 for the fan. This is what I have 
```
Rule1 on Switch1#State do event toggling=%mem1% endon on event#toggling=OFF do backlog mem1 Cooling; event cooling endon on event#toggling=Cooling do backlog mem1 Heating; event heating endon on event#toggling=Heating do backlog mem1 OFF; event off endon 

on event#cooling do backlog power1 1; power2 0; power3 1 endon on event#heating do backlog power1 0; power2 1; power3 1 endon on event#off do backlog power1 0; power2 0; power3 0 endon 
```
As you can see there is a switch that toggles the state of mem1. and they turn on the relays accordingly. now I want to control it via MQTT by setting the value of mem1 to cooling etc. I was just wondering if I could just check the value of mem1 and then act accordingly it will be simple and intuitive. for this example, the rule will look something like `on Timer#minute %mem1%=cooling do event cooling endon`.

I didn't realize the fact it was a complex rule so I thought more about it and I got an idea. is it possible to make a var1#State trigger? it will trigger when the state changes and will have the value of the corresponding value/mem. The following is an example `on mem1#State=cooling do event cooling endon`. Let me know how that sounds.

Thank you for all the work you had put on maintaining the repository. 👍  @ascillato2

ascillato2, 2018-12-27:
@AalianKhan 

Your example and your rules are really good. I think that this can give some ideas for other users. If you want, please add it to the wiki at rules examples.

Your rules works just fine as they are:

```
Rule1
on Switch1#State do event toggling=%mem1% endon

on event#toggling=OFF do backlog mem1 Cooling; event cooling endon
on event#toggling=Cooling do backlog mem1 Heating; event heating endon
on event#toggling=Heating do backlog mem1 OFF; event off endon 

on event#cooling do backlog power1 1; power2 0; power3 1 endon
on event#heating do backlog power1 0; power2 1; power3 1 endon
on event#off do backlog power1 0; power2 0; power3 0 endon 
```

If you want to make them a little more clear to read (simple and intuitive as you said), may be an option could be:

```
Rule1
on Switch1#State do event THERMOSTAT=%mem1% endon

on event#THERMOSTAT=OFF do event COOLING endon
on event#THERMOSTAT=COOLING do event HEATING endon
on event#THERMOSTAT=HEATING do event OFF endon 

on event#COOLING do backlog mem1 COOLING; power1 1; power2 0; power3 1 endon
on event#HEATING do backlog mem1 HEATING; power1 0; power2 1; power3 1 endon
on event#OFF do backlog mem1 OFF; power1 0; power2 0; power3 0 endon 
```

So, from MQTT you can send just:

`cmnd/your_sonoff_topic/EVENT HEATING`

and it will work just fine saving the actual status to MEM.

ascillato2, 2018-12-27:
Ok, now if we apply the VAR to VAR checking as proposed, the rules could be something like:

_**(NOT ACTUAL TASMOTA RULES)**_

```
Rule1
on Switch1#State do event THERMOSTAT=%mem1% endon

on event#THERMOSTAT=OFF do mem1 COOLING endon
on event#THERMOSTAT=COOLING do mem1 HEATING endon
on event#THERMOSTAT=HEATING do mem1 OFF endon 

on MEM1#STATE=COOLING do backlog power1 1; power2 0; power3 1 endon
on MEM1#STATE=HEATING do backlog power1 0; power2 1; power3 1 endon
on MEM1#STATE=OFF do backlog power1 0; power2 0; power3 0 endon 
```

_**(NOT ACTUAL TASMOTA RULES)**_

So, there isn't much change to the EVENT type set of rules, so this is not adding much functionality. Sorry. Is not going to be implemented soon.

Anyway, Thanks a lot for testing and sharing your ideas. Your Thermostat rules are really good. thanks :+1:

AalianKhan, 2018-12-28:
Hello @ascillato2 ,
I'm back again with another debate. This one is specifically with the `VAR3#State`.  I want to modify var3 when it is changed using MQTT. Right now I couldn't think of any good alternatives.  you can do it by using `Time#Minute` but it will not make the change instantly. My concept is to use the following Rule 
**(These are not real rules you can use with Tasmota)**

```on VAR3#State do backlog ADD3 1; event hot_tolerance```

I couldn't think of doing something with events because how would it know if the value changed. 

Thank you

ascillato, 2018-12-28:
Hi,

By mqtt you can pass values also with events. So, instead of changing a var by mqtt, you can trigger an event with a value. 

Example:

By mqtt:

`Cmnd/sonofftopic/event setvar=26`

rule:

```
Rule on event#setvar do backlog var3 %value%; add3 1; event hot_tolerance endon
```

Also you can check the value you are passing by mqtt against another value or variable:

```
Rule on event#setvar>%var1% do backlog var3 %value%; add3 1; event hot_tolerance endon
```

If this don't satisfy what you want to achieve, please explain further what you want to do with var3.

AalianKhan, 2018-12-28:
Hello,

I had also looked into that but the problem was I am using Home Assistant. in that component it sends numbers to the command topic. you can't edit the payload with `setvar=23` it can only send `23`. 
Right now my command topic is 

`cmnd/thermostat/Var1`
 
which sets the sets `var1` to the payload eg `23`.

What I want to achieve with this is the following. var1 is the target temperature and var3 is the point where the thermostat will turn off the relays. suppose my target is 23°C, therefore, var3 should be a bit higher than 23°C. For example, if the target is 23°C then the heater should stop at 24°C. to do that I must tell an `event` to make the var3 1 degree higher than the target(var1). I can trigger the event when the target changes locally on the thermostat buttons but I am not able to trigger it when it is done via MQTT.

I hope this is clearer than before.
Thanks

P.S. the same thing happens with var4 but that is the cold tolerance, therefore, it will turn off the A/C

ascillato, 2018-12-28:
Hi

Home Assistant allows the use of a payload like `setvar=23`. How is your config for that HomeAssistant element? Which type of element are you using?

AalianKhan, 2018-12-28:
here is my config 
```YAML
climate:
  - platform: mqtt
    name: Thermostat
    current_temperature_topic: "tele/thermostat/SENSOR"
    current_temperature_template: "{{ value_json['AM2301'].Temperature }}"
    temperature_command_topic: "cmnd/thermostat/Var1"
    temperature_state_topic: "stat/thermostat/RESULT"
    temperature_state_template: "{{ value_json.Var1 }}"
    mode_command_topic: "cmnd/thermostat/event"
    mode_state_topic: "stat/thermostat/RESULT"
    mode_state_template: "{{ value_json.Mem1 }}"
    modes:
    - "OFF"
    - "Cooling"
    - "Heating"
    availability_topic: "tele/thermostat/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
```

AalianKhan, 2018-12-28:
the MQTT command will look like `cmnd/thermostat/event 23` but we want it to be `cmnd/thermostat/event setvar=23`

ascillato, 2018-12-29:
Ok. I understand. Let me think about this. Can you share all your rule sets? 

By now you can continue sending the target temperature to var1 and add the following rule:

```
Rule on time#minute do backlog var3 %var1%; add3 1 endon
```

This rule will do the adjustment for `var3` every minute. 

Let me think about this thermostat case.

Thanks for sharing your ideas and for your patience.

AalianKhan, 2018-12-29:
Ok, I will add that rule for now. Here are my Rules. Keep in mind that I am using the display version so Rule 3 is all about the display to work.
```
mem1 = mode
var1 = target temp
var2 = current temp
var3 = hot tolerance
var4 = cold tolerance
Rule1 
on Switch1#State do event toggling=%mem1% endon on event#toggling=OFF do backlog mem1 Cooling; event cooling endon on event#toggling=Cooling do backlog mem1 Heating; event heating endon on event#toggling=Heating do backlog mem1 OFF; event off endon 

on event#cooling do backlog power1 1; power2 0; power3 1; mem1 Cooling endon 
on event#heating do backlog power1 0; power2 1; power3 1; mem1 Heating endon 
on event#off do backlog power1 0; power2 0; power3 0; mem1 OFF endon 
on Switch2#State do ADD1 1 endon

Rule2 on Switch3#State do SUB1 1 endon on time#minute do backlog var3 %var1%; add3 1; var4 %var1%; sub4 1 endon

Rule3 on System#Boot do backlog var1 21; event off; event refresh endon 
on tele-AM2301#temperature do backlog var2 %value%; event refresh; event turn_off=%value% endon on Event#off do event refresh endon 
on Event#cooling do event refresh endon 
on Event#heating do event refresh endon 
on event#refresh do DisplayText [z][s1l0c0]Your Climate Control [s1l3c1]Currently %mem1% [s1l1c2]Target   Current [s1l2c2]%var1%C     %var2%C endon
```

AalianKhan, 2019-01-01:
The big difference between `event` and `var#state` is the following. For `event` you have to set the value and call it at the right time. which is difficult at times, especially for MQTT purposses. For `var1#state` you only have to set the value and it will trigger when it changes so you don't need to go through the hassle of triggering it at the right time.

ascillato2, 2019-01-03:
Working on `var#state`

ascillato, 2019-01-04:
PR https://github.com/arendst/Sonoff-Tasmota/pull/4819

ascillato2, 2019-01-04:
Thanks everyone for sharing their ideas. :+1: 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-04 by ascillato2

ascillato, 2019-01-05:
@AalianKhan 

Hi,

The new trigger has been merged.

Please, could you make a full example of a thermostat in the HomeAssistant Chapter of the Tasmota wiki?

Thanks :+1:

ascillato, 2019-01-06:
Gabriel, merged means that it is already in the code in the development branch.

And your rules and logs are correct. What is the question?

gabrielklein, 2019-01-06:
Why is the rule fired even if I always set the same value to var1. I expect the rule to be fired when I change the value....

ascillato, 2019-01-06:
The rule is fired everytime the VAR is modified. No matter if the value you put is the same.

Makk1983, 2019-01-26:
Hello, I tried to use the following syntax Var1 # state in the rules to run a command every time you modify the variable VAR1 but it does not work! Here is the example I was running: 

rule1 on var1 # state do power1 1 endon

where am I wrong?

Makk1983, 2019-01-26:
oh I forgot, my version is 6.4.1 I suppose the last one

ascillato, 2019-01-26:
This feature is new. It is not in 6.4.1
Latest Tasmota version from development branch is 6.4.1.11

Makk1983, 2019-01-26:
Where can I download version 6.4.1.11?

ascillato, 2019-01-26:
http://thehackbox.org/tasmota/

Makk1983, 2019-01-27:
Thanks for giving me the link and for helping to solve my problem, I wanted to ask a question maybe OT, I saw in the link he gave me that there are releases to 4 mb, what is the advantage? Are there any more features? thanks a lot

balajisraghavan, 2019-02-09:
Does not work for ADDn commands. For e.g the `power off` command in the below rule never fires if `ADD1` is used. But works fine if `VAR1` is used.

`rule1
on system#boot do 
	VAR1 0
endon 
on ENERGY#Current>0.619 do
	VAR1 0
endon
on Time#Minute do 
	ADD1 1
endon  
on var1#STATE=1 do
	power off
endon`

ascillato, 2019-02-09:
@balajisraghavan

Please, use latest version from the development branch (6.4.1.14)

If it is not working for you that means you have an old version. Thanks

balajisraghavan, 2019-02-17:
@ascillato  I'm on Sonoff-Tasmota 6.4.1.15. The greater than operator for comparision works, equals does not.

So `var1#STATE>1` works while `var1#STATE=1` never triggers

ascillato, 2019-02-17:
Please update to latest Tasmota 6.4.1.17 and use `var1#state==1`

Thanks

laurentdong, 2019-03-07:
Sorry, guys.
I raised this issue because I think the implementation of MQTT Subscription can perfect solve this use case.
The Var/Mem event trigger are still not released at this moment. I would suggest a discussion about necessity of this feature. While it did not spend too much code space and global memory, the check of Var/Mem changes are fired every 50ms even if there's no any rule trigger catch this event.

bary73, 2022-11-20:
How to set a rule so that when event#t1 is running, event#t2 will be turned on and event#t1 will be turned off ?

 
Rule1
 on DS18B20-1#Temperature do Backlog event  t1_demand=%value%;DisplayText [f0s2p15x0y0]Tb. %value% C endon
  on DS18B20-2#Temperature do Backlog event  t2_demand=%value%;DisplayText [f0s2p15x0y50]Tcwu %value% C endon 
 on Time#Minute|4 do DisplayText [z] endon
 on event#t1_demand>%mem1% DO Backlog dimmer 45;DisplayText [f0s2p15x0y25]Power = 10% break
 on event#t1_demand>%mem2% DO Backlog dimmer 58;DisplayText [f0s2p15x0y25]Power = 20% break
 on event#t1_demand>%mem3% DO Backlog dimmer 64;DisplayText [f0s2p15x0y25]Power = 30% break
 on event#t1_demand>%mem4% DO Backlog dimmer 73;DisplayText [f0s2p15x0y25]Power = 40% break
 on event#t1_demand>%mem5% DO Backlog dimmer 84;DisplayText [f0s2p15x0y25]Power = 60% break 
 on event#t1_demand>%mem6% DO Backlog dimmer 94;DisplayText [f0s2p15x0y25]Power = 80% break
 on event#t1_demand<%mem6% DO Backlog dimmer 100;DisplayText [f0s2p15x0y25]Power = MAX endon
 on event#t2_demand<%mem7% DO Backlog dimmer 73;DisplayText[z][s2l1c2]CWU = ON  break
 on event#t2_demand>%mem8% DO rule1 1 endon

 mem1 40; mem2 39; mem3 38; mem4 37; mem5 36; mem6 35; 
mem7 39; mem8 44