arendst/Tasmota #4731: Add not equal (!=) for rules

AalianKhan, 2018-12-26:
pretty straight forward. Add != for triggers

ascillato, 2018-12-26:
Use case scenario?

AalianKhan, 2018-12-26:
```
rule
on event#power!=0 do Power1 1 endon
```

AalianKhan, 2018-12-26:
@ascillato 
```
rule
on event#power1!=0 do Power1 1 break
on event#power1=1 do var1 1 endon
```

makejoint, 2019-01-21:
@ascillato2 what about 

... `SHT1X#Temperature!=%var1%` ... when `var1` is backlogged?

ascillato, 2019-01-21:
Hi

Temperature is a float number. So, what use case will require:

If temp!= 25 do
If 25.21!= 25 do

ascillato, 2019-01-21:
I understand the math and then logic of your example. But in which use case scenario would be this useful for any automation.

makejoint, 2019-01-21:
e.g., heating control... or simpler rules in the following case: https://github.com/arendst/Sonoff-Tasmota/issues/4958#issuecomment-455131887

in order to send my sensor state (temperature) via MQTT, I need to use two variables, instead of one:

`Rule1 on tele-SHT1X#Temperature do var1 %value% endon on SHT1X#Temperature>%var1% do backlog var1 %value%; publish stat/sonoff-WZ-fan/TEMPERATURE %value%; var2 %value%; endon on sHT1X#Temperature <%var2% do backlog var2 %value%; publish stat/sonoff-WZ-fan/TEMPERATURE %value%; var1 %value%; endon`

instead of something simpler like

`Rule1 on tele-SHT1X#Temperature do var1 %value% endon on SHT1X#Temperature!=%var1% do backlog var1 %value%; publish stat/sonoff-WZ-fan/TEMPERATURE %value%
`

I'm not sure it would work the way I described.

ascillato, 2019-01-21:
Ok, 

I understand. Your rules is a modification of the example:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8a-sending-the-value-of-a-sensor-to-mqtt-only-when-a-delta-is-reached

But with delta = 0

So, to not receive a mqtt every second, you are reducing the tempres to 0 (decimal resolution) that this will become again the example of the wiki but with delta = 1

One way or the other, this has the same output. You can control the delta using 2 vars or as you are proposing you are reducing resolution acting as a delta.

Personally I prefer the example of the wiki because you can have full temp resolution and also you can set the delta you want like changes to 0.01, or changes in 2 degrees up and 1 down, etc. 

Let me think about the !=

Thanks