arendst/Tasmota #4732: sonoff POW R2 V2 - use of GPIO4 and GPIO5

nromensky, 2018-12-26:
Dear Theo! Thank you very much for such a good product! here are couple question on POW R2V2

(1) there are GPIO4 and GPIO5, any plan to use them for sensors? currently when we select POW R2 we do not have any additional GPIOs.

(2) wiki states "Also do not connect any additional sensors to serial pins until you are 100% sure. It can at least destroy your Pow!". is that the case for V2R2?

wongnam, 2018-12-26:
As i know the Pow that it is live leak. so, be carefuly the electric shock via gpio.

Jason2866, 2018-12-26:
Never, ever connect sensors to POW! Due to design there can be live AC!
No support fot that scenario, because it is dangerous (and not allowed...)!

ascillato2, 2018-12-26:
Do not use sensors in a pow. This was already addressed in several issues. Please review safety in electricity. It is not worth to risk yourself, your family and your home for that. Use a nodemcu instead for the sensors you want.

closed 2018-12-26 by ascillato2