arendst/Tasmota #4734: How save current date time to mqtt topic?

clientgithub, 2018-12-26:
Hi everyone! 
Please help, how send to mqtt topic current date and time on example 
format: 2018-12-26T10:58:30? I want save date and time event to retain topic. Thank you.

ascillato, 2018-12-26:
Sorry, that is not supported.

clientgithub, 2018-12-26:
Thank you for answer, but this very strange.. Elementary function..

ascillato, 2018-12-26:
The mqtt messages already contain the date and time. There is no need to add another time and date field. This date and time is already stored in your mqtt broker database if the message is retained. See example below (the date and time in grey letters are the one saved by the mqtt broker) :

[image]

clientgithub, 2018-12-26:
Yes, the time and date of the received message is shown, but if at the time of the message the client was disconnected, the next time the message is displayed not the time of the message, but the time when the message was received.

ascillato, 2018-12-26:
Tasmota does not have a buffer for mqtt messages. If the broker is down, the message will not be retransmitted.

clientgithub, 2018-12-26:
I meant not to turn off the broker, but to turn off the client on the phone, for example, the IoT MQTT Dashboard application.

clientgithub, 2018-12-26:
I tried again. The alarm went off at 4:10 pm, and the time in IoT MQTT Dashboard was 16:42. This is the time to connect to the broker (

[image]

clientgithub, 2018-12-26:
My experiment next again) If i subscribe to topic STATE, SENSOR, RESULT, date and time saving ok. But if i subscribe to my topic,date and time connecting. I send retain message from rules with publish2 option...

ascillato2, 2018-12-27:
Ok, the actual Tasmota format for tele messages is:

```
{"Time":"2018-12-27T12:52:57","Uptime":"10T16:40:54","Vcc":3.471,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":2,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":2,"RSSI":82}}
```

So, what do you think to add a new variable `%timestamp%` to be used in a rule to allow for example:

```
publish stat/topic/sensor {"Time":"%timestamp%","mysensor":"%var1%"}
```

output:
```
{"Time":"2018-12-27T12:52:57","mysensor":"1"}
```

clientgithub, 2018-12-27:
Wow!  Thank you very much!!! it will be very convenient to use, and I think many users will help!

ascillato, 2018-12-27:
Added new variable %timestamp% with PR https://github.com/arendst/Sonoff-Tasmota/pull/4749

```
14:48:53 CMD: rule
14:48:53 MQT: living/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":437,"Rules":"on power1#state=1 do publish stat/sonoff/TIME {"Time":"%timestamp%"} endon"}

14:48:56 MQT: living/stat/RESULT = {"POWER1":"ON"}
14:48:56 MQT: living/stat/POWER1 = ON
14:48:56 RUL: POWER1#STATE=1 performs "publish stat/sonoff/TIME {"Time":"2018-12-27T14:48:56"}"
14:48:56 MQT: stat/sonoff/TIME = {"Time":"2018-12-27T14:48:56"}
```

closed 2018-12-27 by ascillato2

dac2020, 2022-08-12:
I know this is old but I'm brand new to tasmota so trying to learn as I go.

Is this thread saying that it doesn't matter if the tasmota device has the right time or not and that the broker is what matters?

I'm asking because my device shows;
tele/garden/SENSOR {"Time":"1970-01-02T04:11:38"

I've set two NTP servers but the device keeps sending the same 1970 time so I wanted to set the time on it but could not and still cannot find a way of doing so. Then I came across this and now I'm a little confused.

Wondering if someone can enlighten me a little.

barbudor, 2022-08-12:
@dac2020 not only it's old but your question doesn't relate to the original thread question
Your Tasmota must receive a proper time from NTP to behave correctly. If you set some NTP servers but tasmota doesn't get it, you probably have something wro g in your

dac2020, 2022-08-12:
Ok, I'll open a new discussion. Can you delete mine above or if I do, can you delete your erply?