arendst/Tasmota #4745: Tasmota rules: rule on Switch2#Status doesn't send mqtt message

R4C3R, 2018-12-27:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ I don't have discord] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used :Tasmota OTA 6.4.0 and 6.4.1
- [X] Hardware used : Wemos D1 mini
- [X] If a pre-compiled release or development binary was used, which one? :sonoff.bin
- [X] You have tried latest release or development binaries? : yes, upgraded to the latest release
- [X] Provide the output of command ``status 0`` :
```
13:47:56 CMD: status 0
13:47:56 SRC: WebConsole from 192.168.1.144
13:47:56 RSL: Received Topic /status, Data Size 1, Data 0
13:47:56 RSL: Group 0, Index 1, Command STATUS, Data 0
13:47:56 MQT: stat/Garage/STATUS = {"Status":{"Module":18,"FriendlyName":["Garage-Switch","Garage-Sensor"],"Topic":"Garage","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:47:56 MQT: stat/Garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:34","StartupUTC":"2018-12-27T12:21:22","Sleep":50,"BootCount":24,"SaveCount":242,"SaveAddress":"F4000"}}
13:47:56 MQT: stat/Garage/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:47:56 MQT: stat/Garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rolfs_TP-LINK4",""],"TelePeriod":300,"SetOption":["00008009","55A180C0","00000000"]}}
13:47:56 MQT: stat/Garage/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
13:47:56 MQT: stat/Garage/STATUS5 = {"StatusNET":{"Hostname":"Garage-7101","IPAddress":"192.168.1.69","Gateway":"192.168.1.50","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.50","Mac":"84:F3:EB:75:DB:BD","Webserver":2,"WifiConfig":5}}
13:47:56 MQT: stat/Garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_75DBBD","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:47:56 MQT: stat/Garage/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 27 12:47:56 2018","Local":"Thu Dec 27 13:47:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
13:47:56 MQT: stat/Garage/STATUS10 = {"StatusSNS":{"Time":"2018-12-27T13:47:56","Switch2":"OFF"}}
13:47:56 MQT: stat/Garage/STATUS11 = {"StatusSTS":{"Time":"2018-12-27T13:47:56","Uptime":"0T00:26:34","Vcc":2.917,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"rolfs_TP-LINK4","BSSId":"30:B5:C2:B1:7F:06","Channel":9,"RSSI":100}}}

```
The Wemos D1 mini is used to open a garage.

D1 GPIO5 Relay1
D2 GPIO4 Button1
D3 GPIO0 Switch2
D4 GPIO2 Relay2

Button1 simulates a button press on the garage door remote by enabling Relay1 during a certain pulse time (domoticz idx:14)
Switch2 is a reed contact that switches a virtual relay (Relay2) this relay is not present, as it is a virtual relay. The status of the switch tells me if the garage door is open or not. (switch closed = door closed, switch open=door is open) (domoticz idx:15)
I want to use this status to send a mqtt message.

stat/Garage/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":396,"Rules":"on Switch2#State do publish domoticz/in {"idx" : 14,"nvalue" : %value%,"svalue1" : "","Battery":100,"RSSI":0} endon"}

This is how the rule is currently configured, but somehow tasmota doesn't send the message. I have verified this by checking the console as well as the domoticz log files.

After spending a significant amount of time finding a solution, I'm stuck.

Maybe someone can point me in the right direction.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-27:
Gpio0 as a switch is a bad idea. If your switch state is gnd, after a reboot your device will be in programming mode

R4C3R, 2018-12-27:
Thanks for the recommendation on GPIO0 and I will try switchmode2, but I think I already tried that. I'll let you know shortly if it worked.

R4C3R, 2018-12-27:
Unfortunately it doesn't work. Here is the output from the console:

15:52:20 CMD: switchmode2 1
15:52:20 SRC: WebConsole from 192.168.1.144
15:52:20 RSL: Group 0, Index 2, Command SWITCHMODE, Data 1
15:52:20 MQT: stat/Garage/RESULT = {"SwitchMode2":1}
15:52:20 CFG: Saved to flash at F5, Count 257, Bytes 3584
15:52:27 HTP: Main Menu
15:52:29 SRC: WebGui from 192.168.1.144
15:52:29 MQT: domoticz/in = {"idx":15,"nvalue":1,"svalue":"","Battery":31,"RSSI":10}
15:52:29 MQT: stat/Garage/RESULT = {"POWER2":"ON"}
15:52:29 MQT: stat/Garage/POWER2 = ON
15:52:29 SRC: MQTT
15:52:30 CFG: Saved to flash at F4, Count 258, Bytes 3584
15:52:46 HTP: Console

Tasmota is simply not sending any message

ascillato2, 2018-12-27:
Another thing:

Your publish in your rule should be without spaces.

The publish works fine with the following in a wemos with latest Tasmota:

- Setting a GPIO as switch2 (no need for relay2)
- Copying and pasting your rule (without spaces) 
- Setting switchmode to follow (switchmode2 1)
- Activating rules (rule1 1) - (switchtopic needs to remain in 0 to allow the rule to work)

```
11:49:51 CMD: switchmode2
11:49:51 MQT: living/stat/RESULT = {"SwitchMode2":1}

11:49:54 CMD: rule
11:49:54 MQT: living/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":402,"Rules":"on Switch2#State do publish domoticz/in {"idx":14,"nvalue":%value%,"svalue1":"","Battery":100,"RSSI":0} endon"}

11:49:56 RUL: SWITCH2#STATE performs "publish domoticz/in {"idx":14,"nvalue":0,"svalue1":"","Battery":100,"RSSI":0}"
11:49:56 MQT: domoticz/in = {"idx":14,"nvalue":0,"svalue1":"","Battery":100,"RSSI":0}

11:49:58 RUL: SWITCH2#STATE performs "publish domoticz/in {"idx":14,"nvalue":1,"svalue1":"","Battery":100,"RSSI":0}"
11:49:58 MQT: domoticz/in = {"idx":14,"nvalue":1,"svalue1":"","Battery":100,"RSSI":0}
```

Please, recheck your config. Thanks.

ascillato2, 2018-12-27:
The rules are triggered with the physical switch. The webGUI buttons controls directly the relays (they are not switches, so they will not trigger rules)

closed 2018-12-27 by ascillato2

R4C3R, 2018-12-27:
ok, that explains a lot! thanks! I can see the rules sending the message now, thanks!

R4C3R, 2018-12-27:
Thank you for your help, it is much appreciated!